JSBSim-Team/jsbsim #1: Long-time developers' help appreciated

agodemar, 2017-12-17:
This repo is under the github _organization_ named "JSBSim-Team". I've created the organization and the repo in agreement with Jon S. Berndt, the primary architect and coordinator of JSBSim development team.
The first commit is a sync of the official repository that I have cloned from Sourceforge with the command:
`git clone https://git.code.sf.net/p/jsbsim/code jsbsim-code` 
Jon and me are the first two members of this team, but other developers willing to join this effort are welcome, especially the long-time developers and users.

bcoconni, 2018-02-25:
Hi Agostino,

Is the idea now to host the JSBSim development on GitHub rather than on SourceForge ?

agodemar, 2018-02-25:
Hi Bertrand,
Yes, after a discussion with Jon he gave me the green light to start a repository here on GitHub. I think there are more tools on this framework. And there are more options, including: continuous integration, wiki, GithUb Pages, issue management, etc.

bcoconni, 2018-02-25:
Ok, fine but it seems I don't have commit rights on this project. I tried to push the last commits that have been made to SourceForge repo but failed with an error about not having permissions to push.

agodemar, 2018-02-25:
I've changed the current members role to 'Owner'. Now you guys should have commit rights.
Cheers

bcoconni, 2018-02-26:
I just updated the code here to the bleeding edge so it works.
Given the disruption that SourceForge is facing at the moment, I will move to GitHub for my JSBSim development.
Thanks Agostino !

agodemar, 2018-02-27:
One thing I would like to test are the Continuous Integration features in GitHub. We'll have fresh executables of the bleeding edge version

dpculp, 2018-04-01:
Hello Agostino,  could you please add me to the JSBSim-Team?  Thank you,  Dave Culp.

agodemar, 2018-04-01:
@dpculp Hello, I just added you to the team. Welcome!

WillemEerland, 2018-05-01:
@bcoconni @agodemar I just made an attempt at setting up CI for the project on Travis CI

See the result here:

[image]

Two things that would greatly improve it:
1. Add a flag to install in the home directory, so "sudo" powers are not needed
2. This bug test fails, thus failing the build - maybe remove / setup as expected fail?

Other than that, it may be a good starting point?

bcoconni, 2018-05-01:
@WillemEerland Thanks for setting this up. This is definitely a good starting point.
> 1. Add a flag to install in the home directory, so "sudo" powers are not needed

You can customize the path to which JSBSim will be installed via the `CMAKE_INSTALL_PREFIX` flag of CMake such as
```bash
cmake -DCMAKE_INSTALL_PREFIX=/path/to/install -DINSTALL_PYTHON_MODULE=ON ..
```
This should remove the need for "sudo" powers.

> 2. This bug test fails, thus failing the build - maybe remove / setup as expected fail?

I am pretty sure the test `CheckFGBug1503` is failing because it needs the Python module `scipy` which is not in the list of the modules requested by your `pip install` command.

bcoconni, 2018-05-01:
I have just updated the README to document the flag `CMAKE_INSTALL_PREFIX` and the required Python modules for the test suite.

WillemEerland, 2018-05-01:
It's passing now. You have to activate travis yourself (just login with your github credentials) and copy the right details to add a nice button in the readme file. Oh yeah, and add this file to the repo.

Voila, CI all setup

.travis.yml
```
language: python
python:
  - "2.7"
#  - "3.6"
# command to install dependencies
notifications:
  email: false
before_install:
  - pip install cython numpy pandas scipy
  - mkdir build
  - cd build
  - cmake -DCMAKE_INSTALL_PREFIX=~/ -DINSTALL_PYTHON_MODULE=ON ..
  - make
install:
  - make install
# command to run tests
script:
  - ctest # or py.test for Python versions 3.5 and below
```

agodemar, 2018-05-01:
Excellent! Thank you @WillemEerland

bcoconni, 2018-05-01:
OK. Thank you @WillemEerland.

I just set up Travis CI according to your instructions and it works ! I have added the nice button to the README file.

One of the tests is failing (I can reproduce the problem locally on my computer so Travis CI reports a genuine test failure from JSBSim. I guess it is linked to a recent commit). That's great ! Well somehow...

seanmcleod, 2018-05-01:
@bcoconni I see the test failing, TestInitialConditions.py makes use of 737_cruise.xml which I recently committed with examples of using the template functions. In particular it has an output element. I wonder if this is causing a conflict with the output.csv that the test script is expecting/checking etc.

I can't currently run the python test code with my python 3 only installation. But glancing at the test I'm guessing this might be the issue.

agodemar, 2018-05-01:
> That's great ! Well somehow...
In only a few weeks, we have our own CI and a parallel effort to document this project. Not bad at all.

bcoconni, 2018-05-01:
> One of the tests is failing (I can reproduce the problem locally on my computer so Travis CI reports a genuine test failure from JSBSim...

Ok fixed  via commit 74c85f35. We now have a nice green flag.

bcoconni, 2018-05-18:
FYI, I have posted a news blog post on [JSBSim SourceForge project page](https://sourceforge.net/p/jsbsim/news/2018/05/jsbsim-devlopment-has-migrated-to-github/) to announce that we have migrated the development to GitHub.

agodemar, 2018-05-18:
Sounds good to me.

bcoconni, 2018-06-04:
I pushed the commit 3946586 that makes Travis CI to automatically build the C++ API documentation with `doxygen` each time a commit is pushed to the repository. If `doxygen` succeeds, the documentation is then automatically deployed on the "GitHub pages" of the JSBSim project https://jsbsim-team.github.io/jsbsim/

From now on, we have no excuses not to have an up to date API doc ! That's the beauty (or the curse ?) of continuous integration :-)

bcoconni, 2020-11-08:
I have just updated the README to show off a bit :smile:

I have added a reference to NASA verification check cases, a list of software that uses JSBSim in a new section "Applications & Usages" and, of course, mentioned that JSBSim was used in the DARPA competition where a USAF fighter pilot was defeated by an AI. Also rather than adding a list of academic papers mentioning JSBSim, I have used a link to the citations returned by Google Scholar.

The general idea is that visitors understand what an FDM is used for and that JSBSim is a robust and mature software (it is used by the NASA, DARPA and mentioned in academic papers)

Feel free to update to your liking if you have some references to add or if you spot a typo/misspelling/strange wording.

closed 2020-12-11 by bcoconni