JSBSim-Team/jsbsim #9: Clean up build system files

agodemar, 2018-04-01:
Since JSBSim has moved to CMake and having realized that the autotools no longer work, I guess we no longer need to maintain them. To make things clean I suggest to delete all the obsolete files in JSBSim main directory.

Regarding Visual Studio, Microsoft has included CMake to VS2017 and everything builds like a breeze. So according to @bcoconni suggestion we should drop VS support at some point and just keep CMake alone.

bcoconni, 2018-04-01:
Well, @seanmcleod mitigated my initial statement :-)
The idea would be to drop the autotools build files but we could keep VS build files for version earlier than 2017.

agodemar, 2018-04-01:
Agreed.
So: we keep only the CMake files and a VS solution earlier than 2017.

bcoconni, 2018-04-01:
Autotools build files have just been removed via commit adec698

WillemEerland, 2018-04-18:
I build jsbsim and ran a test script in Linux (building directly in the directory, and I build the Python libraries). Sharing my approach here: 

```
git clone git@github.com:JSBSim-Team/jsbsim.git
cd jsbsim
```

(*) make sure your path points to Python 2 (I use Anaconda for this)

```
cmake -BUILD_SHARED_LIBS -INSTALL_PYTHON_MODULE .
make
```

(*) test the build

`python tests/CheckTrim.py .`

This works for me, however I'm struggling a bit with making the jsbsim available everywhere, possibly because I'm using an Anaconda environment. I would prefer to use:

`pip install ./tests/ `

however, this returns "error: can't copy 'tests/jsbsim.so': doesn't exist or not a regular file".

Other than that, at least I got the core build, hopefully it can be helpful in updating the README file

bcoconni, 2018-04-28:
@WillemEerland Thanks for the report, I will investigate about making the JSBSim Python module installable via pip and let you know.

bcoconni, 2018-04-29:
@WillemEerland Not sure at the moment how to make a package compatible with pip. But have you tried the command `python setup.py install` ?
The file `setup.py` should be located in `build\tests` where `build` is the directory where you built JSBSim.

Alternatively you can run the following commands from your build directory. However this will install JSBSim C++ headers and libraries in addition to the Python module.
```
cmake -DINSTALL_PYTHON_MODULE=ON /path/to/JSBSim/root/directory
make
make install
```

bcoconni, 2018-04-29:
Woohoo  #! I finally updated the README files in the commit 659458a9.
Updates or corrections are welcome. Let me know if something is missing or unclear.

bcoconni, 2018-04-30:
@WillemEerland Thanks for your feedback! I just pushed a new commit 7f2ff742 that should fix the two errors that you reported
- `CMakeLists.txt` has been fixed to reference `README.md` instead of `README`
- The correct command to install manually the Python module is `python tests/setup.py install` that should be called from the `build` directory.

I took the opportunity of this commit to fix a few other minor details as well.

bcoconni, 2018-05-01:
@agodemar I guess we can close this issue since the only remaining open topic is addressed in the ticket #14 ?