JSBSim-Team/jsbsim #15: Make Python methods name consistent with C++ code

bcoconni, 2018-04-30:
At the moment, the JSBSim Python module is using underscore separating convention for methods name whereas the C++ code using CamelCase.
For example, the C++ method `FGFDMExec::LoadModel` is named `FGFDMExec.load_model` in the Python module.

The plan would be to move all the Python methods name to CamelCase convention just as the C++ code for consistency.

@seanmcleod and @WillemEerland since you have reported to have successfully built and used the Python module, would that change be acceptable to you ?

WillemEerland, 2018-04-30:
Who can say no to consistency?

mikepsn, 2018-05-28:
While consistency can be a good thing, the convention in the Python community is to use underscores for methods and functions and camel case for classes. 

The Python community conventions are to follow the [PEP 8 -- Style Guide for Python Code](https://www.python.org/dev/peps/pep-0008/). 

I've first used the C++ version of JSBSim back in 2003, and I'm now planning on revisiting it but using it from a Python project. Most modern Python projects run their code through automatic linters or style guide checkers like pep8 to check the code is following the community conventions. 

TL:DR; keep the underscores please.

bcoconni, 2018-05-31:
@mikepsn 
> the convention in the Python community is to use underscores for methods and functions and camel case for classes.

Sure, but PEP 8 specifically allows mixedCase *"in contexts where that's already the prevailing style, to retain backwards compatibility"* so I can assert that due to the fact that JSBSim Python module is merely a wrapper to the C++ code, then it is falling in this category.

bcoconni, 2020-10-19:
So I never proceeded with this changed and I guess anyway that the Python interface has been released for too long a time now to consider a renaming of the methods as a serious option.

🎉 I hereby officially drop that project and let JSBSim be (somewhat) conformant to PEP 8 🎉

closed 2020-10-19 by bcoconni