JSBSim-Team/jsbsim #18: Make our own CI pipeline deliver binaries for Linux and Windows

agodemar, 2018-05-02:
We could provide users with the results of our automatic build. The binearies could be downloaded via a "Download" section of this GitHub repository. Having both Linux and Windows binaries (x64) would be nice.

agodemar, 2018-05-05:
I have set up automated builds of JSBSim for Win64 (Release). The builds are managed via AppVeyor and are based on Sean's JSBSim.vcxproj project file. The build environment on AppVeyor site uses Visual Studio 2015. The `JSBSim.exe` artifact is uploaded automatically on our GitHub repository as a specifically tagged release ([see for example here](https://ci.appveyor.com/project/agodemar/jsbsim/branch/master)). The builds are found [in the Releases section](https://github.com/JSBSim-Team/jsbsim/releases).

This is a preliminary setup. Since our official build workflow is based on CMake, future Win64 releases should be accomplished with this approach.

seanmcleod, 2018-05-06:
@agodemar I noticed that AppVeyor supports Visual Studio 2017, but if you don't select a specific version then it defaults to VS2015. So in theory we should be able to select VS2017 and then make use of it's CMake support to use CMake to build the Windows binaries as well.

See - https://www.appveyor.com/docs/build-environment/

I initially looked at Travis in terms of Windows build support but I see there was a comment from a couple of years ago stating that they were going to support Windows but it doesn't look like it's happened yet.

agodemar, 2018-05-06:
@seanmcleod Thanks for the suggestion, I'll look into that and make the switch to VS 2017. I've seen that they also support both Python 2.7 and 3.6 x64, so we can set up the present test suite and in future the new one based on Python 3.

agodemar, 2018-05-06:
Now we build on AppVeyor with Visual Studio 2017, Release, x64 from JSBSim.vcxproj

bcoconni, 2018-05-08:
@agodemar Unless I am mistaken, every commit is now tagged with `JSBSim-win64-v1.0.x`. It seems to be a bit excessive doesn't it ? Each commit has its own unique ID, what is the purpose to add a tag over it ? Isn't there a means to a release a Windows executable without tagging every commit ?

agodemar, 2018-05-10:
@bcoconni I've set a unique tag `JSBSim-win64-v2018a` for our deployment workflow via AppVeyor. Hoping that upcoming automatic builds will overwrite the old releases with the same tag.

bcoconni, 2018-05-10:
@agodemar OK thanks. We'll see if that works.

I'm currently investigating how to make releases from Travis and AFAICS it requests tags as well. It seems however that Travis can use existing tags so I hope we should be able to use common tags between Travis and AppVeyor.

agodemar, 2018-05-10:
@bcoconni Apparently a new build triggered by a new commit was successfully uploaded to GitHub as a release with the same tag. From the last few messages in the log you'll see that: 

- AppVeyor deletes the existing release asset "JSBSim.exe" in the GitHub repository.
- Uploads the newly built "JSBSim.exe" to release assets with the same tag.

```
  101 Warning(s)
    0 Error(s)
Time Elapsed 00:01:34.18
Discovering tests...OK
Collecting artifacts...
Found artifact 'Release\JSBSim.exe' matching 'Release\JSBSim.exe' path
Uploading artifacts...
[1/1] Release\JSBSim.exe (1,384,448 bytes)...100%
Deploying using GitHub provider
Creating "JSBSim-win64-v2018a" release for repository "JSBSim-Team/jsbsim" tag "JSBSim-win64-v2018a" commit "4b41c844281fb941145cb53309d63515b80dbba8"...Skipped - release with tag "JSBSim-win64-v2018a" already exists
Deleting existing release asset "JSBSim.exe"...OK
Uploading "JSBSim.exe" to release assets...OK
Build success
```

See the whole log [here](https://ci.appveyor.com/project/agodemar/jsbsim/branch/master).

bcoconni, 2018-05-15:
Made a first try for Trusty (Ubuntu Linux) binary upload. It works on my fork but fails on `master`. I think this is due to the API token that needs to be modified.

I will investigate that later.

agodemar, 2018-05-16:
FYI
Now also AppVeyor supports Linux builds. Ubuntu images are available to all accounts, free and paid alike. The current working image users will see is based on Ubuntu 16.04 (Xenial Xerus) and a new one based on Ubuntu 18.04 (Bionic Beaver) will be added soon.

There is the [Getting started with AppVeyor for Linux guide](https://www.appveyor.com/docs/getting-started-with-appveyor-for-linux/) with a list of supported features and software pre-installed on the image.

bcoconni, 2018-05-16:
Thank you but no, thank you :-)

I spent hours reading and understanding Travis documentation to set up
something which is about to work and have not the motivation to restart the
process from the ground up with yet another CI client.

bcoconni, 2018-05-16:
I just fixed the issue with the API token (I was trying to use Travis token instead of GitHub token)

bcoconni, 2018-05-17:
> Now you should make sure that each new deploy from Travis to GitHub always uses the same tag: `JSBSim-trusty-v2018a`.

Yes, that's what I'm trying to do... without much success so far.

bcoconni, 2018-05-17:
Not there yet but at least Travis is now releasing builds with a legible name `JSBSim-trusty-v2018a-{build}` rather than the cryptic `untagged-{travis-id}` (commit 4e2869b)

agodemar, 2018-05-19:
@bcoconni It's weird that all the builds on travis are successful but the button in the README file is red and says that builds fail.

bcoconni, 2018-05-20:
@agodemar The windows build from AppVeyor seems to be stuck to commit faf4f75 despite the fact that AppVeyor successfully built the last commit. However there is a **skipped** message that is suspect about the deployment
```
Deploying using GitHub provider
Creating "JSBSim-win64-v2018a" release for repository "JSBSim-Team/jsbsim" tag "JSBSim-win64-v2018a" commit "9a24aabc5b83d74fa9dcd4d126e27de53ecbb507"...
Skipped - release with tag "JSBSim-win64-v2018a" already exists
Deleting existing release asset "JSBSim.exe"...OK
Uploading "JSBSim.exe" to release assets...OK
Build success
```

Also could be please add the aeromatic executable to the list of files included in the release ?

bcoconni, 2018-05-20:
FYI I have updated the Travis process to release an Ubuntu package which is more in line with the way a project is distributed on Linux.
For the moment, I will keep the executables `JSBSim` and `aeromatic` in the releases but I plan to drop them in favor of the Ubuntu package alone since the executables are only built for Ubuntu Trusty  anyway.

bcoconni, 2018-05-21:
Just pushed a new commit 2594021 that splits the Ubuntu/Debian package in 2 parts `runtime` and `devel`. The former contains the executables `JSBSim` and `aeromatic` as well as the required shared libraries. The latter contains the files needed for development (headers and static libraries).

We are getting closer to a normal release process for Linux packages so I also took this opportunity to remove the stand-alone executables from the release.

Next step, build an Ubuntu/Debian package for the python module.

bcoconni, 2018-05-30:
> Next step, build an Ubuntu/Debian package for the python module.

Done via commit 3dd1b95

bcoconni, 2018-05-31:
@agodemar  
> Now you should make sure that each new deploy from Travis to GitHub always uses the same tag: `JSBSim-trusty-v2018a`.

Done via commits e5cc599 and 496c29f. I declare this issue closed for 
- [x] Linux binary releases (Travis Ci)
- [ ] Windows binary releases (AppVeyor)

I will let you and @seanmcleod tell if the task can be ticked for Windows as well and close the issue accordingly.

bcoconni, 2018-06-17:
I think we have a pretty decent binary delivery for both Linux and Windows now.
I am closing this issue. If someone wants to bring additional requirements regarding this topic then I would suggest to open a new ticket.

closed 2018-06-17 by bcoconni