JSBSim-Team/jsbsim #19: Update the tests to Python 3

agodemar, 2018-05-02:


agodemar, 2018-05-02:
FYI
I have added a folder `Jupyter_Notebooks` in the reference manual repository [here](https://github.com/JSBSim-Team/jsbsim-reference-manual/tree/master/docs/_mypages/Jupyter_Notebooks).

The folder contains a couple of preliminary tests made via Jupyter notebooks (formerly IPython) running a Python 3 kernel. There are some helper classes defined in the subfolder `python` which are designed to perform multiple JSBSim simulations. The classes are adapted from [this work](http://www.mdpi.com/2226-4310/5/2/42/htm) that I did in collaboration with one of my former students and colleagues at Fraunhofer Institute for Wind Energy Systems, IWES (Germany).

I run my simulations in Windows and I've put a recent build of `JSBSim.exe` in the subfolder `JSBSim`.

seanmcleod, 2018-05-02:
So you didn't have any issues then with using Python 3 to call into the JSBSim code? Actually taking a quick glance at some of your Python code it looks like you're using os.system('JSBSim.exe --script...') to execute instances of JSBSim then and you're not using the Python API wrapper to call the JSBSim classes directly from Python?

agodemar, 2018-05-03:
> ... you're not using the Python API wrapper to call the JSBSim classes directly from Python?

In fact, that was my approach. But I would love to adat that work in order to use a Python 3 API wrapper. We might use it to make reports on engineering flight simulations via Jupyter.

seanmcleod, 2018-05-05:
@agodemar perfect, I was actually starting to write something very similar. My aim is to given an FDM generate a set of plots for a visual/graph representation, e.g. Cl vs AoA, Cd vs AoA, Cl vs Cd, plus graph the individual items that make up for example Cd, so Cd0, Cdi etc.

Did you notice the funny kink in the ground effect curve of your C172 versus the one retrieved from GitHub? I checked in a fix for the kink a while ago.

agodemar, 2018-05-05:
@seanmcleod Jupyter notebooks can help a lot in creating reports on FDM config files as well on results. The classes that I've put [here](https://github.com/JSBSim-Team/jsbsim-reference-manual/tree/master/docs/_mypages/Jupyter_Notebooks/python) have been designed with that in mind. They need to be enhanced with the task you were mentioning:

> given an FDM generate a set of plots for a visual/graph representation, e.g. Cl vs AoA, Cd vs AoA, Cl vs Cd, plus graph the individual items that make up for example Cd, so Cd0, Cdi etc.

seanmcleod, 2018-05-05:
@agodemar yep I've used Jupyter notebooks on other non-JSBSim related projects. 

In terms of plotting data from the FDM I was thinking of using the JSBSim python API in order to be able to evaluate functions that are defined rather than parsing the xml looking for table data etc.

So for example if Cdi is defined as a function using Cl^2 etc. the graphing code would evaluate it as well as handling functions with table data.

The Cdi based on Cl^2 is actually more complicated since the code would need to first lookup the Cl values from the Cl vs AoA function and then use those when evaluating the Cdi function etc.

bcoconni, 2018-05-08:
FYI @ezietsman made a pull request to update the tests to Python 3.6 and keep them compatible with Python 2.7. The PR is currently needing some changes before being merged but this is just a matter of a couple of details. I am confident they should be resolved quickly.

bcoconni, 2018-05-09:
@agodemar and @seanmcleod The PR #23 has been merged. I think we can close this issue.

closed 2018-05-10 by agodemar