JSBSim-Team/jsbsim #20: Build fails on macos

ezietsman, 2018-05-03:
Repos/jsbsim/build  master ✔                                                                                                                                                                                                                                               1d
    ▶ cmake ..
    -- The C compiler identification is GNU 7.3.0
    -- The CXX compiler identification is GNU 7.3.0
    -- Checking whether C compiler has -isysroot
    -- Checking whether C compiler has -isysroot - yes
    -- Checking whether C compiler supports OSX deployment target flag
    -- Checking whether C compiler supports OSX deployment target flag - yes
    -- Check for working C compiler: /usr/local/bin/gcc-7
    -- Check for working C compiler: /usr/local/bin/gcc-7 -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Checking whether CXX compiler has -isysroot
    -- Checking whether CXX compiler has -isysroot - yes
    -- Checking whether CXX compiler supports OSX deployment target flag
    -- Checking whether CXX compiler supports OSX deployment target flag - yes
    -- Check for working CXX compiler: /usr/local/bin/g++-7
    -- Check for working CXX compiler: /usr/local/bin/g++-7 -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- Found PythonInterp: /usr/local/bin/python (found version "2.7.14")
    -- Found Cython: /usr/local/bin/cython
    -- Found PythonLibs: /usr/lib/libpython2.7.dylib (found version "2.7.10")
    version: 3.3.10
    version: 3.3.10
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /Users/ewald/Repos/jsbsim/build

    ▶ make
    Scanning dependencies of target libJSBSim
    [  1%] Building CXX object src/CMakeFiles/libJSBSim.dir/FGFDMExec.cpp.o
    [  2%] Building CXX object src/CMakeFiles/libJSBSim.dir/FGJSBBase.cpp.o
    [  3%] Building CXX object src/CMakeFiles/libJSBSim.dir/initialization/FGInitialCondition.cpp.o
    [  4%] Building CXX object src/CMakeFiles/libJSBSim.dir/initialization/FGTrim.cpp.o
    [  5%] Building CXX object src/CMakeFiles/libJSBSim.dir/initialization/FGTrimAxis.cpp.o
    [  5%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/atmosphere/FGMSIS.cpp.o
    [  6%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/atmosphere/FGMSISData.cpp.o
    [  7%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/atmosphere/FGMars.cpp.o
    [  8%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/atmosphere/FGStandardAtmosphere.cpp.o
    [  9%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/atmosphere/FGWinds.cpp.o
    [  9%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGDeadBand.cpp.o
    [ 10%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGFCSComponent.cpp.o
    [ 11%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGFilter.cpp.o
    [ 12%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGGain.cpp.o
    [ 13%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGKinemat.cpp.o
    [ 13%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGSummer.cpp.o
    [ 14%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGSwitch.cpp.o
    [ 15%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGFCSFunction.cpp.o
    [ 16%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGSensor.cpp.o
    [ 17%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGPID.cpp.o
    [ 17%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGActuator.cpp.o                                                                                                                                                                                                                                        1d
    [ 18%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGAccelerometer.cpp.o
    [ 19%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGGyro.cpp.o
    [ 20%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGMagnetometer.cpp.o
    [ 21%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGAngles.cpp.o
    [ 21%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGWaypoint.cpp.o
    [ 22%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/flight_control/FGDistributor.cpp.o
    [ 23%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/propulsion/FGElectric.cpp.o
    [ 24%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/propulsion/FGEngine.cpp.o
    [ 25%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/propulsion/FGForce.cpp.o
    [ 25%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/propulsion/FGNozzle.cpp.o
    [ 26%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/propulsion/FGPiston.cpp.o
    [ 27%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/propulsion/FGPropeller.cpp.o
    [ 28%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/propulsion/FGRocket.cpp.o
    [ 29%] Building CXX object src/CMakeFiles/libJSBSim.dir/models/propulsion/FGTank.cpp.o
[... 67 lines cut ...]
    [ 81%] Built target JSBSim
    Scanning dependencies of target fpectl_module
    running build_ext
    running build_ext
    cc1plus: warning: command line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
    /Users/ewald/Repos/jsbsim/tests/fpectlmodule.cpp: In function 'PyObject* turnon_sigfpe(PyObject*, PyObject*)':
    /Users/ewald/Repos/jsbsim/tests/fpectlmodule.cpp:107:14: error: 'feenableexcept' was not declared in this scope
       fp_flags = feenableexcept(FE_DIVBYZERO | FE_INVALID | FE_OVERFLOW);
                  ^~~~~~~~~~~~~~
    /Users/ewald/Repos/jsbsim/tests/fpectlmodule.cpp:107:14: note: suggested alternative: 'feraiseexcept'
       fp_flags = feenableexcept(FE_DIVBYZERO | FE_INVALID | FE_OVERFLOW);
                  ^~~~~~~~~~~~~~
                  feraiseexcept
    /Users/ewald/Repos/jsbsim/tests/fpectlmodule.cpp: In function 'PyObject* turnoff_sigfpe(PyObject*, PyObject*)':
    /Users/ewald/Repos/jsbsim/tests/fpectlmodule.cpp:120:3: error: 'fedisableexcept' was not declared in this scope
       fedisableexcept(fp_flags);
       ^~~~~~~~~~~~~~~
    /Users/ewald/Repos/jsbsim/tests/fpectlmodule.cpp:120:3: note: suggested alternative: 'feraiseexcept'
       fedisableexcept(fp_flags);
       ^~~~~~~~~~~~~~~
       feraiseexcept
    error: command 'gcc-7' failed with exit status 1
    make[2]: *** [tests/CMakeFiles/fpectl_module] Error 1
    make[1]: *** [tests/CMakeFiles/fpectl_module.dir/all] Error 2
    make: *** [all] Error 2

agodemar, 2018-05-05:
Hi Ewald, thank you for bringing out this issue.

At the moment I do not have a MacOS available. Let's see if someone can shed some light on the compilation errors.

WillemEerland, 2018-05-05:
Right you are!

I don't have a fix -- I'm going to leave that to the coding guru's. I did find the following article, which may be useful to that particular guru:

https://philbull.wordpress.com/2012/07/27/portability-issues-with-floating-point-exception-handling/ 
https://philbull.wordpress.com/2012/12/09/update-floating-point-exception-handling-on-mac-os-x/

What I do have, is a workaround, which you can find here:

https://github.com/WillemEerland/jsbsim/tree/macos-fix

It comments out two lines, and switches around a declaration of Aeromatic++ (Why does the compiler on MacOS complain, and not on Linux?)

It compiles on MacOS, and completes almost all tests, except "TestInputSocket".

> 97% tests passed, 1 tests failed out of 34
> 
> Total Test time (real) = 158.06 sec
> 
> The following tests FAILED:
> 	 10 - TestInputSocket (Failed)
> Errors while running CTest

I believe it will only influence the Python library and only the input socket exception handling... so depending on what you're planning, this may work for you!

bcoconni, 2018-05-07:
@ezietsman and @WillemEerland the `fpectl` Python module is only used to make sure that none of the tests raise a floating point exception. JSBSim does not need it for a day to day usage. Even the tests can be run without `fpectl`, it is just that if a floating point exception is raised, it will be ignored and the test will continue.

I will investigate later how the `fpectl` code can be modified to run on MacOS X as well.

Regarding `TestInputSocket` it fails from time to time. This is most likely due to a race condition in it, you can run it again using the following command
```bash
> ctest -R TestInputSocket
```
If the failure was due to the race condition then it should succeed after a couple of try. If it does not, then you can get more information about the test failure with the `-V` (verbose) flag.
```bash
> ctest -R TestInputSocket -V
```

WillemEerland, 2018-05-07:
I tried it five times, failed five times. Here's (part of) the more detailed report

> 10: End of vehicle configuration loading.
> 10: -------------------------------------------------------------------------------
> 10: ok
> 10: 
> 10: ======================================================================
> 10: FAIL: test_no_input (__main__.TestInputSocket)
> 10: ----------------------------------------------------------------------
> 10: Traceback (most recent call last):
> 10:   File "TestInputSocket.py", line 156, in test_no_input
> 10:     TelnetInterface(fdm, 5., 1137)
> 10: AssertionError: error not raised
> 10: 
> 10: ----------------------------------------------------------------------
> 10: Ran 3 tests in 2.856s
> 10: 
> 10: FAILED (failures=1)
> 1/1 Test #10: TestInputSocket ..................***Failed    3.33 sec

ezietsman, 2018-05-08:
OK so it seems fpectl isn't in python 3, which for various reasons, I need to use at the moment. I'm trying to integrate jsbsim into a game engine, so all I really want to do is to pass in (throttle, rudder, roll, pitch, delta_time) inputs and want back (X, Y, Z, roll, pitch, yaw, Vx, Vy, Vz) for each time step.

I'm happy to hack at JSBSim to allow it to build python 3 bindings but I may need some guidance.

bcoconni, 2018-05-09:
@WillemEerland OK thanks for the report. It seems that `TestInputSocket` fails for a reason different from a race condition on your system. Could you open a separate issue to address this topic ?

And in the meantime could you please check that the following lines of Python code raise a `socket.error` exception on your platform as well ?
```python
import telnetlib
telnetlib.Telnet("localhost", 1137, 2.0)
```

bcoconni, 2018-05-09:
@ezietsman and @WillemEerland the PR #23 has been merged. I think this issue can now be closed ?

closed 2018-05-10 by ezietsman