JSBSim-Team/jsbsim #24: 'TestInputSocket' fails on MacOS

WillemEerland, 2018-05-10:
@bcoconni Opening a new issue as requested. More information following when I have access to MacOS.

> OK thanks for the report. It seems that TestInputSocket fails for a reason different from a race condition on your system. Could you open a separate issue to address this topic ?
> 
> And in the meantime could you please check that the following lines of Python code raise a socket.error exception on your platform as well ?
> 
> import telnetlib
> telnetlib.Telnet("localhost", 1137, 2.0)

bcoconni, 2018-05-10:
OK thanks. Please make sure to test the last code from the branch `master` since a potential fix has been introduced for this issue since the merge of PR #23 (using port 2222 instead of 1137)

WillemEerland, 2018-05-10:
Test still fails

> Start 10: TestInputSocket
> 10/35 Test #10: TestInputSocket ..................***Exception: SegFault  5.63 sec

`telnetlib.Telnet("localhost",2222,2.0)`

> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/Users/will/anaconda/lib/python3.5/telnetlib.py", line 218, in __init__
>     self.open(host, port, timeout)
>   File "/Users/will/anaconda/lib/python3.5/telnetlib.py", line 234, in open
>     self.sock = socket.create_connection((host, port), timeout)
>   File "/Users/will/anaconda/lib/python3.5/socket.py", line 711, in create_connection
>     raise err
>   File "/Users/will/anaconda/lib/python3.5/socket.py", line 702, in create_connection
>     sock.connect(sa)
> ConnectionRefusedError: [Errno 61] Connection refused

FYI, they removed telnet from MacOS (apparently, it took up too much space) -- Could this be the issue?

WillemEerland, 2018-05-10:
Test passes - my bad!

> Start 10: TestInputSocket
> 1/1 Test #10: TestInputSocket ..................   Passed    3.34 sec
> 
> 100% tests passed, 0 tests failed out of 1

closed 2018-05-10 by WillemEerland

ezietsman, 2018-05-11:
you need to install it yes, `brew install telnet`