JSBSim-Team/jsbsim #27: Clean up code base

bcoconni, 2018-05-12:
There is a number of outdated files or redundant data currently in the code base.
### Files to delete
- `.cvsignore` since we are no longer using CVS
- `JSBSim.cbx` since [Borland C++ builderX is abandoned](https://en.wikipedia.org/wiki/C%2B%2BBuilder)
- `JSBSim.vcproj` since we are no longer working Microsoft Visual Studio earlier than 2015
### Redundant files
- `INSTALL` which contains information redundant with `README.md`

Unless someone objects I will remove them from the code base

agodemar, 2018-05-14:
I vote for the deletion of those outdated files.

bcoconni, 2018-05-17:
Done in commit 6bb1f49

closed 2018-05-17 by bcoconni

bcoconni, 2018-06-03:
More clean up : removed the directory `include/mipspro` (commit 5ec55ca) and the mentions of JSBSim being supported on SGI platforms.

According to [Wikipedia](https://en.wikipedia.org/wiki/IRIX), SGI went bankrupt in 2009 and released the last version of IRIX (6.5.30) in 2006. The files included in `include/mipspro` never were used by CMake anyway and nobody complained.

bcoconni, 2018-06-03:
Also removed the empty files `NEWS` and `ChangeLog` (commit 4644450) which were there to prevent the GNU/autotools from complaining. Since we dropped the GNU/autools in favor of CMake, I removed these 2 files.