JSBSim-Team/jsbsim #35: Python tests all fail on Windows

seanmcleod, 2018-05-25:
With the recent checkins to support Python 3 I'm now able to build the Python module and also run the Python tests using VS2017 and it's CMake support.

However all the Python tests currently fail due to an issue with the cleanup mechanism used in the Python tests. 

Basically the Sandbox class executes shutil.rmtree() to cleanup the temporary sandbox directory. However this fails since there are still files open, e.g. output.csv so it fails on Windows with a message that it can't delete the file since a process has the file open.

If I tell shutil.rmtree() to ignore errors then the test passes.

These 2 lines should also probably be swapped around in JSBSimTestCase.tearDown() otherwise it may fail on Windows even if there are no open files inside the sandbox directory.

self.sandbox.erase()
os.chdir(self.currentdir)

So we'll need to come up with a reliable mechanism on Windows to cleanup the temporary sandbox directory.

bcoconni, 2018-05-25:
And what if we delete the fdm prior to calling `JSBSimTestCase.tearDown()` ? Would that force all the files to close ?

And yes it makes sense to swap these 2 lines
```python
self.sandbox.erase()
os.chdir(self.currentdir)
```

seanmcleod, 2018-05-27:
I've just submitted a pull request - https://github.com/JSBSim-Team/jsbsim/pull/37 which improves the test results in terms of going from 0/35 passing to 32/35 passing now. The largest improvement came from changing the order in `JSBSimTestCase.tearDown()`.

I had to in a handful of cases also explicitly delete the fdm before the teardown started, and in a handful of cases change the order in terms of an explicit call to delete .csv files happening before the deletion of the fdm.

bcoconni, 2018-06-17:
@seanmcleod 
I think this has been fixed ? If so, could we close this issue ?

seanmcleod, 2018-06-17:
At the time I submitted pull request - https://github.com/JSBSim-Team/jsbsim/pull/37 the pass rate for the Python tests improved from 0/35 to 32/35, i.e. there were still 3 tests failing on Windows.

```
CheckFGBug1503
TestInitialConditions
TestSuspend
```

In the meantime I've been busy/stuck in the atmosphere code so I haven't spent any time trying to figure out the issues and fix them for the 3 tests that were still failing.

I've just run the 3 now to confirm that they still fail on Windows. `TestInitialConditions` now succeeds but the other 2 still fail.

seanmcleod, 2018-06-17:
@bcoconni in terms of `TestSuspend` it's failing when performing `np.abs()` on the 2 Panda data frames.

```python
out = pd.read_csv('BallOut.csv', index_col=0)
# Find all the data that are differing by more than 1E-8 between the
# two data sets.
diff = FindDifferences(self.ref, out, 1E-8)

def FindDifferences(ref, other, tol):
    delta = np.abs(ref - other)

-- Exception TypeError: unsupported operand type(s) for -: 'str' and 'str'
```

So it appears that some of the data in the Panda dataframes are strings rather than floats? Although glancing at the .csv file there isn't anything obvious, maybe some of the integer values in the .csv are being read in as strings? Or is the header row possibly being included?

bcoconni, 2018-06-17:
According to [this StackOverflow article](https://stackoverflow.com/questions/43970972/typeerror-unsupported-operand-types-for-str-and-str-in-python-3-x-anac), yes it is possibly that headers are incorrectly handled.

They are suggesting to add `header=0` to the `read_csv` command which would be
```python
out = pd.read_csv('BallOut.csv', index_col=0, header=0)
```
I guess, the same trick should be applied to the data contained in `self.ref` ?

seanmcleod, 2018-06-17:
Had a quick look in the debugger and it looks like the value of the data is of type `object` as opposed to `float` and it looks like there is a mix of floats and strings.

Will take a closer look at the .csv files tomorrow to see if there is any obvious reason for Pandas to be reading some values in as strings.

```
array([0.0, -0.001048238798741804, 0.0, '3.763875093686821e-08',
       '-8.803574563915573e-08', '-2.046347373786989e-07',
       0.002451794545730645, -0.003733853083297775, 0.0020572998990450514,
       '219.4452737', 2483629.885, 500.0, 1825.4844798000001,
       500.0000000000007, -1.13686837721616e-13, -5.684341886080802e-14,
       '-20.60563849655216', '-0.03590665693927631', '24.5008232850669',
       '-1.969722398725504e-15', '-32.16800795293633', '0', '0', '0', '0',
       499.99999999999983, 1.845346400166839e-14, 3.378685916172798e-13,
       -1772.8533442988698, 321.3938048432702, 293.412044416733,
       -246.20193825305242, 321.3938048432703, 293.4120444167331,
       293.4120444167331, 246.20193825305242, -321.3938048432703, '-0',
       '0', '-0', 0, '0', '0', '-0', 0, 0, 0, 0, 0, 0, '0', '0', '0', 0, 0,
       0, '-32.13334594667513', '1.067000358096352e-14',
       '38.29503047402267', '0', '0', '0', '0', '0', '0', 0, 0, 0, 0, 0, 0,
       0, 0, 0, '0', '0', '0', 0, 0, 0, '0', '0', '0', 20, 0, 0, 0, 10, 0,
       0, 0, 10, 1.5540474869999998, 50, 0, 0, 0, 10000.0, 10000.0,
       '1.86836366938607e-14', '40.00000000000001', '40.00000000000002',
       '0.8830222215594891', '-0.1169777784405109', '0.3213938048432697',
       0.3213938048432697, '0.5', '0.06162841671621939', '-0.5',
       '0.7044160264027588', '0.5', '0.06162841671621938', '-0.5',
       '0.7044160264027588', 3.8716888659399e-14, 2.1146112093844e-15, 0.0,
       0.0, 90.0, 1.281938613027782e-09, 20935646.32546, 0.0,
       1.281938613027782e-09, 20935646.32546, 0.0, 0.0, 10000.0, 0, 0.0,
       90.0, 20935646.32546], dtype=object)
```

seanmcleod, 2018-06-18:
So taking a look at the .csv file and doing some tests it looks like the `-nan(ind)` entries are causing an issue at least with my Pandas version (`pd.__version__ = 0.18.1`) running on Python 3.

The .csv file has a combination of floats, `nan` and `-nan(ind)`.

```python
# BallOut.csv with '-nan(ind)' entries
df = pd.read_csv('BallOut.csv', index_col=0)
df - df
TypeError: unsupported operand type(s) for -: 'str' and 'str'
```

If I replace all the `-nan(ind)` entries with `nan` then the above snippet succeeds.

Tried forcing it to assume all data was `np.float64` but that also fails, and suceeds for the .csv file where I've replaced `-nan(ind)` with `nan`.

```python
df = pd.read_csv('BallOut.csv', index_col=0, dtype=np.float64)
TypeError: Cannot cast array from dtype('O') to dtype('float64') according to the rule 'safe'
ValueError: could not convert string to float: '-nan(ind)'
```

Adding `-nan(ind)` as a `na_values` works.

```python
df = pd.read_csv('BallOut.csv', index_col=0, na_values="-nan(ind)")
```

If I add `na_values="-nan(ind)"` to all the calls of `pd.read_csv()` in the test then the test succeeds on Windows.

So on Linux/gcc do you end up with `-nan(ind)` entries in addition to `nan`. If so what version of Pandas do you have installed?

seanmcleod, 2018-06-18:
Looking at the documentation for `pandas.read_csv()`.

```
na_values : scalar, str, list-like, or dict, default None

Additional strings to recognize as NA/NaN. If dict passed, specific per-column NA values. By default the following values are interpreted as NaN: ‘’, ‘#N/A’, ‘#N/A N/A’, ‘#NA’, ‘-1.#IND’, ‘-1.#QNAN’, ‘-NaN’, ‘-nan’, ‘1.#IND’, ‘1.#QNAN’, ‘N/A’, ‘NA’, ‘NULL’, ‘NaN’, ‘n/a’, ‘nan’, ‘null’.
```

So no `-nan(ind)` in the list, what do you see on Linux/gcc? Maybe `-1.#IND` instead of `-nan(ind)`?

bcoconni, 2018-06-18:
@seanmcleod 
> So taking a look at the .csv file and doing some tests it looks like the `-nan(ind)` entries are causing an issue at least with my Pandas version `(pd.__version__ = 0.18.1)` running on Python 3.

On my side `pd.__version__` returns 0.22.0 on Python 2 and 0.23.1 on Python 3.

> The .csv file has a combination of floats, `nan` and `-nan(ind)`.

I don't have any `nan` in `BallOut.csv` (I searched the file with `grep -i nan BallOut.csv` and nothing was returned. And if there were `nan(ind)` `grep` would have found them). May be the problem is coming from that in the first place ?

> Adding `-nan(ind)` as a `na_values` works.
```python
df = pd.read_csv('BallOut.csv', index_col=0, na_values="-nan(ind)")
```
It does not make any difference on my side, obviously because there are no `nan`. I guess the actual issue is that there are `nan` in `BallOut.csv` which suggests that the test is failing for a good reason...

Could you be more specific about the scripts and conditions that are run to generate the CSV file with NaNs ?

seanmcleod, 2018-06-19:
I set breakpoints on all the calls to `pd.read_csv()` and then looked at the BallOut.csv generated when I hit the first breakpoint that was hit, which was in the `setup()` method.

At this point BallOut.csv is 124,170 bytes in size and has a large number of `nan` and `-nan(ind)` entries.

seanmcleod, 2018-06-20:
If I run the `cannonball.xml` script, which is what `TestSuspend.py` runs, directly I don't get any `nan` or `-nan(ind)` values.

The difference is that `TestSuspend.py` sets the `integrators`. If you don't see any `nan` values on Linux are we seeing some difference between the Windows build and the Linux build due to differences in the integrators when set to 5?

```python
        script_path = self.sandbox.path_to_jsbsim_file('scripts',
                                                       'cannonball.xml')
        fdm.load_script(script_path)
        fdm['simulation/integrator/rate/rotational'] = 5
        fdm['simulation/integrator/rate/translational'] = 5
        fdm['simulation/integrator/position/rotational'] = 5
        fdm['simulation/integrator/position/translational'] = 5
        fdm.run_ic()
        fdm.set_dt(0.05)
```

bcoconni, 2018-06-21:
Indeed, on my side `cannonball.xml` raises a floating point exception when I try to run the script from the command line
```bash
> JSBSim scripts/cannonball.xml
```
It seems that the ball falls through the Earth to finally escape from the its gravitational attraction. The script fails the ball reaches an altitude of 36281033 ft ! It crashes because the altitude is so high that the air density is reaching 0.

The ball hits the ground before t=37s which is the moment at which `nan` are issued on your side.

May be if you make sure that the script stops before 36s then `TestInitialConditions.py` will succeed ? For that purpose, you could modify the `end` parameter in `cannonball.ml` and set it to 37:
```xml
<run start="0.0" end="36" dt="0.00833333"/>
```
However I don't understand why you don't get a floating point exception as well ? The code in [`JSBSim.cpp` lines 283-291](https://github.com/JSBSim-Team/jsbsim/blob/93f89276968a223aa0bc8e3e34fc9f32f70f1fd3/src/JSBSim.cpp#L283) is supposed to enable floating point exceptions on Windows as well ?
```c++
int main(int argc, char* argv[])
{
#if defined(_MSC_VER)
  _clearfp();
  _controlfp(_controlfp(0, 0) & ~(_EM_INVALID | _EM_ZERODIVIDE | _EM_OVERFLOW),
           _MCW_EM);
#elif defined(__GNUC__) && !defined(sgi) && !defined(__APPLE__)
  feenableexcept(FE_DIVBYZERO | FE_INVALID);
#endif
```

bcoconni, 2018-06-22:
@seanmcleod 
> You mentioned the code in `JSBSim.cpp` in `main()` enabling floating point exceptions, but in the case of the Python test case where we are seeing differences in the output between Windows and Linux (VS and gcc) this code in `main()` probably isn't being executed right?

That's correct. The code in `JSBSim.cpp` is only used for the command line executable `JSBSim[.exe]`. As far as tests are concerned, this code is replaced by the Python code.

> Isn't the floating point control for the Python module case handled by the fpectl module?

It is, but `fpectl` is not imported in `TestInitialConditions.py` so floating point exceptions cannot be reported.

> And maybe there is some VS/gcc difference or missing #ifdef there which is resulting in different floating point exception handling?

That's not supposed to be. Anyway, the tests we conducted with the `cannonball.xml` script clearly show there is a problem with this script.

Have you tried to shorten the simulated duration of `cannonball.xml` to 36 seconds as I suggested earlier ? Does that fix the problem for `TestInitialConditions.py` ?

seanmcleod, 2018-06-22:
@bcoconni yes if the end time in `cannonball.xml` is changed to 36s down from it's current 800s then the test `TestSuspend.py` succeeds without having to deal with `-nan(ind)` values.

But I'm still trying to understand why we see a difference between the Linux and Windows builds when we run `TestSuspend.py`. It would appear in the Windows case the floating point exceptions are disabled and we get `nan` and '-nan(ind)` values.

You said in the Linux case of running `TestSuspend.py` you don't see any `nan` or `-nan(ind)` in `BallOut.csv`. Now if a floating point exception is being raised in the Linux case while running `TestSuspend.py` wouldn't the process die and the test fail?

bcoconni, 2018-06-23:
@seanmcleod 
> yes if the end time in `cannonball.xml` is changed to 36s down from it's current 800s then the test `TestSuspend.py` succeeds without having to deal with `-nan(ind)` values.

Good. So we have isolated the cause of the test failure.

> You said in the Linux case of running `TestSuspend.py` you don't see any `nan` or `-nan(ind)` in `BallOut.csv`. Now if a floating point exception is being raised in the Linux case while running `TestSuspend.py` wouldn't the process die and the test fail?

Yes, when FPEs (floating point exception) are enabled, the process dies and the test fails. In that case, no `nan` are exported to `BallOut.csv`. I have attached the `.csv` files I am obtaining for several settings:

[BallOut.zip](https://github.com/JSBSim-Team/jsbsim/files/2129796/BallOut.zip)
1. `BallOut1.csv` is the output of `cannonball.xml` when the integrators are set to their default values and the time step is 120 Hz / 0.0083333 s
2. `BallOut2.csv` is the output of `cannonball.xml` when the integrators are set to "5" and the time step is 20 Hz / 0.05 s

For both these files, no `nan` are exported. The execution is terminated shortly after the ball hits the ground (`CONTACT`). The integrator and time step settings does not make much difference before the contact (the content of `BallOut1.csv` and `BallOut2.csv` is very similar).

When I disable FPEs, the process runs for 800 simulated seconds and terminates normally without any error message (which what happens on Windows if I understood correctly). The tests have been run in the same conditions `1` and `2` than above, the file names have been appended with `_noFPE`.
* `BallOut1_noFPE.csv` contains `nan` for t >= 582 s
* `BallOut2_noFPE.csv` contains `nan` for t >= 39 s

So it seems that the test is issuing the same garbage on Linux than on Windows if it is not interrupted as soon as a floating point error is occurring. The only difference that I can see is that Linux consistently reports `nan` and never outputs `nan(ind)`.

seanmcleod, 2018-06-23:
@bcoconni should I modify `TestSuspend.py` to limit the run time to 39s?

bcoconni, 2018-06-25:
@seanmcleod 
> should I modify `TestSuspend.py` to limit the run time to 39s?

That's my opinion, but I have this strange feeling that you are not convinced this is the way to go :worried:

seanmcleod, 2018-06-27:
Happy with 'this way' :wink: - just submitted a pull request for this - https://github.com/JSBSim-Team/jsbsim/pull/79

So just 1 last Python test failing on Windows which I'll look into next. Then we'll be able to enable the Python tests on AppVeyor to match Linux.

seanmcleod, 2018-07-09:
@bcoconni I'm looking into the final test - `CheckFG1503` that fails on Windows.

It currently throws an exception when `Process.start()` is called:

```python
p = Process(target=self.ScriptExecution, args=(fdm,))
p.start()
```

The exception thrown is:
```
TypeError: cannot serialize '_io.TextIOWrapper' object
```

Doing a search the couple of hits I got mentioned this being an issue when the object can't be serialized, e.g. "Seems like Martin is right, the error seems to be that your object isn't
serializable, and it indeed seems to be 'pickle' that's used for
serialization (though I'm not 100% sure). The _io.TextIOWrapper you're
trying to queue is probably a 'foreign' object, implemented in C or C++ or
sth, and can therefore not be serialized."

So my assumption was the issue was with the fdm argument, thinking that since it's a wrapper around our JSBSim C++ objects etc. that it was failing to be serialized. Ignoring the question of why this doesn't seem to be an issue on Linux for the moment.

However changing the argument to a simple string instead of passing `fdm` resulted in the same exception. So it appears the serialization issue is with the `target`, and sure enough the following avoids the exception.

```python
def dummy(fdm):
    print('hello')

class CheckFGBug1503(JSBSimTestCase):
   def test_regression_bug_1503(self):
      ...
      p = Process(target=dummy, args=(fdm,))
      p.start()
```

Now before I dig too much further I wanted to make sure that this isn't an issue on Linux. The test passes on Travis etc. but I just want to make sure it isn't passing somehow with this sort of exception being thrown but somehow being 'handled' such that it doesn't fail the test?

What I did also notice after making my `dummy` changes is that things didn't appear to work as expected when the sub-process is started. Instead of my `dummy` function being called in the new sub/child process the whole test `CheckFGBug1503` seems to be executed again. In other words the new process is simply an exact repeat of the parent process.

seanmcleod, 2018-07-09:
I had glanced at the documentation for the `multiprocessing` module but after a bit more RTFM I came across this which specifically talks about Windows specific differences/requirements and seems to answer most of the questions I asked above.

[Multiprocessing - Windows](https://docs.python.org/2/library/multiprocessing.html#windows)

In particular the extra requirements around `pickability` and `safe importing of main module`.

```python
if __name__ == '__main__':
    info('main line')
    p = Process(target=f, args=('bob',))
    p.start()
    p.join()
```

"For an explanation of why (on Windows) the if __name__ == '__main__' part is necessary, see Programming guidelines."

So it looks like it will take a bit more work to get this part of the test working on Windows.

seanmcleod, 2018-07-09:
So as a quick hack test the following works on Windows. First off add a `__main__` check:

```python
if __name__ == '__main__':
    RunTest(CheckFGBug1503)
```

Then based on further testing in terms of which objects do serialize/pickle successfully.

```python
def SubProcessScriptExecution(sandbox, script_path, time_limit=1E+9):
    # Run the script with the modified aircraft
    fdm = CreateFDM(sandbox)
    fdm.set_aircraft_path('aircraft')
    fdm.load_script(script_path)
    fdm.run_ic()

    while fdm.run() and fdm.get_sim_time() < time_limit:
        aileron_pos = fdm['fcs/left-aileron-pos-rad']
        #self.assertEqual(aileron_pos, 0.0,
        #                msg="Failed running the script %s at time step %f\nProperty fcs/left-aileron-pos-rad is non-zero (%f)" % (script_path, fdm.get_sim_time(), aileron_pos))
```

So now pass the `sandbox` object and the `script_path` to the sub-process.

```python
def test_regression_bug_1503(self):
   …..
   p = Process(target=SubProcessScriptExecution, args=(self.sandbox, self.script_path,))
   p.start()
   …….
```

bcoconni, 2018-07-11:
@seanmcleod 
I am not sure to understand what the test would look like after the changes you proposed ?

seanmcleod, 2018-07-12:
@bcoconni I've added a commit to my fork of JSBSim to show the changes more clearly for my quick hack/test to get the test to run successfully on Windows. Take a look at - https://github.com/seanmcleod/jsbsim/commit/b97c5fe098357a80d3d19850f258b21bc7103cdf

bcoconni, 2018-07-14:
> Take a look at - seanmcleod@b97c5fe

Thanks. Now that I am re-reading my test 4 years later and I have to say that I made it quite convoluted :smiley:

At the time I wrote this test the problem was that JSBSim was entering an infinite loop when the code in `CheckFGBug1503.ScriptExecution` was run. My convoluted code was therefore designed to run `ScriptExecution`, wait for some time and if the test was not completed after that time, declared the test failed.

I understand `multiprocessing.Process` is quite complicated to use under Windows so may be you should use another means to wait for some time and if the test is not completed by that time, declare the test failed. I found [this on StackOverflow](https://stackoverflow.com/questions/34743448/how-to-specify-test-timeout-for-python-unittest) which should hopefully allow you to drop the `multiprocessing` stuff. I like their concept of the class `test_timeout` because you can raise any exception in `test_timeout.handle_timeout`.

Anyway, anything that simulate the time out concept would be fine for that particular test as far as I am concerned.

seanmcleod, 2018-07-14:
@bcoconni the problem with `test_timeout` is the use of `signal.alarm()` which limits it to Unix.

"The basic idea is to use signal handlers to set an alarm for some time interval and raise an exception once that timer expires.
Note that this will only work on UNIX.
The code is still using signal and thus UNIX only:"

The big difference between the Windows and Unix multiprocessing Python implementations seems to be due to the use of `fork` on the Unix versions which allows you to use pretty much any pre-existing objects whereas in the Windows case a new 'blank' process is created and so the arguments have to be serialized across to this new blank instance etc.

The modifications I made for my quick test/hack should also work for the Unix case so we could use something like that for both, or some runtime check to see whether we're executing on Windows and only use my modified code on Windows etc.

bcoconni, 2018-07-14:
> the problem with `test_timeout` is the use of `signal.alarm()` which limits it to Unix.

OK so what about the other option (`timeout-decorator`) then ?

What I is annoying me in your current proposal is that you commented out the assertion in the code.

seanmcleod, 2018-07-14:
Hmm, taking a look at `timeout-decorator` - https://pypi.org/project/timeout-decorator/ it looks like it's default is to use `signals` but it does offer an option `use_signals=False` which then falls back to using multiprocessing with warnings about serialization/pickling which is what the original issue was :wink:

It's also a separate Python module that we would need to install.

In terms of commenting out the assertion that was done just in order to quickly test whether I could get the basics to work in Windows. There is no reason it has to remain commented out.

It would simply need a bit more additional code for example to either create a new class derived from JSBSimTestCase in order to inherit the `assertEqual` method and to create an object instance in the `SubProcessScriptExecution` function or to duplicate the `assertEqual` code.

bcoconni, 2018-07-15:
@seanmcleod 
I have tested your code and it succeeds on my platform (Linux Fedora).
In addition I validated that the test fails as expected when the value in the assertion of `ScriptExecution` is modified from `0.0` to `1.0`.

So it's all good !

seanmcleod, 2018-07-15:
Thanks for testing on Linux. I've submitted a merge request - https://github.com/JSBSim-Team/jsbsim/pull/97

Once it has been merged then I'll enable the Python tests in AppVeyor so that we get them run in a Windows build for each commit.

seanmcleod, 2018-07-16:
All the Python tests are now working on AppVeyor.

closed 2018-07-16 by seanmcleod