JSBSim-Team/jsbsim #39: JSBSim failed to open the configuration file

Gor-Ren, 2018-05-30:
Hello and thanks for the great project. Unfortunately I am falling at the first hurdle: I cannot run any simulations.

```
$ JSBSim --aircraft=ball --initfile==reset00

JSBSim Flight Dynamics Model v1.0 May 30 2018 09:53:30
            [JSBSim-ML v2.0]

JSBSim startup beginning ...

Could not open file: Path "aircraft/ball/ball.xml"
  JSBSim failed to open the configuration file: Path "aircraft/ball/ball.xml"
  JSBSim could not be started
```

I have tried sudo privileges and specifying the root dir:
 ```
$ JSBSim --aircraft=ball --initfile==reset00 --root=~/Apps/jsbsim-code/
...
JSBSim failed to open the configuration file: Path "~/Apps/jsbsim-code/aircraft/ball/ball.xml"
...
```
This despite the fact that the files are clearly there in the specified root:

```
$ ls ~/Apps/jsbsim-code/aircraft/ball/
ball.xml  INSTALL  reset00_v2.xml  reset00.xml  reset01_v2.xml  reset01.xml
```
I am running a fresh install of Ubuntu 18.04. JSBSim was built and installed as per the repo instructions from latest commit https://github.com/JSBSim-Team/jsbsim/commit/b4dd258a9c2dc9c70ca35ecf577788e528630702, with the `DINSTALL_PYTHON_MODULE=ON` flag. All tests run successfully (which means the simulations run perfectly happy from Python, so I am at a bit of a loss). Many thanks in advance for any help!

bcoconni, 2018-05-30:
@Gor-Ren `JSBSim` does not expand `~` in paths (although shells usually replace `~` by the content of the environment variable `$HOME`). So you have to specify the root dir either by a relative path
```bash
$ JSBSim --root=../../relative/path/to/JSBSim/root/ --initfilreset00 --aircraft=ball
```
or by an absolute path
```bash
$ JSBSim --root=/home/username/Apps/jsbsim-code --initfilreset00 --aircraft=ball
```

Gor-Ren, 2018-05-30:
@bcoconni aha, thank you! It is now working with the absolute path.

closed 2018-05-30 by bcoconni