JSBSim-Team/jsbsim #46: :sparkles: **JSBSim Jupyter Notebooks** :sparkles:

agodemar, 2018-06-05:
Establish a set of Jupyter notebooks to show how JSBSim can be used for engineering simulations. The notebooks will be based on JSBSim interface for Python 3, i.e. the same that we use for unit testing.
Here's a list of fundamental tasks:
- [ ] Establish a folder to collect all notebooks. Subfolders are encouraged in order to explore with ease both input and output files.
- [ ] Draft a list of examples covering as many use cases as possible.
- [ ] Enjoy the results of each single example!

seanmcleod, 2018-06-05:
@agodemar if you think either of the following are useful enough/good enough I could re-do them as Jupyter examples. I originally wrote them up based on running JSBSim scripts and using matplotlib etc. to graph the .csv file output etc.

[Landing – Ground effect, flare – JSBSim](https://mcleodsean.wordpress.com/2018/02/06/landing-ground-effect-flare-jsbsim/)

[Pitch Rate Damping – JSBSim Test](https://mcleodsean.wordpress.com/2018/02/02/pitch-rate-damping-jsbsim-test/)

agodemar, 2018-06-05:
@seanmcleod Sure. Meanwhile I'll make a folder `doc` in the main repository where we're going to put all our notebooks.

bcoconni, 2018-06-05:
Just for fun. Some of the things that can be achieved with JSBSim/Python/Jupyter lab.

On this YouTube video, the bottom curve displays the aircraft altitude as the simulation progresses (the aircraft takes off just before t=25s). I did not have the time to find a means to remove the flickering but the proof of concept is there.

[image]

agodemar, 2018-06-05:
@bcoconni Cool stuff! By the way, I know that `bokeh` is the plotting package of choice for this kind of dynamic things. See [this video](https://youtu.be/aXNy1r5UDEE)