JSBSim-Team/jsbsim #49: [Jupyter Notebooks] make Python classes in `jsbsim/tests visible` in Jupyter notebooks launched from `jsbsim/doc/basic`

agodemar, 2018-06-05:
The Jupyter notebooks that we want to use to narrate workflows with JSBSim are located in `jsbsim/doc` subfolder, e.g. in `jsbsim/doc/basic` we have [01_JSBSim_Setup.ipynb](https://github.com/JSBSim-Team/jsbsim/blob/master/doc/basic/01_JSBSim_Setup.ipynb).

Jupyter is launched from command line as follows:

```bash
$ jupyter notebook
```

Then, through the browser, the user navigates the `.ipynb` notebook files and opens one of her choice. When the notebook is run the Python kernel should be able to find all the classes located in `jsbsim/tests`, e.g. `JSBSim_utils.py`

seanmcleod, 2018-06-05:
@agodemar on Linux at least from what I've seen in terms of @bcoconni's commits he has things setup to allow a user to install the JSBSim python module etc. (although I'm not sure if this includes JSBSim_utils.py). So I would assume that once the user does the python install of the JSBSim module on Linux it would be available from any Jupyter notebook.

We'll need to look into the equivalent Python module install option on Windows.

For now as a very brief initial test on Windows with the JSBSim python module `jsbsim.cp35-win_amd64.pyd` with Jupyter I've done the following:

Built all using CMake with VS2017 and then changed into the `tests` directory where the python module/dll `jsbsim.cp35-win_amd64.pyd` ends up.

```
cd \Users\Sean\CMakeBuilds\3f00c6d9-d323-5a32-8a90-665138817fd4\build\x64-Release\tests
jupyter notebook
```
Then created a new Jupyter notebook and sucessfull executed the following code in the notebook:

```python
import jsbsim
fdm = jsbsim.FGFDMExec(root_dir='\\source\\jsbsim')
fdm['atmosphere/rho-slugs_ft3']

0.0023768988323037404
```

seanmcleod, 2018-06-05:
@bcoconni is there a way for a Python script to retrieve the output from an `<output>` element without having to read it in from a temporary .csv file or setup an output socket?

So for example I'm thinking of something along these lines from a Python script, inside and outside of a Jupyter notebook.

```python
fdm.run('some script')
data_output = fdm.retreive_data_output()
# Use data_output, e.g. matplotlib the data etc.
```

agodemar, 2018-06-05:
@seanmcleod I'm working on Ubuntu (WLS on Windows 10). I've built all for Linux in jsbsim/build using CMake. What do you mean by "install the JSBSim python module"?

I feel a little dummy in these pythonic things, which I still do not handle easily.

seanmcleod, 2018-06-05:
@agodemar there are multiple ways to install Python modules in general, e.g. `pip install` etc.

But take a look at .travis.yml and note the `-DINSTALL_PYTHON_MODULE=ON`

```
  - pip install cython numpy pandas scipy
  - gem install gist
  - mkdir build
  - cd build
  - cmake -DCMAKE_INSTALL_PREFIX=~/ -DINSTALL_PYTHON_MODULE=ON -DCPACK_GENERATOR=DEB -DBUILD_SHARED_LIBS=$USE_SYSTEM_EXPAT -DSYSTEM_EXPAT=$USE_SYSTEM_EXPAT ..
  - make -j2
```

And take a look at `jsbsim\python\CMakeLists.txt`

```
# Install the JSBSim Python module
if (INSTALL_PYTHON_MODULE)
  execute_process(COMMAND ${PYTHON_EXECUTABLE} ${CMAKE_CURRENT_SOURCE_DIR}/findInstallDir.py OUTPUT_VARIABLE PYTHON_INSTALL_DIR)
  execute_process(COMMAND ${PYTHON_EXECUTABLE} ${CMAKE_CURRENT_SOURCE_DIR}/findModuleFileName.py OUTPUT_VARIABLE PYTHON_MODULE_NAME)
  install(FILES ${CMAKE_BINARY_DIR}/tests/${PYTHON_MODULE_NAME} DESTINATION ${PYTHON_INSTALL_DIR} COMPONENT pymodules)
endif()
```

So in theory if you execute CMake with the arguments you see in `.travis.yml` you should end up with the JSBSim python module installed.

agodemar, 2018-06-05:
@seanmcleod thanks for the instructions. Now the first Jupyter *very basic* test works.

bcoconni, 2018-06-05:
> Still, `JSBSim_utils` is not found. @bcoconni Should it be?

I did not plan to release `JSBSim_utils.py` since it mostly contains stuff for the test suite (inheriting a test case, moving around files from source to build directory, etc.). So no, it is not installed by the installer.

If there are a couple of lines of general interest in there, I would rather move them in a separate file that would be located in the `python` directory for instance and make `JSBSim_utils.py` import that file.

agodemar, 2018-06-05:
@bcoconni @seanmcleod I've created a file `jsbsim/doc/python/JSBSim_nb_utils.py` which can be called in the first cell of a Jupyter notebook. It defines utility functions and classes. Initially, they are copied from `jsbsim/tests/JSBSim_utils.py`. In the future it will be cleaned up.

agodemar, 2018-06-06:
@seanmcleod I've closed this issue because it is resolved now. Regarding your question
> @bcoconni is there a way for a Python script to retrieve the output from an <output> element without having to read it in from a temporary .csv file or setup an output socket?

> So for example I'm thinking of something along these lines from a Python script, inside and outside of a Jupyter notebook.

> fdm.run('some script')
> data_output = fdm.retreive_data_output()
> ... Use data_output, e.g. matplotlib the data etc.

please open a separate issue.

closed 2018-06-06 by agodemar