JSBSim-Team/jsbsim #58: Keep in the Releases area only the releases associated to the last commit

agodemar, 2018-06-08:
I have manually deleted a bunch of older releases. At the moment we have only the release tagged `1.0.0-152`. We should decide our release policy and automate it via Travis and AppVeyor.

Tentative proposal:
1. Make a new release every time we have a new commit. Delete the previous one (or keep only the current and the previous one).
2. Make a new release every time a given *flag* is set (in the commit text or elsewhere). Delete the previous one (or keep only the current and the previous one).
3. Make a new release every 6 months. Delete the previous one (or keep only the current and the previous one).

Ideas?

seanmcleod, 2018-06-08:
I would go with option 1, i.e. every commit that passes build and test on both Travis and AppVeyor, i.e. the latest passing commit.

Then periodically potentially we could mark specific commits as say version 1.0, 1.1 etc. and have those as releases in addition to the latest passing commit.

bcoconni, 2018-06-10:
@agodemar 
I guess option 2 is not very difficult to implement but why not keeping it simple and go for option 1 ?

IMHO option 3 is definitely not the way to go since experience showed that this policy results in no release at all :cry:

agodemar, 2018-06-10:
I close this issue for now. We'll keep it simple and go with a release every new commit.

closed 2018-06-10 by agodemar