JSBSim-Team/jsbsim #64: P-Factor gives no effect?

dany93, 2018-06-14:
Hi,

Sorry, I'm not sure that the subject of my question is a bug (hard to believe after such a long time), but I don't know where to place it. The problem seems so simple that I don't know where to look at.
Prior to disturbing you, I've posted in the FG forum (Development ‹ Aircraft ‹ Flight dynamics model) to insure that I was not doing a mistake but I've had no response.
https://forum.flightgear.org/viewtopic.php?f=49&t=34376

In the propeller file, the
<p_factor> number </p_factor>
seems having no effect, even at high RPM and AoA angles. (I've done tests on two different single-engine aircraft, having different propeller files)

I've tried p_factor values up to 10000, which should give a very exaggerated yaw effect, at no avail.

An excerpt of my test file (c172p):
(attention: the c172p already has a spiralling propwash effect, which preferably should be cancelled to isolate this P-Factor effect. Property: `/fdm/jsbsim/aero/coefficient/spiral-propwash-coeff = `)

File excerpt:

```
<propeller name="NACA 640 (5868-9) 76-inch 20deg Two-Blade Propeller" version="1.01">

<ixx> 1.67 </ixx>
<diameter unit="IN"> 76.0 </diameter>
<numblades> 2 </numblades>
<p_factor> 60 </p_factor>

<table name="C_THRUST" type="internal">
    <tableData>
        0      0.1040
        0.1    0.1040
        (etc...)
    </tableData>
</table>

<table name="C_POWER" type="internal">
    <tableData>
        0      0.0660
        0.1    0.0650
        (etc...
    </tableData>
</table>

</propeller>

```
I could not find a property which would enable me to check the effect in-sim.

I tried understanding it by doing the math calculation (based on velocity-vector additions) simplified for a pure-AoA aircraft disorientation but, very possibly due to my more than basic knowledge of C++, I couldn't understand the FGPropeller.cpp code at SF.net (lines 254-).
The only thing that is clear for me: "the effect is simulated by a shift of the acting location of the thrust".
After that, I think it should be proportional to the aircraft velocity, propeller RPM, AoA (small angle approximation).

Thank you if you can help, and sorry if that's my fault.

Daniel Dubreuil

bcoconni, 2018-06-14:
@dany93 

The p factor effect in JSBSim is a hack, not sure how realistic it is. You are correct that it simulates the p factor effect by a moment generated by a shift of the point of application of thrust.

* In normal operation (i.e. when the aircraft is in level flight and no cross wind), the thrust direction is parallel to the propeller axis of rotation and is applied at the center of the propeller. There is no p factor effect.
```
______ (propeller)              (propeller) _______
   |                                           \---> (tangential velocity)
   |                                            \
   |                                             \
   V                                              V
(aircraft velocity                        (aircraft velocity
in normal operation)                    with cross-wind or AoA)

```

* When operating with an AoA or with some cross wind (or a combination of both), the air velocity has a component parallel to the propeller plane (this component is stored in the variable called `tangentialVel`). The p factor effect is modeled by a moment proportional to the thrust and which lever arm is the product of the user supplied `P_Factor` and of the angle between the propeller axis and the aircraft velocity (variable `angle`)

So, the aircraft will be influenced by the p factor if, and only if, the aircraft velocity is not perpendicular to the propeller plane (non zero AoA or cross wind). Have you tested your aircraft in this condition ?

Also it seems that it is assumed that `angle` is small because the tangential component should be much smaller than the total aircraft velocity. However this assumption breaks when operating the aircraft on ground with strong cross wind. For the purpose of your tests, you should get an exaggerated p factor effect when operating the aircraft on ground with a strong cross wind.

> I could not find a property which would enable me to check the effect in-sim.

Indeed you cannot access the results of these calculations via the properties. However this data can be obtained by another means. Just add the following XML code to your aircraft definition
```xml
<output name="propulsion.csv" type="CSV" rate="20">
  <propulsion> ON </propulsion>
</output>
```
This will output data at 20 Hz in a CSV file named `propulsion.csv`. If you open that file with a spreadsheet (MS Excel or LibreOffice Calc) you will find 2 columns `NACA 640 (5868-9) 76-inch 20deg Two-Blade Propeller PFactor Pitch (engine 0)` and `NACA 640 (5868-9) 76-inch 20deg Two-Blade Propeller PFactor Yaw (engine 0)`. These 2 columns contain the moment (in lbf.ft) generated by the p factor along the pitch and yaw directions.

I don't think this feature has been investigated very much so far, so we are very much interested by any feedback you can give :smiley:

dany93, 2018-06-16:
@bcoconni 

Hi Bertrand, and thank for your detailed response and explanations.

First, you made me clear what the tangential velocity was. I believed that it was the (circular) linear velocity of the prop blade due to its spinning.

To be sure, please (in the FGPropeller.cpp file):
`tangentialVel` is the magnitude of the tangential velocity (component in the propeller plane),
`localAeroVel(eU)` is the x velocity component in the body (aircraft) axis (usual velocity)
l`ocalAeroVel(eV)` and` (eW)` are the y and z velocity components in the body (aircraft) axis
Correct?

Also, I discovered the way to output the data in a .csv file. Fine.

>  (non zero AoA or cross wind). Have you tested your aircraft in this condition ?

Yes, of course. Preferably up to 10 deg AoA or a bit more
@dany wrote:
><p_factor> number </p_factor>
seems having no effect, even at high RPM and AoA angles

I've observed the `NACA 640 (5868-9) 76-inch 20deg Two-Blade Propeller PFactor Pitch` and `yaw` values in the csv files.

I've set different `p_factor` values (from 0 to 100000) and have observed the yaw and pitch moments with varying the AOA.

There is at least one thing weird: 
**Whatever the p_factor value in the prop file, p_factor from 0 to 100000, the pitch and yaw moments are in the 10, ...30 ... 50 range.**
These values vary with AoA, becoming low or slightly negative for negative AoA. Which is correct at first view.
In any case, they are rather weak (compared with the rudder moment `aero/coefficient/Cndr`, about 180 for 10% deflection at 60 kt). Which may explain why I don't see the effect.

@bcoconni wrote
> The p factor effect in JSBSim is a hack, not sure how realistic it is
 it simulates the p factor effect by a moment generated by a shift of the point of application of thrust.

Yes, but this hack seems close to reality.
Much more complicated is the shift variation with prop RPM, aircraft velocity and AoA (even for small AOA). But a mere variation with AoA might do the trick, enough to give enough subjective effect.

_I would prefer if someone could confirm (or not) my observations. I'm surprised that no one noticed that, I would not like making everybody losing his time if this is a mistake from me._

bcoconni, 2018-06-17:
@dany93 
> `tangentialVel` is the magnitude of the tangential velocity (component in the propeller plane),

Correct.
> `localAeroVel(eU)` is the x velocity component in the body (aircraft) axis (usual velocity)
`localAeroVel(eV)` and `(eW)` are the y and z velocity components in the body (aircraft) axis.

Not exactly. This is the velocity component in the propeller frame. Remember that you can provide an `<orient>` element for the propeller which is mainly used to tilt the propeller such as is done for the `p51d`. In the propeller frame `x` is the axis of rotation and the other axes correspond to the propeller plane.

> There is at least one thing weird:
**Whatever the p_factor value in the prop file, p_factor from 0 to 100000, the pitch and yaw moments are in the 10, ...30 ... 50 range.**
These values vary with AoA, becoming low or slightly negative for negative AoA. Which is correct at first view.
In any case, they are rather weak (compared with the rudder moment `aero/coefficient/Cndr`, about 180 for 10% deflection at 60 kt). Which may explain why I don't see the effect.

I'll investigate that.

dany93, 2018-06-17:
On the Cap10B, with the` p_factor` value in the prop file = 5.29 or 10000, the pitch and yaw moments are  always 0, even at full thrust.
Also, I don't understand why, 
- exactly 0 with the Cap10B (one engine file)
- and 10, ... 30. ... 50 with the C172P (engine 1 out of 0 and 1).
The C172P has two "virtual" engines + propellers, 160 hp and 180 hp, only one is running at a time.

dany93, 2018-06-17:
**I've just discovered that a `p_factor` had already been set in the FDM file**!!! Instead of the propeller file. Weird idea..
I have to investigate that.

dany93, 2018-06-17:
- If I set the `p_factor` = 100 **in the prop file**, it does nothing (and moments = 0 in the csv file).
- If I set it at 100 **in the FDM file**, it gives a P_Factor effect in flight at high AoA (which seems correct), and values up to +/- 500,... 600  in the csv file. But curiously, the pitch moment values are greater than the yaw moment, even by trying to keep the slip ball centered.

At least now, the P-Factor has an effect, correct at first view in flight, but only if set it in the FDM file. Is it expected?

bcoconni, 2018-06-17:
@dany93 
> **At least now, the P-Factor has an effect, correct at first view in flight, but only if set it in the FDM file.** Is it expected? The JSBSim Thrusters wiki does not say like this.

Well, according to the documentation that is not expected but that is indeed what the code does. Below is an extract of `src/models/propulsion/FGPropeller.cpp` (lines 135-138). The first line makes a reference to `GetParent()` which means that, **as you mentioned,** `p_factor` **is read from the FDM file.**
```c++
local_element = prop_element->GetParent()->FindElement("p_factor");
if (local_element) {
  P_Factor = local_element->GetDataAsNumber();
}
```
Strangely enough `sense` is also read from the FDM file (i.e. the parent level) but that is correctly documented. According to [JSBSim Doxygen documentation](https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGPropeller.html): `Sense is specified in the parent tag of the propeller`. Unfortunately there is no such mention for the documentation of `p_factor`.

Good catch ! :+1: 

I guess that `sense` is located outside the propeller file so that you can specify a different rotation sense for each propeller of an aircraft while using a common definition file for all the propellers. *I suppose* that the same reasoning was applied to `p_factor` for which the coefficient could depend on the propeller installation (right wing/left wing) but that is only a supposition.

I would suggest to leave the code as it is and update the documentation (i.e. mention that `p_factor is specified in the parent tag of the propeller`). The reason why I suggest not to modify the code is that some aircraft designers seem to have already defined `<p_factor>` at the correct place in their FDM.
Is that OK with you ?

@agodemar , @seanmcleod , @djlinse does that make sense to you as well ?

bcoconni, 2018-06-17:
I have searched in the commit history and found an old version of JSBSim (May 2, 2005) where `p_factor` was read in `FGEngine.Cpp` from the XML parent level along with the propeller orientation and sense. Check [the 13 years old FGEngine.cpp](https://github.com/JSBSim-Team/jsbsim/blob/eeb87bc82a597064403b6a9841f7018c5da00924/FGEngine.cpp) lines 270-276.

So this situation is not coincidental but it is the intended behavior for a long time.

dany93, 2018-06-17:
@bcoconni 

> `local_element = prop_element->GetParent()->FindElement("p_factor");`
> `p_factor is specified in the parent tag of the propeller`

You are teaching me the basics of the code language...

> I would suggest to leave the code as it is and update the documentation (i.e. mention that p_factor is specified in the parent tag of the propeller)
> Is that OK with you ?

OK. I'm fully convinced by your arguments.

Also, if possible, if you (or I?) can explain these subtleties in [JSBSim Thrusters wiki](http://wiki.flightgear.org/JSBSim_Thrusters#Configuration_file_format_3). It is the first document that most people read. That's already done for the `sense`.   (on condition that readers understand what  `parent tag` means)
If you prefer, I can do it (also, adding that the "parent tag" is generally the FDM... for newbies like me).

Thank you Bertrand, for your attention and the time you spent to write these explanations.

Dany

agodemar, 2018-06-17:
@bcoconni 

> @agodemar , @seanmcleod , @djlinse does that make sense to you as well ?

That makes sense to me.

djlinse, 2018-06-18:
@bcoconni I haven't dug into this at all, and propellors aren't necessarily my strongest area, so I can't really offer an opinion.  Sorry.

bcoconni, 2018-06-18:
According to our discussion, I've just updated the JSBSim Doxygen [documentation for FGPropeller](https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGPropeller.html) via the commit 9b1b598.
In particular a new paragraph has been added:

*The parameters \<sense\> and \<p_factor\> must be specified at the parent level i.e. in the \<thruster\> element. This allows to specify different sense and P factor values for each propeller of the model while using the same definition file for all the propellers.*

dany93, 2018-06-18:
Thank you, Bertrand.

dany93, 2018-06-19:
I've updated [FGPropeller](http://wiki.flightgear.org/JSBSim_Thrusters#FGPropeller) in the JSBSim Thrusters wiki.

bcoconni, 2018-06-19:
Excellent ! Can we declare the issue closed ?

dany93, 2018-06-19:
Yes, it's solved.
And thank you once more.

closed 2018-06-21 by bcoconni