JSBSim-Team/jsbsim #67: [C++ Language] Impact of new C++ standards (e.g. C++11, or C++14) on JSBSim

agodemar, 2018-06-17:
Looking at the [ISO C++ Standard current status](https://isocpp.org/std/status) and considering the [support of major compilers to the various standards](http://en.cppreference.com/w/cpp/compiler_support), a question comes naturally: Shall we consider adopting the newest, consolidated, C++ features?

bcoconni, 2018-06-18:
That's an interesting option. I am not familiar with all the goodies brought by the last revisions of the C++ standard but there are a couple of interesting features in C++11:
* `constexpr`
* The type inference `auto`
* The range based `for` loops

:+1: to that proposal.

agodemar, 2018-06-19:
@bcoconni I like those new features as well as the `std::shared_ptr` and lambda functions. [See for example here](http://cppdepend.com/blog/?p=319).

My proposal: From now on switch to C++11, whether we are reviewing old code or we are adding new features to the library.

bcoconni, 2018-06-21:
Since we are embedding the code of JSBSim in FlightGear and since they are using C++11, I suggest we do not use features of C++ earlier than C++11.

seanmcleod, 2018-06-21:
@bcoconni I presume you meant to write not **later** than C++11 (from 2011), i.e. don't use any features in C++14 or C++17?

bcoconni, 2018-06-21:
@seanmcleod 
Ooops ! Yes you are right, I meant **later** :smile:

agodemar, 2018-06-22:
I propose this additional configuration in `src/CMakeLists.txt`
```
if(NOT CMAKE_VERSION VERSION_LESS 3.1)
    set(CMAKE_CXX_STANDARD 11)
else()
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
endif()
```

bcoconni, 2018-06-22:
@agodemar Is the flag `-std=c++11` recognized by all compilers (Visual C++, gcc and clang) ?

agodemar, 2018-06-22:
- Visual C++: [The compiler does not offer a specific C++11 mode and defaults to C++14, but that standard is fully inclusive of C++11](https://stackoverflow.com/questions/47043869/is-c11-available-in-visual-studio-2017).
- g++: [yes](https://stackoverflow.com/questions/10363646/compiling-c11-with-g)
- clang: [yes](https://clang.llvm.org/cxx_status.html)

bcoconni, 2018-06-22:
Thanks.

So I guess Visual C++ will complain if we provide it with the flag `-std=c++11` ? Which means that the code should be something like below (swapping the two `if` branches to save the negation `NOT` on the CMake version condition) ?
```cmake
if (CMAKE_VERSION VERSION_LESS 3.1)
  if (NOT MSVC)
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
  endif()
else()
  set(CMAKE_CXX_STANDARD 11)
endif()
```

agodemar, 2018-06-22:
It should work.

There is no C++11 flag for Visual C++. For later versions of the standard the flags are: `/std:c++14` and `/std:c++17` (or `/std:c++latest`).

bcoconni, 2018-06-23:
@agodemar 
Will you push that change to `master` ? I suggest that you skip the pull request process and push directly to `JSBSim:master`

agodemar, 2018-06-23:
I've changed the title of this issue, which is now closed.

closed 2018-06-23 by agodemar