JSBSim-Team/jsbsim #70: JSBSim bindings for other languages

agodemar, 2018-06-19:
We can provide JSBSim *bindings* to other languages, such as
- Python
- Matlab
- Java

The bindings would come as Application Programming Interfaces (APIs), that is, wrapper libraries that bridge between a set of selected internal JSBSim C++ functions/data structures and other programming languages. This enhancemnt would expand the usability of our software. 

I am aware of automatic workflows, driven by Python scripts, that produce bindings for different languages. The automation is based on functions and class attributes in the C++ library. As an example, [here is TiGL software by the German DLR](https://github.com/DLR-SC/tigl) and [here is TiGL folder where bindings are stored](https://github.com/DLR-SC/tigl/tree/cpacs_3/bindings).

bcoconni, 2018-06-21:
We are currently [providing a Python binding to JSBSim](https://github.com/JSBSim-Team/jsbsim#building-the-python-module-of-jsbsim) so I don't understand why you are requesting that ?

agodemar, 2018-06-21:
@bcoconni because [Python is not the only language out there](https://www.tiobe.com/tiobe-index/). Not sure if [Python is the language of choice in Aerospace](https://www.aversan.com/top-programming-languages-in-aerospace/).

I did the example of TiGL, which I use in my Java projects, but others use as well in Python or Matlab.

bcoconni, 2018-06-21:
@agodemar 
OK I am rephrasing my statement because it was unclear: I understand you might want to add other languages bindings but why did you mention Python among the languages for which we *should* provide a binding while we are already *providing* one ?

agodemar, 2018-06-22:
@bcoconni That's simply a list. I envisage a folder named `bindings` with subfolders `python`, `matlab`, `java`, etc. where we put the sources we need to generate the interfaces to those languages. Then we would have binaries packaged and distrinuted in each release.

seanmcleod, 2018-06-22:
In terms of MATLAB integration I seem to remember coming across one many years ago. Doing a quick search it looks like @agodemar you were involved in it?

[JSBSim S-function Project](http://jsbsim.sourceforge.net/matlab.html)

How different would the MATLAB integration be in terms of the S-function project and the DLR binding approach?

agodemar, 2018-06-22:
@seanmcleod I guess it was 2009 when I first came out with a MEX-file (Matlab Executable) that exposed the FGFDMExec class to Matlab environment. I did it with MS Visual C++ 2008 and it was more or less as simple as producing a DLL. Then it came all the subsequent work to build on my MEX-file an S-function to be used in Simulink.

DLR uses a [Python script](https://github.com/DLR-SC/tigl/blob/cpacs_3/bindings/bindings_generator/matlab_generator.py) that generates M-files and MEX-files by scanning a set of C++ classes and their member functions that return an object of type TIGL_EXPORT. I still have to grasp all the details of this process but I like the idea. Also because there are similar Python scripts that generate the other bindings for Java, Python, and Fortran.

Regarding the Matlab binding I like the basic approach of providing only the MEX file. Then the general user may build on it his own Matlab or Simulink project.

benhager, 2019-10-05:
@agodemar I'd be interested in resurrecting the ability to run JSBSim in simulink via an s-function... is this something you could help me get started with? I can't seem to find any files that run in newer Matlab/Simulink as a starting point.

agodemar, 2019-11-10:
@benhager sure we can work jointly at this. I'll have to collect the old files on my side. But I think we can start from [the Matlab page on JSBSim website](http://jsbsim.sourceforge.net/matlab.html).

benhager, 2019-12-03:
@agodemar that we be good, the links on that page to download files don’t work for me. Were you able to locate any old files to use as a starting point?

agodemar, 2019-12-03:
@benhager 
The latest available sources are on [Matlab Central: ](https://www.mathworks.com/matlabcentral/fileexchange/25042-jsbsim-s-function-gui-0-3).

This is an [alternative link: ](https://www.dropbox.com/s/8b45xknrpjppoee/JSBSimGUI_SFunction.zip?dl=0)

The project used to work back in 2010. To make it compile and work with the current versions of JSBSim *and* Matlab, a careful revision is required.

benhager, 2019-12-06:
@agodemar thank you for this. I’m swamped at work the next couple weeks, but I’ll probably start taking a look at it in detail over my holiday break (last week of December).

rega0051, 2019-12-14:
I'm interested in getting the matlab mex working again. I work on linux and Windows with MinGW as the compiler (which is now the MATLAB default for mex). I'm not so interested in the s-function, but it's likely that most of the challenges are common.

I had been looking at this repo: https://github.com/podhrmic/JSBSim-Matlab

agodemar, 2019-12-14:
Ok, I think I'll make a new branch `JSBSim_Matlab` and put some code there.

bcoconni, 2019-12-14:
@agodemar Do you mind if I assign this ticket to you ?

agodemar, 2019-12-14:
Sure
--
Inviato da myMail per Android sabato,

rega0051, 2019-12-16:
I forked the podhrmic repo and cleared out all the non-essentials. For example the repo had a GUI and some additional functionality built on top of JSBSim that I didn't necessarily want.

https://github.com/rega0051/JSBSim-Matlab

I was able to get JSBSim to compile on Windows10 with MinGW, and also got the mex to compile in R2019b. I'm still getting NaNs on the outputs and haven't tracked those down.

Launchpad-McQ, 2020-05-16:
Hi, did you resolve the NaNs on the outputs and get it all working?
@rega0051

rega0051, 2020-05-22:
@Launchpad-McQ  I didn't look into it. My need for the MATLAB bindings was overcome by other events. There were quite a number of changes on the Mathworks side of things, particularly on windows, that have changed since the last version of this worked.

bcoconni, 2021-01-09:
As per the discussion in #366, we could add Julia to this list !

seanmcleod, 2021-07-17:
@agodemar @bcoconni now that we've added bindings/integration support for Julia and Matlab since this issue was opened do we want to keep this issue open? Or do we close it and re-open it if/when someone requests support for another language?

bcoconni, 2021-07-17:
To everyone, thanks to @sthelia, the PR #445 has just been merged. JSBSim can now be used as an S-Function from within Matlab.

If you need support about how to use it, then please open a [new discussion](https://github.com/JSBSim-Team/jsbsim/discussions). If you find bugs or want to suggest a new feature, please open an [issue](https://github.com/JSBSim-Team/jsbsim/issues).

bcoconni, 2021-07-17:
> Now that we've added bindings/integration support for Julia and Matlab since this issue was opened do we want to keep this issue open? Or do we close it and re-open it if/when someone requests support for another language?

Good point: I'd suggest to close it.

agodemar, 2021-07-17:
> Good point: I'd suggest to close it.

Agreed, @seanmcleod

closed 2021-07-17 by seanmcleod