JSBSim-Team/jsbsim #71: [Dependencies] Boost libraries

agodemar, 2018-06-20:
Considering the relevance of the [Boost libraries](https://www.boost.org/) in the C++ community, and considering the amount of functionalities it provides, ire there a good reasons why we should not depend on them?

seanmcleod, 2018-06-20:
A fair number of (more than 10) of the Boost libraries were included in the C++11 standard and continue to be added to later C++ standards. So with some lag the most generic libraries that the C++ committee thinks will be useful in the C++ standard are added.

[Boost Features Overlap with C++11](https://stackoverflow.com/questions/8851670/which-boost-features-overlap-with-c11)

Sometimes the leading/bleeding edge boost libraries aren't compatible with as many current compilers compared to say an older C++ standard like C++11 etc.

I wouldn't say that we should automatically depend on Boost, in general the fewer dependencies the better, unless there is a compelling reason/example where something would be a LOT better/easier if we used Boost as opposed to standard C++11, 14 etc.

So I'd suggest rather waiting for some new feature or large refactoring where someone motivates a good case for why it would be much better to use Boost for the new feature/refactoring etc.

agodemar, 2018-06-20:
I think the [Asio](https://www.boost.org/doc/libs/1_67_0/doc/html/boost_asio.html) for network and low-level I/O programming would be good to have in the standard. They are working on it.

djlinse, 2018-06-20:
I would agree with @seanmcleod that there would need to be a serious case made before requiring Boost libraries.  Isn't one of the "claims to fame" of JSBSim that it doesn't rely on external libraries?

I *love* python, but one of the issues I always had with distributing anything was requiring the recipient to have python plus X plus Y plus Z plus ...  Sure there are distributions that give most things, but there always seems like there is one more thing.

bcoconni, 2018-06-21:
Same as @seanmcleod and @djlinse JSBSim has never relied on any external libraries and that is one of its strength: you just need a C++ compiler to build it.

In addition I think PR #73 is making sense and that means that, in some cases, we might need to build JSBSim locally with Cython and Python distutils. In that case not having any dependency will make our lives easier.

closed 2018-08-16 by agodemar