JSBSim-Team/jsbsim #75: query_property_catalog() -> TypeError: expected bytes, str found

Gor-Ren, 2018-06-22:
I am writing a Python (3.6) application which wraps an instance of JSBSim.

I would like to retrieve a list containing the simulation property catalog, and am trying to use the wrapper method `query_property_catalog(self, check)` for this. This method accepts one arg, `check: str`.

However, calling this method with any string (empty or non-empty) raises a TypeError with the following trace:
```
Traceback (most recent call last):
  File "~/dev/gym-jsbsim/test/query_props_minimal_example.py", line 11, in <module>
    properties = fdm.query_property_catalog('ic/')
  File "jsbsim.pyx", line 512, in jsbsim.FGFDMExec.query_property_catalog
  File "stringsource", line 15, in string.from_py.__pyx_convert_string_from_py_std__in_string
TypeError: expected bytes, str found
```
Trying to call the function with a `bytes` object of a string encoded in UTF-8 raises the error `TypeError: a bytes-like object is required, not 'str'`.

I found a [similar issue in another project](https://github.com/clab/dynet/issues/1187) which suggested the problem may be related to string encoding changes Python 3.6.

Here is a minimal example which reproduces the issue:
```
import jsbsim

fdm = jsbsim.FGFDMExec(root_dir="/your/root/dir/jsbsim-code")
fdm.load_model('c172x')
fdm.load_ic('reset00', useStoredPath=True)
fdm.run_ic()

# empty string raises TypeError: expected bytes, str found
properties = fdm.query_property_catalog('')
# non-empty string raises same
properties = fdm.query_property_catalog('ic/')
# byte encoding of empty string raises TypeError: a bytes-like object is required, not 'str'
properties = fdm.query_property_catalog(''.encode('utf-8'))
```
I cannot find any usage of `query_property_catalog()` in the existing Python tests.

I am using Python 3.6, Cython 0.28.3, JSBSim built from cc9beb70 and the JSBSim Python3 library installed from release python3-JSBSim_1.0.0-184.trusty.amd64.deb.

seanmcleod, 2018-06-22:
I see the same issue on Windows with Python 3. Busy looking into the issue, I think I have a fix.

seanmcleod, 2018-06-22:
@Gor-Ren I've submitted a pull request to fix the issue you reported, see pull request [77](https://github.com/JSBSim-Team/jsbsim/pull/77).

However take note of the additional bug that I filed while investigating your issue which you may run into depending on what you want to do with the property catalog, issue [76](https://github.com/JSBSim-Team/jsbsim/issues/76).

bcoconni, 2018-06-23:
Can this topic be closed now that the PR #77 has been merged ?

closed 2018-06-23 by seanmcleod