JSBSim-Team/jsbsim #76: Python fdm.get_property_catalog(check) crashes

seanmcleod, 2018-06-22:
In fixing issue [75](https://github.com/JSBSim-Team/jsbsim/issues/75) which related to an issue with Python 3 string incompatibility with `query_property_catalog()` I tested `get_property_catalog()` and noticed that it crashes the process.

Internally `get_property_catalog()` calls `query_property_catalog()'.

```python
    def get_property_catalog(self, check):
        """
        Retrieves the property catalog as a dictionary.
        """
        catalog = {}
        for item in self.query_property_catalog(check):
            catalog[item] = self.get_property_value(item)
        return catalog
```

The issue is that the property names returned by `query_property_catalog()` have a read-write status string appended to them, e.g. `'ic/vc-kts (RW)'`.

And the call to `get_property_value()` crashes if this read-write status string is included, e.g.

```python
fdm.get_property_value('ic/vc-kts')
0.0
fdm.get_property_value('ic/vc-kts (RW)')
--crash
```

So we need to remove the read-write status before calling `get_property_value()`.

Gor-Ren, 2018-06-22:
Thanks for the fast fix on issue #75! I have reproduced this new crash on Ubuntu also. The crash appears to be specifically caused by the space and parenthesis characters, because you can remove them and it works:
```python
fdm.get_property_value('ic/vc-ktsRW')
0.0
```
The new property name `'ic/vc-ktsRW'` is not valid. Despite this, the program returns `0.0`, and it does so for all invalid property names.

I would suggest there is a wider issue here, that `get_property_value()` and `set_property_value()` do not verify properties exist before setting/retrieving them. This could lead to some nasty, subtle bugs when properties are misspelled.

For example,
```python
fdm.get_property_value('qwerty')
0.0
fdm.set_property_value('qwerty', 42.0)
fdm.get_property_value('qwerty')
42.0
```
My preferred behaviour would be for Python to raise a `KeyError` or similar so that attention is drawn to the problem. Unfortunately I don't see a nice and simple way to do this in the Python library (the list of valid properties depends on the loaded aircraft and will change during runtime, but we don't really want to retrieve and check against the property catalog every access). 

Perhaps it could be more efficiently implemented on the C++ side? Looking at the relevant C++ methods [`GetPropertyValue()`](http://jsbsim.sourceforge.net/JSBSim/classJSBSim_1_1FGFDMExec.html#a40480255797aace3ea845ff1d46ff597), [`GetNode()`](http://jsbsim.sourceforge.net/JSBSim/classJSBSim_1_1FGPropertyNode.html#a0a656ba4d87d627eff0f6f52326d68d0) and [`GetDouble()`](http://jsbsim.sourceforge.net/JSBSim/classJSBSim_1_1FGPropertyNode.html#abec94b9a386364af233294f118d37059) it does seem that returning 0.0 on an unrecognised property is the intended behaviour, so I would be scared of breaking other things.

Anyway, the immediate issue here can be fixed by doing something like:
```python
for item in self.query_property_catalog(check):
    property_name = item.split(" ")[0]  # remove any (RW) flags
    catalog[item] = self.get_property_value(property_name)
```
since space characters do not occur in property names, only when a (R)/(W)/(RW) flag is on it.

bcoconni, 2018-06-23:
@Gor-Ren 
> Anyway, the immediate issue here can be fixed by doing something like:
```python
for item in self.query_property_catalog(check):
    property_name = item.split(" ")[0]  # remove any (RW) flags
    catalog[item] = self.get_property_value(property_name)
```
> since space characters do not occur in property names, only when a (R)/(W)/(RW) flag is on it.

Pushed to `master` via commit 94677b7

bcoconni, 2018-06-23:
@Gor-Ren 
> I would suggest there is a wider issue here, that `get_property_value()` and `set_property_value()` do not verify properties exist before setting/retrieving them. This could lead to some nasty, subtle bugs when properties are misspelled.

I agree with you regarding the getter part but I would tend to disagree regarding the setter. If we can only set the value of existing properties, we will be unable to create new ones ?

> My preferred behavior would be for Python to raise a `KeyError` or similar so that attention is drawn to the problem.

This is now implemented via commit d05d9ed with a small modification with respect to your suggestion. The method `get_property_value()` keeps its legacy behavior i.e. returns 0.0 for non existent property. This is because `get_property_value()` is a wrapper to the C++ method `FGPropertyManager::GetPropertyValue()` so the Python getter should have the same behavior than the underlying C++ routine.

**However** property values can also be accessed in a more Pythonic way `fdm['prop_name']`. And when this syntax is used, JSBSim now checks the existence of the requested property and raises `KeyError` if it does not exist.

So now your example gives the following output
```python
>>> fdm['qwerty']
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "jsbsim.pyx", line 206, in jsbsim.FGFDMExec.__getitem__
KeyError: 'No property named qwerty'
>>> fdm['qwerty'] = 42.0
>>> fdm['qwerty']
42.0
```

closed 2018-06-23 by seanmcleod