JSBSim-Team/jsbsim #82: Unit test do not work with Anaconda/Python3 on Windows

agodemar, 2018-07-02:
I generate makefiles for JSBSim with CMake on ubuntu within my WSL (Windows System for Linux) box on Windows. I have Anaconda installed with Python 3.4.

This is what I do:

```bash
agodemar@DEMARCO-PC:/mnt/c/Users/DeMarco-PC/JSBSim-agodemar/jsbsim/build$ cmake ..
-- Found CxxTest: /usr/include
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE)
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/Users/DeMarco-PC/JSBSim-agodemar/jsbsim/build
agodemar@DEMARCO-PC:/mnt/c/Users/DeMarco-PC/JSBSim-agodemar/jsbsim/build$ make
Scanning dependencies of target libJSBSim
[  1%] Building CXX object src/CMakeFiles/libJSBSim.dir/FGFDMExec.cpp.o
[  1%] Building CXX object src/CMakeFiles/libJSBSim.dir/FGJSBBase.cpp.o
[  2%] Building CXX object src/CMakeFiles/libJSBSim.dir/initialization/FGInitialCondition.cpp.o
[  3%] Building CXX object src/CMakeFiles/libJSBSim.dir/initialization/FGTrim.cpp.o
[  4%] Building CXX object src/CMakeFiles/libJSBSim.dir/initialization/FGTrimAxis.cpp.o
...
[ 68%] Building CXX object src/CMakeFiles/libJSBSim.dir/simgear/misc/strutils.cxx.o
[ 69%] Building CXX object src/CMakeFiles/libJSBSim.dir/simgear/io/iostreams/sgstream.cxx.o
[ 70%] Linking CXX static library libJSBSim.a
[ 70%] Built target libJSBSim
Scanning dependencies of target JSBSim
[ 70%] Building CXX object src/CMakeFiles/JSBSim.dir/JSBSim.cpp.o
[ 71%] Linking CXX executable JSBSim
[ 71%] Built target JSBSim
[ 71%] Compiling Cython CXX source for jsbsim...
[ 72%] Building Python modules...
running build_ext
building 'jsbsim' extension
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
running build_ext
building 'fpectl' extension
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
[ 72%] Built target PythonJSBSim
[ 73%] Generating FGMatrix33Test.cpp
Traceback (most recent call last):
  File "/usr/bin/cxxtestgen", line 7, in <module>
    import cxxtest.cxxtestgen
  File "/usr/share/cxxtest/cxxtest/__init__.py", line 33, in <module>
    from cxxtest.cxxtestgen import *
  File "/usr/share/cxxtest/cxxtest/cxxtestgen.py", line 18, in <module>
    import __release__
ModuleNotFoundError: No module named '__release__'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/cxxtestgen", line 14, in <module>
    sys.path.append( os.path.abspath( os.path.sep.join([currdir, '..', python3]) ) )
NameError: name 'python3' is not defined
tests/unit_tests/CMakeFiles/FGMatrix33Test1.dir/build.make:61: recipe for target 'tests/unit_tests/FGMatrix33Test.cpp' failed
make[2]: *** [tests/unit_tests/FGMatrix33Test.cpp] Error 1
CMakeFiles/Makefile2:581: recipe for target 'tests/unit_tests/CMakeFiles/FGMatrix33Test1.dir/all' failed
make[1]: *** [tests/unit_tests/CMakeFiles/FGMatrix33Test1.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

seanmcleod, 2018-07-02:
Not sure offhand about the _release_ issue. In terms of the python3 issue have you installed Anaconda into WSL/Ubuntu or have you installed it into Windows?

I've done all my Jsbsim python testing in Windows using VS2017's built in CMake support.

agodemar, 2018-07-02:
I have Anaconda into WSL/Ubuntu. This is my Ubuntu version:
```bash
> lsb_release -a
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.4 LTS
Release:        16.04
Codename:       xenial
```
This is my Anaconda:
```bash
> conda info

     active environment : None
       user config file : /home/agodemar/.condarc
 populated config files :
          conda version : 4.5.4
    conda-build version : 3.4.1
         python version : 3.6.4.final.0
       base environment : /home/agodemar/anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/free/linux-64
                          https://repo.anaconda.com/pkgs/free/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/pro/linux-64
                          https://repo.anaconda.com/pkgs/pro/noarch
          package cache : /home/agodemar/anaconda3/pkgs
                          /home/agodemar/.conda/pkgs
       envs directories : /home/agodemar/anaconda3/envs
                          /home/agodemar/.conda/envs
               platform : linux-64
             user-agent : conda/4.5.4 requests/2.18.4 CPython/3.6.4 Linux/4.4.0-17134-Microsoft ubuntu/16.04 glibc/2.23
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

bcoconni, 2018-07-07:
@agodemar 
```
Traceback (most recent call last):
  File "/usr/bin/cxxtestgen", line 7, in <module>
    import cxxtest.cxxtestgen
  File "/usr/share/cxxtest/cxxtest/__init__.py", line 33, in <module>
    from cxxtest.cxxtestgen import *
  File "/usr/share/cxxtest/cxxtest/cxxtestgen.py", line 18, in <module>
    import __release__
ModuleNotFoundError: No module named '__release__'
```
The module `__release__` is part of cxxtest and should be installed under `/usr/lib/python3.6/site-packages/cxxtest/__release__.py`. If it is not then it is likely that something went wrong during the installation of cxxtest.

seanmcleod, 2018-07-07:
@bcoconni I was wondering how come you added these. Wasn't sure if it was because of some specific limitation with the Python test setup.

seanmcleod, 2018-07-08:
Okay in terms of using CxxTest on AppVeyor it looks like we either need to get it installed or at least configured. From the latest AppVeyor build:

```
-- Could NOT find CxxTest (missing: CXXTEST_INCLUDE_DIR CXXTEST_PYTHON_TESTGEN_EXECUTABLE) 
```

bcoconni, 2019-03-17:
@agodemar 
The AppVeyor interface has been updated to clone and build CxxTest on Windows.
May be you can use it to fix your problem ?
https://github.com/JSBSim-Team/jsbsim/blob/574f8553cd680657f05af0f9fb33a5af9c322987/appveyor.yml#L23-L29
@seanmcleod 
> Okay in terms of using CxxTest on AppVeyor it looks like we either need to get it installed or at least configured. From the latest AppVeyor build:
```
-- Could NOT find CxxTest (missing: CXXTEST_INCLUDE_DIR CXXTEST_PYTHON_TESTGEN_EXECUTABLE) 
```
This is now fixed since commit eecc318a0fa1cacc01b4c13e3df5b32013561f90

agodemar, 2019-03-18:
@bcoconni 
Great! Thanks

closed 2020-12-11 by bcoconni