JSBSim-Team/jsbsim #83: No way to handle repetitive work. (Was "XML copypasta" on jsbsim-users).

mike402, 2018-07-02:
Is there a better way to program repetitive chains of functions than
simply copypasting?

For example, here I had to do a lot of copypasta for each pump and valve, and
now if I decide to fix something I will have to do it all over again:

https://gitlab.com/mdanil/Tu-144/blob/master/Tu-144-jsbsim/propulsion/fuel.xml

And is there a way to include a system passing some XML entities to
it in the <system include=""> tag? The way I have to handle repetitive
systems now is creating multiple files and editing the entities
manually, which is tedious and error prone, like:

https://gitlab.com/mdanil/Tu-144/blob/master/Tu-144-jsbsim/propulsion/RD-36-51A_0.xml
https://gitlab.com/mdanil/Tu-144/blob/master/Tu-144-jsbsim/propulsion/RD-36-51A_1.xml
https://gitlab.com/mdanil/Tu-144/blob/master/Tu-144-jsbsim/propulsion/RD-36-51A_2.xml
https://gitlab.com/mdanil/Tu-144/blob/master/Tu-144-jsbsim/propulsion/RD-36-51A_3.xml

I have been looking for the proper XML way on and off, and realised I
don't even know what question to search to get info about how the
'something="something"' part of XML tags is related to entities.

mike402, 2018-07-05:
One more example -- distance functions for each INS waypoint:

https://gitlab.com/mdanil/Tu-144/blob/master/Tu-144-jsbsim/NPK/Orbita/func/dist.xml

bcoconni, 2018-07-05:
@mike402 
There are already some features in JSBSim to handle the duplication of XML definitions. At the moment they are limited to:
* engines definition where you can use the special character `#` to replace the engine number.
* file inclusion with the `file=` attribute

If we plan to extend that to other elements, I think we should design how the thing should work before going any further. An important point to my view is consistency with the existing features.

Suggestions ?

mike402, 2018-07-05:
On Thu, Jul 05, 2018 at 08:48:53PM +0000, Bertrand Coconnier wrote:
> @mike402
> There are already some features in JSBSim to handle the duplication of XML
> definitions. At the moment they are limited to:
>
>   • engines definition where you can use the special character # [...]

The only thing I can think about so far is allowing user to add other things
than include="" (sorry, I don't know the right name for this in XML) in the
system tag that would define some parameters for the inside of the file being
included -- like

<system include="asd.xml" param1="fgh" param2="jkl">

so that then &param1; is expanded to "fgh" and &param2; is expanded into "jkl"?

As for repetition within a system file itself, I'm guessing maybe something like
an ID for that channel by which one would instruct JSBSim to create a new
channel of same content.

<channel id="asd" param="fgh">
 ...
</channel>

so that you can
create channels of the same content but with different parameters like this?

<channel copyfrom="asd" param="jkl"/>

Probably related: it would probably be nice to be able to specify the "current
working dir" in the property tree for either channels or systems, so that one
does not have to type out full paths in every place? But there is a problem that
paths beginning with a "/" refer to FlihtGear tree, so that can not be used to
refer to the root of JSBSim's tree.

bcoconni, 2018-12-29:
@mike402 @abassign
To respond to the request of reducing the duplication of XML code in your aircraft models, I just pushed a new commit to FlightGear that allows to use "template functions" everywhere in JSBSim. The idea is to have generic functions which can be repeatedly used in the aircraft models.

For example, say you need to multiply some values by 2 in several places in your model. First, you need to declare a `template` function like below:
```xml
<function name="twice" type="template"> <!-- the type "template" tells JSBSim that it is a special function that will be evaluated later -->
  <product>
    <property> # </property> <!-- The character # stands for the variable of the function -->
    <value> 2.0 </value>
  </product>
</function>
```
Notice in particular the usage of the special character `#`. It can appear several times in a given template function.

The second step is to use that template function everywhere you need it. For that you must use the attribute `apply` of a property.
```xml
<function name="fct1">
  <sum>
    <pi>
    <property apply="twice"> prop1 </property> <!-- the attribute apply="twice" tells JSBSim to apply the template function "twice" to the property "prop1" before proceeding to the evaluation of the function "fct1"-->
    <sin>
      <property apply="twice"> theta </property> <!-- same as above -->
    </sin>
  </sum>
</function>
```
This is equivalent to
```xml
<function name="fct1">
  <sum>
    <pi>
    <product>
      <property> prop1 </property>
      <value> 2.0 </value>
    </product>
    <sin>
      <product>
        <property> theta </property>
        <value> 2.0 </value>
      </product>
    </sin>
  </sum>
</function>
```
The feature is available in the branch `next` of FlightGear. Could you please test that feature and give feedback in that issue report ? Since you both requested to have a feature similar to that, or asked for new functions, I suppose that you can be beta testers ? :smiley:

mike402, 2019-01-04:
Thanks, it will take me a while to wrap my head around this and come up with an efficient use scheme, but this already looks useful!

I think #84 can be merged with this issue, as now there's cyclic clipto, and beside that sawtooth functions can be done as templates?

mike402, 2019-01-04:
But I can immediately say that if the number of arguments is limited to 1, this would be limited to only the simplest use cases?

bcoconni, 2019-01-05:
> But I can immediately say that if the number of arguments is limited to 1, this would be limited to only the simplest use cases ?

I am sure people can be creative about using such a feature :smiley: And I'd rather discuss about how to solve a problem on a real case rather than jumping to negative conclusions. This is not very supportive regarding the amount of time that has been involved in developing this feature :sob:

bcoconni, 2019-01-05:
> No no I didn't meant to say it's useless, I, for one, can immediately think of a hundred uses for it in my plane.

This was purposely provocative :smiling_imp: to point out that we should focus on problem solving rather than on general statements. I'm sure you will find plenty of creative ways to use the template function feature :smile:

> I'm just saying that with multi-argument support it would have been maybe five hundred.

Right but we should move one step at a time.

My plan is to collect feedback about the current implementation of template functions with uses, problems and bugs issued from real case uses. Only then we might consider improvements or changes to the feature.

I have had bad experiences about over-engineering things due to wrong anticipations of future needs of users. This ended up with additional, almost useless, code to maintain for backward compatibility **and** a redesign of the feature.

OpenUAS, 2021-03-18:
Maybe time then to close this issue? @bcoconni your remarks look wise and indicating years of real life development experience, thanks :1st_place_medal: Hope to see you working on JSBsim in the long long future.

bcoconni, 2021-03-21:
@OpenUAS I guess you are right. We got no feedback on that feature for 2 years now, it seems it's time to close the issue now.
If someone wants to report some bugs/feature request please open a new issue. Feedback and support will be handled in [Discussions](https://github.com/JSBSim-Team/jsbsim/discussions).
> Hope to see you working on JSBsim in the long long future.

I've been involved in the project for more than 10 years now. Who knows may be there are some more 10 years of involvement ahead of me ? :smile: Anyway thanks for the positive feedback it's always very much appreciated :+1:

closed 2021-03-21 by bcoconni