JSBSim-Team/jsbsim #84: FlightGear: unable to control JSBSim engines from within JSBSim.

mike402, 2018-07-02:
The "starter", "cutoff" and "seized" engine properties are not exposed
to JSBSim tree. The ones under "/engines/engine[*]" of FlightGear are
boolean and do not accept other data types.

This means if engine control system is written in JSBSim, it has no
clean way of outputting to those properties, especially because JSBSim
can not output bools.

I have to rely on external logic to convert those values and pass them
to FlightGear properties, which makes the engine control code fragile.

bcoconni, 2018-07-07:
@mike402 
All the properties "starter", "cutoff" and "seized" are internally booleans so the properties through which they are exposed will not accept any other data type.

Regarding the property `seized`, it is already exposed via the property `propulsion/engine[#]/seized`. It is a read/write property which means that the user can modify it at will.

The property `cutoff` and `starter` can be set via the properties `propulsion/start_cmd` and `propulsion/cutoff_cmd` after having set `propulsion/active_engine` to the relevant engine number. At the moment, these properties are write only.

mike402, 2018-07-07:
On Sat, Jul 07, 2018 at 08:43:14AM -0700, Bertrand Coconnier wrote:
> The property cutoff and starter can be set via the properties propulsion/
> start_cmd and propulsion/cutoff_cmd after having set propulsion/active_engine
> to the relevant engine number. At the moment, these properties are write only.

But that effectively rules out ability to make engine systems control those
simultaneously, without ugly hacks with multiplexing?

bcoconni, 2018-07-07:
@mike402 
> But that effectively rules out ability to make engine systems control those simultaneously, without ugly hacks with multiplexing?

I just pushed the commit 6999e5c that makes the two properties `propulsion/start_cmd` and `propulsion/cutoff_cmd` read/write.

bcoconni, 2018-07-07:
> I just pushed the commit 6999e5c that makes the two properties `propulsion/start_cmd` and `propulsion/cutoff_cmd` read/write.

Now it is pushed to FlightGear as well.

bcoconni, 2018-07-11:
> But why not per-engine variables like in FlightGear ?
1. Legacy FDM models must not be broken
2. The property must be read from the same property than it was written to
3. Several properties should not access the same piece of data
4. JSBSim has its own agenda and has no obligation whatsoever to mimic FlightGear

mike402, 2018-07-11:
1. How would it break legacy models if the new behaviour was
implemented as opt-in?

2, 3, 4. I don't think I understand this mechanism enough to argue --
but do you mean those vars have been the only option all along, and
FlightGear does the multiplexing from /engines/engine[*]?

mike402, 2018-07-11:
But anyway, multiplexing is not as difficult as I had imagined, in
fact only took 5 minutes to get right -- thank you for implementing
this!

https://gitlab.com/mdanil/Tu-144/blob/2018.3.0/Tu-144-jsbsim/workarounds/enginemux.xml

But is the part marked with the "XXX" comment going to be reliable (as
in, won't skip over some index by accident or confuse the <switch>es
into picking the wrong signals), or should I use sim-time-sec instead?

Now the only thing missing is "propulsion/seized_cmd" -- can you
please expose it as well?

bcoconni, 2018-07-13:
> 1 How would it break legacy models if the new behaviour was implemented as opt-in?

> 2, 3, 4. I don't think I understand this mechanism enough to argue --

Properties `propulsion/start_cmd` and `propulsion/cutoff_cmd` were already existing but were write only (values could not be read). So we basically had 4 options
1. We drop what you call "multiplexing" so that cutoff status could be read from `propulsion/engine[#]/cutoff` but that would mean that the cutoff value would be read from another property than it was written to
2. We duplicate the cutoff status so it can be accessed by 2 properties `propulsion/cutoff_cmd` and `propulsion/engine[#]/cutoff` but we don't want that (would need code duplication and to make sure the data is synchronized for the 2 properties)
3. We move `propulsion/cutoff_cmd` to `propulsion/engine[#]/cmd` but this breaks backward compatibility.
4. The cutoff status is read from `propulsion/cutoff_cmd`

For the reasons I gave previously, option 4 was implemented.

> and FlightGear does the multiplexing from /engines/engine[*]?

The interface is coded in C++ and the engine status is copied from FlightGear to JSBSim in a loop.

bcoconni, 2018-07-13:
> But is the part marked with the "XXX" comment going to be reliable (as in, won't skip over some index by accident or confuse the <switch>es into picking the wrong signals)

I have not tested your code however, unless there is a bug in JSBSim, it is guaranteed to be executed once every time step and if I understood correctly your code, the cutoff and starter status will be copied every 4 time steps for a given engine.

If that does not work, let us know. As I said above it is supposed to work and if it does not this is likely a bug in JSBSim.

This is a creative usage of JSBSim capabilities; I like that :smiley:

mike402, 2018-07-13:
I see, in this situation making the new behaviour optional woud make it too complex -- and the solution we have now seems good.

Yes, that piece of code works -- I just wanted to know that relying on things being done in the same timestep is not dangerous.

Thank you again!

closed 2018-07-13 by mike402

bcoconni, 2018-07-13:
> things being done in the same timestep is not dangerous.

I am not sure my statement was very clear but the `<switch>` statement you wrote will update only one engine per time step. And since you have 4 engines, this means that a given engine is updated every 4 time steps.

mike402, 2018-07-13:
Yes, even every 5th timestep, because also APU uses it as well -- but that is fine, the systems generating their status have slow execrate, and I intended it.

I am worried about another thing -- if there is any possibility for the output of those <switch> statements to lag behind the output of the <function>, causing the wrong engine to be start up / shut down. As they all are evaluated on the same timestep, I would expect that since <switch>es read the property that the <function> outputs, they would receive its value from the previous step -- but everything is working fine somehow?