JSBSim-Team/jsbsim #85: fcs/{left,right}-brake-cmd-norm used directly for brake output (from jsbsim-devel)

mike402, 2018-07-02:
fcs/{left,right}-brake-cmd-norm drive the brake groups directly. And
they are command properties, written to by FlightGear controls. That
means there is no way to model brake serviceability and brake
regulator/antiskid entirely in JSBSim without relying on external
logic and fragile hacks.

Case in point: in Tu-144 I had to make external aliases
fcs/{left,right}-brake-inp-norm for FlightGear controls, and overwrite
the -cmd- props with the <output> tag of the brake logic's final
function, because it somehow refused to work if I put it directly in
its name="" instead.

On FlightGear mailing list I was suggested to take -brake-cmd-norm as
input in some function and also put <output> of the same function
there. That would probably work, but it is unobvious to whoever might
discover this code later. Moroever it can break if something in the
way properties are handled changes.

bcoconni, 2018-07-07:
@mike402 
This is purely related to the interface of FlightGear with JSBSim. Not much that can be done JSBSim.

Currently FG brake properties are copied to JSBSim's by the C++ interface `src/FDM/JSBSim/JSBSim.cxx`. And I agree there is currently no way to override that behavior.

My suggestion would be to add a property `/fdm/jsbsim/override-fg-brake-properties` and make the copy of FG brake properties depend on it.

bcoconni, 2018-07-07:
I pushed a commit to FlightGear to fix that. Let me know on FlightGear mailing list if that works.

closed 2018-07-07 by bcoconni