JSBSim-Team/jsbsim #86: FlightGear: atmosphere/P-sl-psf stuck.

mike402, 2018-07-02:
Since around FG 2018.1, "atmosphere/P-sl-psf" seems to no longer reflect weather changes.

I had been using it in a hack to keep abrupt jumps in atmo pressure from disturbing my autopilot (as discussed in this thread: https://sourceforge.net/p/flightgear/mailman/message/36299722/), but now I had to throw on more epicycles, see history of the channel "Aero/atmosphere kludge" in this file:

https://gitlab.com/mdanil/Tu-144/blob/master/Tu-144-jsbsim/functions.xml

mike402, 2018-07-05:
Yes, in fact I have been using flightgear -next branch for a while. The property is always "2116.228" and does not change with METAR or manual QNH setting in weather dialog.

It is just one channel, not the whole file, but I should have summarized to make it clear.

Before I could smooth the pressure by doing something like this:

1. Calculate altitude offset (Hsl-abrupt) from atmosphere/P-sl-psf, apply lag_filter.
2. Sum Hsl-abrupt with position/h-sl-meters to produce indicated altitude above sea level (Habs-m)
3. Either use this altitude directly in systems, or calculate the smoothed pressure from it.

Now I have to:
1. Calculate indicated altitude above sea level (Habs-abrupt-m) from "jerky" pressure atmosphere/P-psf.
2. Subtract position/h-sl-meters to get Hsl-abrupt, apply lag_filter.
3. Do the same as in steps 2 and 3 above.

seanmcleod, 2018-07-05:
@mike402 what Jsbsim atmosphere property or properties do the METAR and QNH settings in Flight Gear modify?

mike402, 2018-07-05:
For sure atmosphere/P-psf and atmosphere/pressure-altitude, there may be others. I think they indirectly affect aero/qbar-psf etc as well.

seanmcleod, 2018-07-05:
@mike402 could you confirm, I don't have a copy of FlightGear handy.

If you take a look at the JSBSim commits over the last month or so you'll notice that a lot of them were around updates to the standard atmosphere model.

Now if you take a look at `atmosphere/P-psf` and `atmosphere/pressure-altitude` you'll notice that there are only getters provided by JSBSim, so FlightGear's METAR and QNH weather settings can't be modifying them directly.

```c++
PropertyManager->Tie("atmosphere/P-psf", this, &FGAtmosphere::GetPressure);
PropertyManager->Tie("atmosphere/density-altitude", this, &FGAtmosphere::GetDensityAltitude);
PropertyManager->Tie("atmosphere/pressure-altitude", this, &FGAtmosphere::GetPressureAltitude);
```

In terms of atmosphere properties that can be set in JSBSim:

```c++
  PropertyManager->Tie("atmosphere/delta-T", this, eRankine,
                                    (PMFi)&FGStandardAtmosphere::GetTemperatureBias,
                                    (PMF)&FGStandardAtmosphere::SetTemperatureBias);
  PropertyManager->Tie("atmosphere/SL-graded-delta-T", this, eRankine,
                                    (PMFi)&FGStandardAtmosphere::GetTemperatureDeltaGradient,
                                    (PMF)&FGStandardAtmosphere::SetSLTemperatureGradedDelta);
  PropertyManager->Tie("atmosphere/P-sl-psf", this, ePSF,
                                   (PMFi)&FGStandardAtmosphere::GetPressureSL,
                                   (PMF)&FGStandardAtmosphere::SetPressureSL);
```

bcoconni, 2018-07-05:
@mike402 
You are entirely right. FlightGear sends the sea level pressure to JSBSim via `Atmosphere->SetPressureSL` but `SLpressure` is not updated in the process so `Atmosphere->GetPressureSL` does not report the correct pressure.

Bug fixed by commit 12808cc8. The fix has also been pushed in FlightGear.

mike402, 2018-07-06:
It works! \o/ Thanks a lot!

bcoconni, 2018-07-06:
@mike402 
> It works! \o/ Thanks a lot!

Thanks for the bug report

closed 2018-07-06 by bcoconni