JSBSim-Team/jsbsim #87: Mention LD_LIBRARY_PATH in build docs

rebcabin, 2018-07-04:
For those who have not set up their Cython LD_LIBRARY_PATH but otherwise follow the directions on the build page, all the tests will mysteriously fail with missing module 'jsbsim' etc. I fixed this on my machine with a prefix environment variable such as

    LD_LIBRARY_PATH="/home/xxxxxx/Documents/GitHub/jsbsim-code/build/src/" ctest -j6

or 

    LD_LIBRARY_PATH="/usr/local/lib/" ctest -j6 

after I had done `make install`. Of course, all the tests then passed!

It took me quite a bit of googling and messing around to figure this out, so I thought it might be helpful to save others the trouble by mentioning an appropriate fix in the build doc.

bcoconni, 2018-07-05:
@rebcabin 
I guess you compiled JSBSim with `-DBUILD_SHARED_LIBS=ON` ? Indeed in that case, `LD_LIBRARY_PATH` needs to be specified.

rebcabin, 2018-07-06:
Must have been. I tried a few more build options and can't repro the issue.

closed 2018-07-06 by rebcabin

bcoconni, 2018-07-07:
@rebcabin 
I have updated the docs as per your request. See the new [FAQ section](https://github.com/JSBSim-Team/jsbsim#frequently-asked-questions).

Thanks for the bug report.