JSBSim-Team/jsbsim #88: The tests folder has GPL version 3 files (worth a comment in README?)

rebcabin, 2018-07-04:
Whilst JSBSim itself is LGPL 2.1, the python files in the tests folder are marked "version 3." Because these versions are quite different, it means that some developers will have to remove these files to avoid license issues. It might be worth a comment about this in the License section of the top-level README to help people avoid license accidents.

agodemar, 2018-07-04:
@rebcabin thanks for pointing it out. Would you please summarize the differences between the two LGPL versions?

seanmcleod, 2018-07-05:
Pretty much all the JSBSim tests were created by @bcoconni so we need to ask him why he specifically chose GPL V3 for them as opposed to the LGPL 2.1 used for the rest of JSBSim.

I'm not sure that anyone would need to remove them. It just means that if a developer modifies the test files then those modifications will need to be made available as per the GPL V3 license? And unlike the LGPL the test files couldn't be 'linked' with a non-free program.

bcoconni, 2018-07-05:
@rebcabin 
The license LGPL 2.1 is only meant to be applied to libraries. So it does not make much sense to license the test programs under the LGPL. 

Now we could argue about which version of the GPL the test programs could be licensed but as @seanmcleod said, why would anyone need to release these tests under another license ? They are just tests...

bcoconni, 2018-07-22:
I guess the plan is to keep the current licenses since there was no strong argument in favor of another option.

closed 2018-07-22 by bcoconni