JSBSim-Team/jsbsim #89: Pitch oscillations at rest, Brakes "ON", with pure crosswind

dany93, 2018-07-05:
I've discovered a weird behavior, for which unfortunately I (currently) have no solution to propose. I'm afraid it will not be easy.

Conditions:
- Aircraft at rest on the runway, **brakes (or parking brake) "ON"**,
- **Pure crosswind** (bearing close to + / - 90 deg from the nose, from left or right). 3kt may be enough, 10 - 15 kt worsens the effect.

Observation: cmalpha oscillates, instead of being positive steady for a front wind, or negative but steady for a rear wind. The values of cmalpha do not seem significant, nothing happens if they are steady.

Aircraft tested: (engine stopped)
C172P,
SenecaII PA34,
DR400-JSBSim,
Boeing 707 Lake of Constance.

With engine stopped, for several aircraft with propeller or a jet aircraft ==> the p-factor is not the cause.

Dany

dany93, 2018-07-05:
Yes, depending on the wind velocity, it appears to "jump" on the front gear. Or the front gear remains lifted above the ground for a few seconds prior to fall down again. The period may depend on the gear model (stiffness, damping) but, if the front gear remains lifted above the ground for several seconds you can say that an efficient moment (presumably due to wind) is applied. It cannot be solved only by changing the gear model (and every tested gear model cannot be bad).

dany93, 2018-07-09:
I did an ouput `<moments`> logging.
C172P at rest, brakes ON, engine stopped.
12 kt crosswind from left.
The nose oscillates, the nose wheel lifts up above the ground.

I don't know what it means, but the M moments are very high (which is not surprising, but the hardest is probably to know what it comes from).

L_{Aero} (ft-lbs) | M_{Aero} (ft-lbs)
-- | --
500 to 1600|up to +/- 15000 to 70000

`L_{Aero MRC} (ft-lbs)` and `M_{Aero MRC} (ft-lbs)`: very close to the above values.

In case that would help...

bcoconni, 2018-07-11:
@dany93 
> I guess that `M_{Aero} (ft-lbs)` corresponds to `fdm/jsbsim/moments/m-aero-lbsft`.

Correct.
> Also, `pitch-stab-aero-lbsft` fluctuates up to high values (possibly a consequence, not a cause, of the pitch oscillations).

Interesting. I also noticed on my side that `aero/alphadot-rad_sec` seems not to be correct (not being negative when `aero/alpha-rad` decreases...

FYI I can reproduce the problem on FlightGear but not on JSBSim stand alone. Indeed the cross wind seems not to be the only factor to trigger the pitch oscillation.

Still investigating...

dany93, 2018-07-12:
@bcoconni wrote,
>  I also noticed on my side that aero/alphadot-rad_sec seems not to be correct (not being negative when aero/alpha-rad decreases...

On your side?
I find that it is correct. Unless I've not well done or not well understood something...
[image]

bcoconni, 2018-07-13:
Below is what I got from FlightGear. Notice that `alphadot-rad_sec` is always positive even `alpha-rad` decreases which is obviously wrong.

[image]

agodemar, 2018-07-13:
@bcoconni mind also the values of `alpha-rad`. They reach pretty high values.

dany93, 2018-07-14:
I confirm.
I got the same type of graph, same range and same sign correlation for the values,
- with FlightGear  07fac8c Thu, 5 Jul 2018,
- with (at the end of my c172p.xml test FDM)
```
    <output name="/home/dany/Bureau/alpha.csv" type="CSV" rate="5">
        <property>/fdm/jsbsim/aero/alpha-rad</property>
        <property>/fdm/jsbsim/aero/alphadot-rad_sec</property>
    </output>
```
_(Your recording `rate` is much higher than mine)_

bcoconni, 2018-07-15:
@agodemar 
> mind also the values of `alpha-rad`. They reach pretty high values.

Indeed but since we are examining pure cross wind conditions (β = 90°) I tend to consider alpha values rather meaningless.

bcoconni, 2018-07-15:
I just pushed a script `c172_cross_wind.xml` to JSBSim that reproduces the described behavior. Actually the situation is even worse in the script since the c172 oscillates like a [drinking bird](https://en.wikipedia.org/wiki/Drinking_bird) :smile:

As you can see below the c172 oscillates with a period of approximately 8s. It tilts rearward until the tail touches the ground and bounces back to the horizontal position then the process restarts.

[image]

What is interesting is that nothing happens before t > 5s. It took some time before I figured out what was happening at that moment. Below is the compared evolution of the pitch angle `theta` and of the squared sum of `velocities/u-aero-fps` and `velocities/w-aero-fps`

[image]

It is obvious that the high amplitudes pitch oscillations starts when `u^2+v^2` becomes higher than `0.001`. Why is this important ? Because of the code that computes `alphadot` ([FGAuxiliary.cpp line 170](https://github.com/JSBSim-Team/jsbsim/blob/af2b9db8657013fe535b8f3965b8c4fcb5feda2c/src/models/FGAuxiliary.cpp#L170)) where `mUW` contains the squared sum of `u-aero-fps` and `w-aero-fps`
```c++
    if ( mUW >= 0.001 ) {
      double Vtdot = (vAeroUVW(eU)*in.vUVWdot(eU) + vAeroUVW(eV)*in.vUVWdot(eV) + vAeroUVW(eW)*in.vUVWdot(eW))/Vt;
      adot = (vAeroUVW(eU)*in.vUVWdot(eW) - vAeroUVW(eW)*in.vUVWdot(eU))/mUW;
      bdot = (in.vUVWdot(eV)*Vt - vAeroUVW(eV)*Vtdot)/(Vt*sqrt(mUW));
    }
```
While `mUW` is lower than 0.001, `alphadot` is not updated and is equal to zero. So `mUW` builds up as the cross wind amplitude increases until it reaches the threshold of 0.001 beyond which `alphadot` is computed.

Obviously the value of `alphadot` seems to be erratic in such conditions and it results in high values of `Cmadot`.
[image]

It is possible that the computation of `alphadot` needs some fixes but in the meantime I would suggest to make the computation of the coefficient `Cmadot` less sensitive to the pure cross wind component with a change such that
```diff
@@ -1106,7 +1106,8 @@
             <function name="aero/coefficient/Cmadot">
                 <description>Pitch moment due to alpha rate</description>
                 <product>
-                    <property>aero/qbar-area</property>
+                    <property>aero/qbarUW-psf</property>
+                    <property>metrics/Sw-sqft</property>
                     <property>metrics/cbarw-ft</property>
                     <property>aero/ci2vel</property>
                     <property>aero/alphadot-rad_sec</property>
```
The idea is to drop the spanwise velocity `v-aero` in the dynamic pressure since it does not contribute to the lift and hence to `Cmadot`.

After that modification, `Cmadot` stay within reasonable ranges although `alphadot` still goes crazy...
[image]
and the pitch oscillation is cancelled
[image]

seanmcleod, 2018-07-16:
@bcoconni good analysis.

I'm guessing that most coefficients like Cm-alphadot are calculated and/or measured in wind tunnels assuming very small beta angles that are typically found in flight? Unless the model has specifically been tested in the wind tunnel with large beta values for use in for example modeling spins? In which case the M-alphadot will then more than likely be table based with a beta column as opposed to the single Cm-alphadot coefficient like we have in the C-172 example?

So in this sort of case (single Cm-alphadot coefficient with no beta lookup) we have a Cm-alphadot based on/valid for small beta values that is now seeing a very large beta in this non-spinning ground environment with very small wind velocities generating very large alpha and alpha-dot values.

Ideally we don't want every FDM to have to 'protect' itself against this sort of low velocity extreme alpha and alpha-dot scenario.

seanmcleod, 2018-07-16:
To get a more detailed feel for the `alphadot` values being calculated I logged the following values for the first 14 simulation frames (120Hz) after `mUW >= 0.001`.

[image]

I wonder if the spring response from the gear isn't also playing a large role here?

dany93, 2018-07-17:
@seanmcleod wrote

> I wonder if the spring response from the gear isn't also playing a large role here?

I don't remember by how much, but yes.
However, as I wrote in my [first message](https://github.com/JSBSim-Team/jsbsim/issues/89#issue-338615597), I had this behavior on every four aircraft that I tested.

Pitch moment coefficients:
> single Cm-alphadot coefficient with no beta lookup....
 we don't want every FDM to have to 'protect' itself against this sort of low velocity extreme alpha and alpha-dot scenario

Also, I still have these oscillations at 3 or 5 kt crosswind from right, even by canceling  (in c172p.xml)

 (setting `<value>0</value>`)
```
            <function name="aero/coefficient/Cmalpha">
                <description>Pitch_moment_due_to_alpha</description>

            <function name="aero/coefficient/Cmq">
                <description>Pitch_moment_due_to_pitch_rate</description>

            <function name="aero/coefficient/Cmadot">
                <description>Pitch_moment_due_to_alpha_rate</description>

```

seanmcleod, 2018-07-17:
@dany93 I only really looked in detail at @bcoconni's post, yours was 12 days ago :wink: Just re-read your first post and some of the follow-up posts.

I was trying to understand (looking at the data at 120Hz) what could be generating `alpha-dot` values of over 1000 deg/s across a handful of simulation frames given there is no real change in the wind. Hence my guess it may have something to do with the gear.

I'll take a closer look by logging the forces and moments at 120Hz to see if I spot anything obvious.

Just so we're all looking at the exact same setup I see that @bcoconni's script that he added to JSBSim - `C172_cross_wind.xml` uses the `C172x` in JSBSim. Whereas you mentioned the `C172p` and it's not clear if you're refering to the `C172p` in the JSBSim distribution or the one from the C172p team.

dany93, 2018-07-17:
@seanmcleod I use the c172p from the C172P team.
I even didn't know that there was a C172x in JSBSim (I'm at best an aircraft developer, not a code developer).

But as every aircraft that I tested did this, you should see something.
However, I never got the values reported by @bcoconni (my log is at 5 Hz). And I always found that the alphadot are consistent with the alpha variations.
[image]

bcoconni, 2018-07-17:
@seanmcleod 
> I see that @bcoconni's script that he added to JSBSim - `C172_cross_wind.xml` uses the `C172x` in JSBSim. Whereas you mentioned the `C172p` and it's not clear if you're refering to the `C172p` in the JSBSim distribution or the one from the C172p team.

You can get the same results for the `c172p` from JSBSim. The curves are slightly different than those obtained with the `c172x` but the principle and the conclusion remain the same.

> what could be generating `alpha-dot` values of over 1000 deg/s across a handful of simulation frames given there is no real change in the wind. 

My guess is that high `alpha-dot` values generates high `Cmadot` values which modifies the pitch angle `theta`. Since the wind direction is fixed in the Earth frame, the modification of `theta` modifies `alpha` and therefore `alpha-dot`. As a consequence there is a coupling between `alpha-dot` and `theta`.

In addition, I am pretty sure that the computation of `alpha-dot` is incomplete in that its formula is based on the assumption that there is no wind. At the moment, I think it should use a modified value of the acceleration
```c++
FGColumVector3 vAeroUVWdot = in.vUWVdot-in.vPQR*(Tl2b*in.vTotalNEDWind)
```
However this modification does not fix the problem so there is something else at stake.

@dany93 
> I still have these oscillations at 3 or 5 kt crosswind from right, even by canceling (in c172p.xml)

Right, could you please show the time evolution of the following parameters when `Cmadot`, `Cmalpha` and `Cmq` are canceled out ?
* Pitch angle `theta`
* Aero angles `alpha` and `beta`
* Pitch moments `M_{Aero}`, `M_{Prop}`, `M_{Gear}`, `M_{ext}` and `M_{Total}` ?

seanmcleod, 2018-07-17:
@bcoconni I follow your logic/reasoning regarding `Cmadot` changing `theta`, but looking at the slope of your `theta` plot just after the 5s mark I don't see how that sort of pitch rate would help generate `alpha-dot` values of more than 1000 deg/s that I was seeing and the oscillations during the same first 14 simulation frames.

dany93, 2018-07-18:
@bcoconni wrote

> Right, could you please show the time evolution of the following parameters when Cmadot, Cmalpha and Cmq are canceled out ?

[moments-alpha-beta(7).csv.zip](https://github.com/JSBSim-Team/jsbsim/files/2205216/moments-alpha-beta.7.csv.zip)
[moments-alpha-beta(7).ods.zip](https://github.com/JSBSim-Team/jsbsim/files/2205217/moments-alpha-beta.7.ods.zip)

I've also sent the original csv.
You will probably like zooming the plots.

This time, alpha reaches high values, like you reported. In fact, it is very sensitive to beta, the beta range for this seems very sharp (less than 1 deg).
Beta never goes above 90 deg, even with a crosswind slightly from back (more than 90 deg).

I will be less and less available in the time to come. Will resume later.

seanmcleod, 2018-07-18:
@dany93 so the pitch oscillations are much smaller now compared to what you see with @bocconni's example, which we would expect with Cmadot, Cmalpha and Cmq effectively removed. It looks like the maximum deflection in `theta` is on the order of about 3 degrees now?

When I was glancing through the various C-172 FDMs I noticed that there is also a Cladot, so if adot is large then we'll see some delta lift and if the aeroRP isn't exactly at cg then that will also result in a pitching moment, which could explain some of the small pitching movements you still see. Plus a bit of lift even from Clalpha (also increased slightly by the ground effect factor), again adding to the moment.

I don't think I spotted any betadot terms, but anything with beta, e.g. roll moment due to beta will also generate a moment, and maybe not insignificantly with beta close to 90 degrees even with a small qbar.

Which probably explains the moment you see in L_{Aero} which builds up to around -300 ft-lbs and then oscillates between that and around -50 ft-lbs as the plane rolls/compresses the gear's spring on the one side, springs back a bit etc. So if you log the roll angle I'm sure you'll see small oscillations in that in addition to theta.

bcoconni, 2018-07-22:
@seanmcleod 
> I noticed that there is also a Cladot, so if adot is large then we'll see some delta lift and if the aeroRP isn't exactly at cg then that will also result in a pitching moment,

Good point. `alphadot` is driven by `udot` among other parameters and `udot` depends on forces along the body X axis among which `CLadot`. So I checked the forces along X for the script `c172_cross_wind.xml` (and no modifications to `aircraft/c172x/c172x.xml`)
[image]
Unsurprisingly the total force oscillates due to oscillations of the aerodynamics and gear forces. Now if I cancel out the oscillations of `CLadot` with the following modification to `c172x.xml`
```diff
@@ -635,7 +635,8 @@
             <function name="aero/coefficient/CLadot">
                 <description>Lift due to alpha rate</description>
                 <product>
-                    <property>aero/qbar-area</property>
+                    <property>aero/qbarUW-psf</property>
+                    <property>metrics/Sw-sqft</property>
                     <property>aero/alphadot-rad_sec</property>
                     <property>aero/ci2vel</property>
                     <value>1.7</value>
```
We now get the following forces along X
[image]
What is interesting here is that the total force is still oscillating due to the gear forces oscillations only.
> I wonder if the spring response from the gear isn't also playing a large role here?

Indeed this is another good point. The landing forces along X oscillates even though the other forces do not.

These oscillations are not due to oscillations of gears compressing forces
[image]

But the friction forces are oscillating both in the roll and side directions
[image]
[image]
So it seems that the culprit is the computation of the landing gears friction forces which are probably overshooting and thus generating these oscillations. Furthermore this is consistent with the condition *Brakes "ON"* that @dany93 reported.

I'll investigate in that direction.

bcoconni, 2019-03-17:
@dany93 
I have further investigated about the dynamic pressure modification that I was suggesting earlier in this thread:
```diff
@@ -635,7 +635,8 @@
             <function name="aero/coefficient/CLadot">
                 <description>Lift due to alpha rate</description>
                 <product>
-                    <property>aero/qbar-area</property>
+                    <property>aero/qbarUW-psf</property>
+                    <property>metrics/Sw-sqft</property>
                     <property>aero/alphadot-rad_sec</property>
                     <property>aero/ci2vel</property>
                     <value>1.7</value>
```
I have computed the lift coefficient of a wing for different values of `beta` using [XFLR5](https://sourceforge.net/projects/xflr5/) 3D panels method.

[image]

Below is the plot of the lift coefficient `CL` computed by XFLR5 vs `beta` for angles ranging from 0° to 85° and a constant angle of attack `alpha=2 deg`
[image]
The green curve corresponds to what is currently implemented in the c172 FDM: the lift is only a function of `alpha` and `qbar`. In that case `CL` does not depend on `beta` and the green curve is an horizontal line.

However the computations performed by XFLR5 (the blue curve) clearly show that CL is also a function of `beta`. This is no surprise since according to the Kutta-Joukowski theorem the lift is obtained by the cross product of the velocity `V` and of the bound vortex with a circulation `Gamma`.
```
  L = rho*V x Gamma
```
[image]
As a consequence, the velocity component parallel to the bound vortex (or spinning cylinder in the illustration above) does not produce lift.
The same is true for a wing that can be modeled by a series of vortices/spinning cylinders (see the Wikipedia article on [Prandtl lifting line theory](https://en.wikipedia.org/wiki/Lifting-line_theory))
[image]
Given that the c172p has an unswept wing, the bound vortices of its wing are parallel to the `y` direction hence the `Vy` component of the velocity does not produce lift.

Therefore the lift produced by the c172p wing (or any unswept wing) is
```
  L = 0.5*rho*(U^2+W^2)*S*CL
```
The orange curve represents the correction of the lift coefficient with the ratio `qbarUW/qbar` i.e.
```
   CL_corrected = CL0*(U^2+W^2)/(U^2+V^2+W^2)
```
Although it does not perfectly fit the 3D panels curve, it is a much better match than the current implementation of the lift computation in the c172p.

This correction can easily be obtained by replacing the property `qbar-area` by the product of `qbarUW` and the wing surface `Sw`
```diff
                 <product>
-                    <property>aero/qbar-area</property>
+                    <property>aero/qbarUW-psf</property>
+                    <property>metrics/Sw-sqft</property>
```

The c172p is therefore overestimating the lift when there is cross wind. Moreover it is obvious that for pure cross wind the overestimation is of several order of magnitudes since `qbar/qbarUW=1.0/cos(beta)**2` which tends to infinity when `beta` is near 90°. 

The same reasoning is applicable for moments and the lift as a function of `alphadot`.

---
Going back to the topic of this issue, my point is that the c172p model (and most likely the other models mentioned in the original post) severely overestimates the effect of cross wind by taking into account the `Vy` component of the velocity in the computation of the wing and elevator aerodynamics forces.

In addition, the computations made by JSBSim to calculate `alpha` and `alphadot` when `beta` is very close to 90° are unreliable. Despite my investigations I have not been able to improve the current status. However my conclusions on the effect of `beta` on the aircraft aerodynamics have convinced me that the improvement of `alpha` and `alphadot` computations is irrelevant for `beta` close to 90°.

Finally, following my post from July 22, 2018, I also investigated about the effect of gear friction on the aircraft oscillations due to pure cross wind. I have obtained some very interesting results in that the amplitude of oscillations could be divided by 10 only by modifying the code of ground reactions. It remained that the only means to completely remove the oscillations was to correct the lift computation by `qbarUW` as I exposed above. The improved gear reactions code also have unexpected side effects in other conditions so I have to investigate further before committing.

---
In order to make sure that I was manipulating correctly XFLR5, I have coded my own Vortex Lattice Method (VLM) to run the same computations than XFLR5 albeit with a simplified model.
[image]
As illustrated above, my VLM code gave the same trend than XFLR5.

dany93, 2019-03-20:
@bcoconni wrote
> The c172p is therefore overestimating the lift when there is cross wind. 

> In addition, the computations made by JSBSim to calculate alpha and alphadot when beta is very close to 90° are unreliable. Despite my investigations I have not been able to improve the current [...]

Thank you, Bertrand, for this detailed response.
I haven't yet tested your solution but I'm convinced.

For beta = 90 °(pure crosswind), qbar remains unchanged but there is no lift at all (at least in the simple case of wings with no dihedral angle and no wing sweep). Or it would necessitate taking into account the angle between the cross wind and the aircraft horizontal symmetry plane. Uselessly complicated for what we are looking for.
Moreover, beta = 90° is a singularity because alpha cannot be defined. As alpha is the relevant parameter I see no rigorous solution.

dany93, 2019-03-24:
It works, thank you @bcoconni.

It was even more subtle than I believed: you cancelled the V-component contribution to qbar only for the effects due to alpha-rate...

seanmcleod, 2019-03-24:
Not sure that @bcoconni was specifically looking for the minimum amount of change in terms of which forces and moments need to use the modified qbar to fix your example issue. I think the alpha-rate was simply a single example/test. 

In general as he mentioned it applies to a lot more forces and moments.

> severely overestimates the effect of cross wind by taking into account the Vy component of the velocity in the computation of the **wing and elevator aerodynamics forces**.

bcoconni, 2019-04-07:
@dany93 
> For beta = 90 °(pure crosswind), qbar remains unchanged but there is no lift at all (at least in the simple case of wings with no dihedral angle and no wing sweep).

That's correct although I would anticipate the dihedral effect to be a second order contributor for the matter. Wing sweep will indeed be a major contributor but this needs further investigation to estimate how much and how this could be taken into account. We should obviously anticipate that the effect also depends on the sweep angle.

Tools like XFLR5 are definitely useful while trying to account for that effect.

bcoconni, 2019-08-15:
@dany93 Considering the discussion in c172p-team/c172p#1297, can we close this issue ?

dany93, 2019-08-15:
@bcoconni 
Yes, I close it (as you seem agree).

Thank you.

closed 2019-08-15 by dany93

dany93, 2019-08-16:
As alpha is undefined with pure crosswind (beta = 90 deg) I'm wondering if the strong pitch moment could not be due to large and rapid alpha fluctuations in crosswind conditions (particularly visible even with moderate turbulence).

Otherwise (your graph above) CL is constant between 0 and 85 deg with qbar-psf (which is wrong, I agree) but it does not diverge.

bcoconni, 2019-08-17:
> As alpha is undefined with pure crosswind (beta = 90 deg) I'm wondering if the strong pitch moment could not be due to large and rapid alpha fluctuations in crosswind conditions (particularly visible even with moderate turbulence).

You are right. The actual problem is, as you mentioned, that `alpha` is undefined when `beta` is equal to 90 degrees. This is a problem which is similar to the [Gimbal lock](https://en.wikipedia.org/wiki/Gimbal_lock) issue: Euler angles (`phi`, `theta`, `psi`) or (`alpha`, `beta`, `gamma`) always lead to that limitation.This is the very reason why JSBSim (and other FDMs, and game physics engines) use [quaternions](https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation) to compute rotations. Unfortunately CL and CD are seldomly expressed as functions of quaternions components, so JSBSim has to reconstruct Euler angles from quaternions to provide them as inputs to the modeller.

In addition, when `beta` is close to 90 degrees even moderate variations of the wind directions can lead to high magnitude changes of `alpha`. See for instance the 2 examples in the table below:

Wind velocity (X, Y, Z) | alpha (deg) | beta (deg)
-|-|-
(0.0, 0.99999, 1E-5) | 90.0 | 89.9999
(1E-5, 0.99999,  0.0) | 0.0 | 89.9999

Although the wind vector only slightly changes in direction (X and Z components are only modified by +/-1E-5), `alpha` greatly varies between 0 and 90 degrees. And if that change happens in 1 second then `alphadot` = 90deg/s. Such high magnitudes variations of `alpha` and `alphadot` are just an illusion because the wind direction is actually almost constant.

dany93, 2019-08-18:
Hi @bcoconni 

I've tried to add a few complements and to fix what I think were a few mistakes in the [JSBSim Aerodynamics](http://wiki.flightgear.org/JSBSim_Aerodynamics) Wiki.

However, I'd prefer if you could check them if possible.

They are:

> QBarUW Use this value for formula involving Alpha. It is the value of QBar **restricted to the vertical plane airspeed components.**
> QBarUV Use this value for formula involving Beta. It is the value of QBar **restricted to the horizontal plane airspeed components.**
> 
> Pitch: Cm
> Using **QBarUW** is more appropriate for Cm than the generic QBar.
> 
> Yaw: Cn
> **QBarUV** is more appropriate for Cn than the generic QBar.
> 
> Lift: CL
> Using **QBarUW** is more appropriate for CL than the generic QBar.

Thank you.

dany93, 2019-08-21:
At the end of Lift: CL and Pitch: Cm as in my previous post, should I add:

> Otherwise, your aircraft will likely be prone to the issue **Pitch oscillations at rest, Brakes ON, with pure crosswind** (link)

??

bcoconni, 2019-08-21:
@dany93 all this makes sense to me. However this has only been tested for the particular case of the c172 so I would make clear that these are not universal laws but rather guidance or at the very least trouble shooting.

And for the sake of completeness, as you mentioned previously in that discussion, swept wings might need a different treatment that we have not investigated further.

dany93, 2019-08-22:
[Done into the Wiki](http://wiki.flightgear.org/JSBSim_Aerodynamics) (correctly, I hope).

bcoconni, 2019-08-22:
> Done into the Wiki (correctly, I hope).

Excellent ! I'm glad this is now documented.
Thank you.

seanmcleod70, 2025-06-11:
Note @bcoconni's last comment in the thread above:

> However this has only been tested for the particular case of the c172 so I would make clear that these are not universal laws but rather guidance or at the very least trouble shooting.

> And for the sake of completeness, as you mentioned previously in that discussion, swept wings might need a different treatment that we have not investigated further.

So what issue are you seeing/are you trying to address, and what is the context?

Octal450, 2025-06-11:
Same issue. Under the winds, the plane shake up and down in pitch when I increased the alphadot coefficient (to correct in-flight behavior) but only when the plane is stopped and alpha is fluctuating heavily (because the alphadot is going crazy).

Apparently, qbarUW works to fix that so for now, I have to do that.
It's close enough to qbar anyway in most conditions I tested with so, fine enough.

https://forum.flightgear.org/viewtopic.php?f=25&t=43383&p=432680#p432680

Kind Regards,
Josh