JSBSim-Team/jsbsim #91: Output socket does not respect debug level

Gor-Ren, 2018-07-11:
The `set_debug_level(level)` method is intended to allow control of JSBSim output. Setting to `level=0` "...requests JSBSim not to output any messages whatsoever" ([FDFDMExec docs](https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGFDMExec.html)).

When using an output socket for visualising JSBSim with FlightGear several messages print to console even with `set_debug_level(0)`. Here is a minimal example using the Python API:

```python
import jsbsim
import os

# init JSBSim and load basic config
root_dir = "/your/root/dir"
fdm = jsbsim.FGFDMExec(root_dir=root_dir)
fdm.set_debug_level(0)
fdm.load_model('c172x')
fdm.load_ic('reset00', useStoredPath=True)

# set socket output using preconfigured flightgear.xml file
output_file = "data_output/flightgear.xml"
output_config_path = os.path.join(root_dir, output_file)
fdm.set_output_directive(output_config_path)

fdm.run_ic()
```

This prints two distinct messages:

>Creating UDP socket on port 5550
>Successfully connected to socket for output ...

>This vehicle has 6 bogeys, but the current 
>version of FlightGear's FGNetFDM only supports 3 bogeys.
>Only the first 3 bogeys will be used.

Any reset of the simulation to initial conditions (e.g. `fdm.reset_to_initial_conditions(0)`) will cause a repeat of these messages. The first message appears to originate from [FGfdmSocket](https://github.com/JSBSim-Team/jsbsim/blob/master/src/input_output/FGfdmSocket.cpp) and the second from [FGOutputFG](https://github.com/JSBSim-Team/jsbsim/blob/master/src/input_output/FGOutputFG). I regret I don't know C++.

In my use case I am running many trials in JSBSim, so the above behaviour results in my console becoming unusable as it is spammed with tens of thousands of lines!

It would be great if these outputs and related messages could be suppressed, for example by making them check the debug level before printing. Perhaps more generally, it would be nice if there was a way suppress all JSBSim output entirely from the Python API (since even `set_debug_level(0)` will still miss some JSBSim launching text and various warnings).

bcoconni, 2018-07-11:
@Gor-Ren 
Thanks for reporting these issues. I will investigate that later this week.

> Perhaps more generally, it would be nice if there was a way suppress all JSBSim output entirely from the Python API (since even set_debug_level(0) will still miss some JSBSim launching text and various warnings).

In the meantime, you should be able to remove the launching text by setting the following environment variable before creating FGFDMExec instances
```python
import os
os.environ["JSBSIM_DEBUG"]=str(0)
```

Gor-Ren, 2018-07-12:
@bcoconni 
Thanks!

I have noticed that `do_trim()` operations also have the same issue, i.e. they output a trim report to console even with debug set to zero. From looking at the source code in [FGTrim](https://github.com/JSBSim-Team/jsbsim/blob/master/src/initialization/FGTrim.cpp), it appears there are some checks on the debug level before reporting so I think there must be a bug in the logic.

bcoconni, 2018-07-15:
@Gor-Ren 
> I have noticed that do_trim() operations also have the same issue, i.e. they output a trim report to console even with debug set to zero.

Just pushed the commit b82c80f which should fix that problem.

bcoconni, 2018-07-15:
The commit 435c17a should remove the message about extra bogeys reported by `FGOutputFG`.

bcoconni, 2018-07-15:
And finally commit 350c1f8 should remove the messages about the UDP socket creation.

Gor-Ren, 2018-07-16:
Many thanks for these fixes - I have updated JSBSim and can confirm they are working.

bcoconni, 2018-07-16:
@Gor-Ren 
> I have updated JSBSim and can confirm they are working.

Great !

closed 2018-07-16 by bcoconni