JSBSim-Team/jsbsim #94: <fcs_function> and <function>: suggestions for new functions

mike402, 2018-07-15:
Hi,

I have noticed that the following functions would be very useful to have for system development, especially navigation system:

- floatng point equivalent of <mod>. For now I have snippets like this to make up for it:

    <product>
     <fraction>
      <product>
       <p>example/input</p>
       <v>0.5</v>
      <product>
     </fraction>
     <v>2.0</v>
    </product>

- sawtooth from -180 to +180 degrees like geo.normdeg180() from FlightGear: http://wiki.flightgear.org/Nasal_library/geo . Without it I ended up with this black magic in many places:

    <difference>
     <product>
      <fraction>
       <product>
        <sum>
         <p>example/input</p>
         <v>180.0</v>
         <product>
          <lt>
           <p>example/input</p>
           <v>0.0</v>
          </lt>
          <v>-360.0</v>
         </product>
        </sum>
        
        <v>0.00277777777777777777</v>
       </product>
      </fraction>
      <v>360.0</v>
     </product>
     <v>180.0</v>
     <product>
      <lt>
       <p>example/input</p>
       <v>0.0</v>
      </lt>
      <v>-360.0</v>
     </product>
    </difference>

- sawtooth from 0 to 360 degrees like FlightGear's geo.normdeg(), which is like floating point modulo by 360, except it should be shifted by 360 for negative input:

    <sum>
     <product>
      <fraction>
       <product>
        <p>example/input</p>
        <v>0.00277777777777777777</v>
       </product>
      </fraction>
      <v>360.0</v>
     </product>
     <product>
      <lt>
       <p>example/input</p>
       <v>0.0</v>
      </lt>
      <v>360.0</v>
     </product>
    </sum>

Kind regards,
Mike

bcoconni, 2018-07-17:
@mike402 
Thanks for this proposal but instead of implementing these functions in C++, I'd rather extend the concept of "template" functions that have been designed for unit conversion (see [scripts/unitconversions.xml](https://github.com/JSBSim-Team/jsbsim/blob/master/scripts/unitconversions.xml)).

Your functions could then be provided in a kind of library just as the unit conversion functions are.

For that to happen, there is some C++ code to write but that is not on the top of my priorities. If someone volunteer for that, I can provide guidance however.

mike402, 2018-08-03:
Thanks, that seems reasonable. Can I try it out right away by inserting the snippets in unitconversions.xml or does it depend on C++ in other way than loading an extra xml that would be the new "library"?

bcoconni, 2018-08-05:
> does it depend on C++ in other way than loading an extra xml that would be the new "library"?

Yes, it needs additional C++ code. At the moment, if you would update `unitconversions.xml` then you would be unable to use the functions anywhere else than in `<output>` elements.

mike402, 2018-10-27:
Also it would be nice to have a "stair" function with adjustable step and transition slope -- which could be used e.g. to represent Geneva drives:

~~~
  <fcs_function name="fcs/NPK/instr/UVO[0]/counter/H-100m[1]">
   <function>
    <!-- Geneva drive -->
    <!-- JSBBUG Moves sharply for negative values, because there is no <floor>, and <integer> is used instead! -->
    <sum>
     <!-- Simple step -->
     <integer>
      <product>
       <!-- lower digit -->
       <p>fcs/NPK/instr/UVO[0]/counter/H-100m[0]</p>
       <!-- = 1 / base -->
       <v>0.1</v>
      </product>
     </integer>
     <!-- Transition -->
     <min>
      <v>1.0</v>
      <max>
       <v>0.0</v>
       <product>
        <difference>
         <product>
          <!-- lower digit -->
          <p>fcs/NPK/instr/UVO[0]/counter/H-100m[0]</p>
          <!-- = 1 / base -->
          <v>0.1</v>
         </product>
         <integer>
          <product>
           <!-- lower digit -->
           <p>fcs/NPK/instr/UVO[0]/counter/H-100m[0]</p>
           <!-- = 1 / base -->
           <v>0.1</v>
          </product>
         </integer>
         <!-- transition length, 0.0 to 1.0 -->
         <v>0.9</v>
        </difference>
        <!-- = 1 / transition length -->
        <v>10.0</v>
       </product>
      </max>
     </min>
    </sum>
   </function>
  </fcs_function>
~~~

Please note that I have used ``<integer>`` instead of the correct  ``<ceiling>``, because it is absent in JSBSim -- which makes it work incorrectly for negative inputs. What is intended with ``<ceiling>`` is, the transition between -2 and -1 of the output should be between -1.0 and -1.1 of the output and such -- just like 1 and 2 transition between 1.9 and 2.0. That is the function should go on the same way into negative domain, instead of being antisymmetric.

bcoconni, 2018-10-27:
From @mike402 suggested in issue #124 that has been marked as a duplicate of the current issue:
> `<floor>`, `<ceiling>` and `<fmod>` would have been very convenient, because their equivalents are very bulky and typo-prone.

bcoconni, 2018-10-28:
@mike402 
> `<floor>`, `<ceiling>` and `<fmod>` would have been very convenient, because their equivalents are very bulky and typo-prone.

These 3 functions have just been added to `<function>` via the commit f0d67c8.

mike402, 2018-10-28:
Wow, thank you, I will try them!

About the requests with sawtooth and stair function: maybe it would be easier to have them in C++, at least for now?