JSBSim-Team/jsbsim #96: Peak power at wrong EGT; possibly wrong power calculation

algefaen, 2018-07-15:
Hi yall, I'm one of the c172p contributors with quite a few hours in real c172's and trying to get it closer to the real thing.

One issue I noted was that the peak power is quite a ways away from where it "ought" to be according to the lycoming o320 / o360 power curves. From [the bug I opened on the c172p project](https://github.com/c172p-team/c172p/issues/1128):

> Hi! I might be entirely wrong, but it seems like our Cessna has a wrong "best power" lean settings compared to the Lycoming manuals. From their charts it looks we should get "max power" at about 130F rich of peak EGT, while my experiments in the sim give me max power at around 250F rich of peak EGT. The sim also seems to be losing power faster than my (anecdotal) real life experience of leaning the C172.

Trying to fully understand the calculations in FGPiston isn't very easy, but I think I may have figured out why this is: The `Mixture_Efficiency_Correlation` table seems to be centered on 12.6, which is where gasoline piston engines get their most power. I _think_ it should have been centered on 14.7, which is where the fuel burns the best. With the old code, I was getting peak power at  ~10 AFR. When trying to crudely just shift the table so that it's centered at 14.7, I'm now getting peak power at ~12AFR (as expected), and the EGT for peak power is indead ~150 degrees F less than it's peak at 14.7 AFR. "Peak economy" also suddenly makes sense, previously flying at 15AFR gave way too little power.

Now there's a lot of calculations and constants in the code that I don't understand, so I might be totally off, and I won't be sad if you tell me I'm wrong. If it's possible to tweak the parameters to our c127p so that it matches real life better with the current code then that's fine. I just can't find those parameters and the simplest way to make it match my understanding better is to center the MEC at 14.7. (Aka 0.068)

I've crudely painted in the power curve I'm getting today (ish) vs what we should be seeing (ish) here:
[image]

This is tested using flightgear next 64f3c1, which is a week old and right after @bcoconni synced jsbsim in.

algefaen, 2018-07-16:
Scratch that, just moving the MEC to be centered at 14.7 moves the power curve down to the right, not just right. So the peak power happens at a more correct temperature but it's producing way too little power. Oh well.

bcoconni, 2018-07-17:
@algefaen 
That's an interesting problem :smiley:
Once issue #89 will be resolved I'll try to investigate this one.

bcoconni, 2024-03-09:
Following discussions in issue #1037:
* The engine power is basically computed from the mixture efficiency correlation, the fuel and the specific fuel consumption:

https://github.com/JSBSim-Team/jsbsim/blob/b9bfc2350e0e00e53246b65979cc18df2922a88d/src/models/propulsion/FGPiston.cpp#L786-L796

So basically we have `power = fuel_flow * ME / ISFC` where `ME` comes from the mixture efficiency correlation and `ISFC` is a constant that is either specified by the user or computed internally by `FGPiston.cpp`.

In FlightGear's c172p model the `<bsfc>` parameter has been commented out:
* https://github.com/c172p-team/c172p/blob/177b702391daa80249a463dbdf27c970ce858d74/Engines/eng_io320.xml#L20
* https://github.com/c172p-team/c172p/blob/177b702391daa80249a463dbdf27c970ce858d74/Engines/eng_io360.xml#L20

So for FlightGear's c172p, the `ISFC` parameter is computed by the following code which uses `MaxHP`:
https://github.com/JSBSim-Team/jsbsim/blob/b9bfc2350e0e00e53246b65979cc18df2922a88d/src/models/propulsion/FGPiston.cpp#L278-L285

On the other hand, if the `<bsfc>` parameter is provided by the engine model such as is the case for `eng_io320.xml`:
https://github.com/JSBSim-Team/jsbsim/blob/b9bfc2350e0e00e53246b65979cc18df2922a88d/engine/eng_io320.xml#L19
then `MaxHP` is ignored. The take away is that `<bsfc>` in `eng_io320.xml` is not set to a value in `eng_io320.xml` that matches the target power of 160 hp.

bcoconni, 2024-03-09:
Also according to the following lines of code:
https://github.com/JSBSim-Team/jsbsim/blob/b9bfc2350e0e00e53246b65979cc18df2922a88d/src/models/propulsion/FGPiston.cpp#L755-L757
and
https://github.com/JSBSim-Team/jsbsim/blob/b9bfc2350e0e00e53246b65979cc18df2922a88d/src/models/propulsion/FGPiston.cpp#L786
The computation of the mixture efficiency correlation can be simplified to the equivalent code:
```c++
ME = Mixture_Efficiency_Correlation->GetValue(1.3 * in.MixturePos[EngineNumber] * 101325.0 / p_amb / 14.7);
```
Which can be simplified further at sea level where `p_amb == 101325.0` (`FGPiston.cpp` is using SI units):
```c++
ME = Mixture_Efficiency_Correlation->GetValue(1.3 * in.MixturePos[EngineNumber] / 14.7);
```
Said in other words, at sea level, the Fuel to Air Ratio (FAR) is computed by the following equivalent code:
```c++
FAR = 1.3 * in.MixturePos[EngineNumber] / 14.7
```
meaning that the maximum value that can reached if mixture is kept in the [0.0 - 1.0] range is `1.3/14.7=0.088`.

Knowing that the mixture efficiency correlation (MEC) goes up to a FAR of 0.125
https://github.com/JSBSim-Team/jsbsim/blob/b9bfc2350e0e00e53246b65979cc18df2922a88d/src/models/propulsion/FGPiston.cpp#L334-L351
then if we would want to cover the whole range of the MEC range then `in.MixturePos[EngineNumber]` shall be multiplied by `0.125*14.7=1.8375` instead of `1.3`.

Now since `in.MixturePos[EngineNumber]` is already multiplied by 1.3 then `fcs/mixture-pos-norm` shall be set to values up to `1.8375/1.3=1.413`:
```xml
  <channel name="Mixture control">
    <fcs_function>
      <function>
        <product>
          <value>1.413</value> <!-- 1.8375 / 1.3 -->
          <property>fcs/mixture-cmd-norm</property>
        </product>
      </function>
      <output>fcs/mixture-pos-norm</output>
    </fcs_function>
  </channel>
```

hbeni, 2024-09-09:
Is there still some information missing here?