JSBSim-Team/jsbsim #99: Use JSBSim in place of NASAL in Flightgear

abassign, 2018-07-17:
In these months I am working at FIAT G91-R1B in FGFS in collaboration with Massimiliano and the friends and former pilots of the G91 group of FB.

I am currently learning to use JSBSim as deeply as possible while removing all the NASAL code I can delete. The reason is that I would like to have a prevailing code in JSBSim that being a data-driven language fits better in my working method (I often use simulation programs like Open-modelics and I myself worked many years ago in the development of a CSMP interpreter in APL).

Obviously in the Flightgear group the collaboration is very poor as few people have a deeper knowledge of JSBSim, many of them think that JSBsim is very heavy in execution.

I am currently working in the electrical system which is starting to work well, and it seems to me that it clearly demonstrates the advantages of using JSBSim in place of NASAL (I am replacing the NASAL code I had written in the previous months with JSBSim). In fact, using a script language like NASAL (but it would be the same thing with Python and Java) it is necessary a continuous use of events and timers, this makes the system weighed as these processes pass through the operating system and the virtual machine. Moreover, NASAL tasks can not define shared variables as they are all separate processes. This requires using external calls to an object called a "property tree". In JSBSim the situation seems to me much better and so it seems to me natural to write in JSBSim everything related to on-board systems (The G91 was conceived in the mid-50s' and therefore many systems are analog).

Sorry for the introduction a bit 'long, but now I pass to more specific questions:

1#. According to your knowledge of the code JSBSim does an optimization of block processing by performing, for each step, only the blocks that have had at least one input modified with respect to the previous step? (It was on of the most powerful optimization criteria I had developed for the interpreter CSMP in APL)?
2#. There are logical blocks (I have only seen the switch), such as a JK flip-flop, (although these days I saw that it is possible to emulate it with a switch used in sample-hold mode ...), but his presence would simplify the code and improve his reading.
#3. I'm using the useful function execrate = "n" where n is a number of intervals between one execution and another of a channel. But I noticed a boring bug, that is, if there is a time-dependent block inside its value dt is not modified (should be multiplied by the execrate value present in the block) so as to make this function transparent and therefore be able to vary execrate without having to intervene in the code. On the contrary, it would be interesting to have a variable execrate depending on some event, since certain blocks must be executed frequently only in certain phases of the flight.
#4. For the performance of JSBSim I can not complain because it is actually very fast and therefore consumes little CPU, but currently the tasks that I give are not many, I would like to know if someone has analyzed the performance problem with more depth, to understand the better optimization criteria.
#5.In Flightgear JSBSim runs in the same system thread, but being a separate application, with its own data that are exchanged only through a specialized process ... it would be very interesting to be able to run it on a separate thread, in order to use a different CPU . At this point the extended use, I would say aggressive, of JSBSim in Flightgear would be very interesting.

Thank you for your cooperation, greetings to the whole group that manages this wonderful program.

bcoconni, 2018-10-27:
@abassign 
Since you made no further comments, I guess your questions have been addressed and this issue can be closed ?

abassign, 2018-10-31:
I think this request is complete, thank for your interest.

closed 2018-10-31 by agodemar