JSBSim-Team/jsbsim #101: Initialisation of running engines confusing

seanmcleod, 2018-07-22:
I have an aircraft with 2 engines and I wanted both engines to be started by the initialization file. Looking at `FGInitialCondition.h' I saw the following comment:

```c++
// - running (-1 for all engines, 0 for no engines, 1 ... n for specific engines)
```

So I added a `running` element to my initialization file.

```
<running> -1 </running>
```

However neither of the engines are started.

I then also noticed that the code in `FGInitialCondition.cpp` looks for multiple `running` elements, and based on the comment above - `1 … n for specific engines` I tried the following based on the comment regarding 0 and 1.

```
<running> 1 </running>
<running> 2 </running>
```

However that only results in engine 1 being started and engine 0 isn't started. Engines are 0 based, from `FGFDMExec.cpp`

```c++
  for (unsigned int n=0; n < propulsion->GetNumEngines(); ++n) {
    if (IC->IsEngineRunning(n)) {
      try {
        propulsion->InitRunning(n);
      } catch (const string& str) {
```

So looking at the code in `FGInitialCondition.cpp`.

```c++
  // Check to see if any engines are specified to be initialized in a running state
  Element* running_elements = document->FindElement("running");
  while (running_elements) {
    enginesRunning |= 1 << int(running_elements->GetDataAsNumber());
    running_elements = document->FindNextElement("running");
  }
```

I then tried the following which now does start both engines (0 and 1).

```
<running> 0 </running>
<running> 1 </running>
```

So assuming that the engine indices to the `running` elements should be 0 based then the comment in `FGInitialCondition.h` needs to be updated.

```c++
// 0 for no engines, 1 ... n for specific engines)
```

And the code needs to be fixed to properly handle the -1 option.

I'm happy to make the fixes, but before I do I just wanted to double-check whether the engine indices in the `running` elements should be 0 or 1 based.

bcoconni, 2018-07-22:
It is very likely that I am the culprit of this (due to the commit 66ceba9a8d195803f9643fb87c245fcdc16a0c5a that does not handle the case where `-1` is specified). 

Not sure that the case `0` ever worked however. After all, if one does not want the engine started then he/she would not specify any `<running>` element ?

So yes please fix this and sorry about that.

seanmcleod, 2018-07-23:
Doesn't look like the code before your commit handled the -1 case either. I'll submit a pull request to fix the handling of the -1 option and update the comment that engine values go from `0 to n-1`.

bcoconni, 2018-07-26:
I just pushed the PR #103.

closed 2018-07-27 by seanmcleod