JSBSim-Team/jsbsim #102: Judging floating point numbers cannot use constant equal sign

jinzhu1911, 2018-07-23:
In Class FGColumnVector3, in the function of Overload constant equal sign whitch it uses "==" to judge double, I think this is not right.

bcoconni, 2018-07-26:
I'm not sure I'm getting your point: your statement is too vague. Can you please point at a place in the code where you think the usage of `==` is *"not right"* ?

bcoconni, 2018-09-02:
Ping !

@jinzhu1911 It's been one month since I asked you some clarifications but got no answer. Could you please be more specific ?

bcoconni, 2018-09-09:
Last call before ~departure~ closing the issue as **won't fix** :smiley:

jinzhu1911, 2018-09-18:
in the function Comparison operator.
 bool operator==(const FGColumnVector3& b) const {
    return data[0] == b.data[0] && data[1] == b.data[1] && data[2] == b.data[2];
  }

agodemar, 2018-09-18:
@jinzhu1911 Your answer isn't clear. Could you elaborate?
Please, do keep in mind that this is not the place to be cryptic.

jinzhu1911, 2018-09-19:
in the function Comparison operator.
 bool operator==(const FGColumnVector3& b) const {
    return data[0] == b.data[0] && data[1] == b.data[1] && data[2] == b.data[2];
  }

> 
> 
> I'm assuming he means the method should be along the following lines:
> 
> ```c++
> bool operator==(const FGColumnVector3& b) const {
>    return fabs(data[0] - b.data[0]) < 1e-13 && fabs(data[1] - b.data[1]) < 1e-13 && 
>       fabs(data[2] - b.data[2]) < 1e-13;
> }
> ```
Yes, I mean this. Thanks very much!

closed 2018-10-29 by jinzhu1911