JSBSim-Team/jsbsim #110: AppVeyor builds fail because they reach the 60 min limit

agodemar, 2018-08-10:
Currently, our AppVeyor builds fail due to the fact that build execution times reach the maximum allowed time of 60 minutes for a free plan. The long build time is probably due to the Python integration tests. On Travis the build time is far less: 4 different builds in 45 min (including tests), more or less.

I propose, for the moment, to disable Python tests on AppVeyor, to make sure that our builds terminate gracefully with a deploy of the artifacts in our Release area on GitHub.

closed 2018-08-16 by agodemar

bcoconni, 2018-08-19:
Just for information, CMake can run all the tests but `TestInputSocket` with the following command
```bash
> ctest -E TestInputSocket
```

seanmcleod, 2018-08-20:
@bcoconni thanks for the `-E` tip. I've just pushed a merge request and confirmed that it works in terms of excluding `TestInputSocket` on AppVeyor. Once we figure out the issue with the test we can then re-enable the test.