JSBSim-Team/jsbsim #111: Question - Local Cartesian Coordinates

Gor-Ren, 2018-08-14:
I am working on a project which involves flights of short distance, and so I would dearly like to use a simplified Cartesian coordinate system (x,y,z / North,East,Altitude) instead of a spherical geographic one.

I see the properties available in JSBSim include velocities in this local frame: `velocities/v-north-fps`, `velocities/v-east-fps`, `velocities/v-down-fps`.

However, I don't see any equivalent for position in the NED frame. Is this correct?

If there is indeed no NED position available, then I have to do my own conversion, and I have a follow-up question: do the geodetic latitude/longitude properties output by JSBSim conform to "WGS84 - World Geodetic System 1984"? (I think this is the standard geodetic system and the answer is yes... but I'm out on a bit of a limb here). 

I need to know so I can tell the [PROJ4](http://jswhit.github.io/pyproj/) library the projection of my input data, to get it converted to a local frame :)

Many thanks!

seanmcleod, 2018-08-14:
One option would be to use the Earth Centered Earth Fixed (ECEF) frame:

```
position/ecef-x-ft
position/ecef-y-ft
position/ecef-z-ft
```

Take a look at the original JSBSim PDF Reference Manual or the new online version that @agodemar has started - https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/user-manual-frames-of-reference/

Although I'm guessing you're ideally looking for a local point of interest tangent frame rather than an earth centered one.

seanmcleod, 2018-08-14:
Actually on further inspection of the `position` properties you could use the following:

```
  PropertyManager->Tie("position/distance-from-start-lon-mt", this, &FGAuxiliary::GetLongitudeRelativePosition);
  PropertyManager->Tie("position/distance-from-start-lat-mt", this, &FGAuxiliary::GetLatitudeRelativePosition);
```

So a local reference to the initial/starting coordinates of the sim run.

seanmcleod, 2018-08-14:
@Gor-Ren but the following aren't geographic coordinates:

```
position/distance-from-start-lon-mt
position/distance-from-start-lat-mt
```

They are the number of meters `-mt` in the `north - lon` and number of meters in the `east - lat` from your starting/initial coordinates that you specify for your initial conditions. In other words you're getting to specify your "point of interest" via your initial starting location.

For small distances from your starting point they would be equivalent to a local tangent plane.

Gor-Ren, 2018-08-14:
@seanmcleod Aha! Thank you, I completely zoned out when I read `-lat` and `-lon`. This is great.

closed 2018-08-14 by Gor-Ren

agodemar, 2018-08-15:
@Gor-Ren @seanmcleod These two properties
```
position/distance-from-start-lon-mt
position/distance-from-start-lat-mt
```
were introduced following my suggestion several years ago, when I used JSBSim to perform Monte Carlo simulations of take-off trajectories. I needed properties that allowed to calculate *distances* travelled from the starting point on runway. 

So, the property names are to be read as: 

*(ground) distance, in meters, travelled from the initial point (at t=0), along the reference direction of longitude (i.e. East) or latitude (i.e. North)*.

seanmcleod, 2018-08-15:
Out of interest @Gor-Ren what sorts of tests are you performing with JSBSim?

Gor-Ren, 2018-08-16:
@seanmcleod I'm doing a dissertation on the subject of deep reinforcement learning. It's a sub-branch of machine learning that allows (software) agents to learn to control arbitrary tasks, and in recent years it has gone from toy problems of simple complexity to state-of-the-art performance in difficult stuff like video games and robotics control.

I like flight simulation. So, my project has been to wrap JSBSim as a reinforcement learning environment (actions on control surfaces come in, the aircraft's various state variables are sent back to the agent) and train agents to fly planes. I'm targeting a fairly simple control task, flying on a commanded heading while maintaining altitude.

I'll be making my repo public in about a month once I finish up, and should have some videos of how the agents do. Working with JSBSim has been a pleasure; its Python wrapper has been essential for my work, so I'm glad it exists!

seanmcleod, 2018-08-16:
@Gor-Ren sounds interesting and I'd be interested in seeing the results/write-up. I remember coming across some similar work at Stanford a couple of years ago, I think in their case they were using it to control a RC helicopter for aerobatic flight.