JSBSim-Team/jsbsim #113: Discrepancy between MachFromVcalibrated and other published conversions

seanmcleod, 2018-08-25:
I noticed some discrepancies between the output from `FGJSBBase::MachFromVcalibrated()` and other published conversions for fairly high CAS/Mach numbers.

In particular for an ISA standard day at 20,000ft `FGJSBBase::MachFromVcalibrated()` produces higher Mach values from 800kt CAS and upwards compared to the results from [luizmonteiro](http://www.luizmonteiro.com/Altimetry.aspx#MachNumber) and from a USAF F-15 data publication's graph.

| CAS (kt) | JSBSim Mach | Luizmonteiro Mach |
| --- | --- | --- |
| 600 | 1.24 | 1.24 |
| 700 | 1.45 | 1.45 |
| 800 | 1.69 | 1.68 |
| 900 | 1.95 | 1.90 |
| 1000 | 2.24 | 2.13 |
| 1100 | 2.57 | 2.36 |
| 1200 | 2.94 | 2.59 |

And a graph from a USAF publication for F-15 data for Airspeed Conversion - High Mach. 

[image]

The reading for a CAS of 900kt matches the Luizmonteiro Mach number of 1.90 versus JSBSim's 1.95.

As a reference here is the code for `FGJSBBase::MachFromVcalibrated()`

```c++
double FGJSBBase::MachFromVcalibrated(double vcas, double p, double psl, double rhosl)
{
  double pt = p + psl*(pow(1+vcas*vcas*rhosl/(7.0*psl),3.5)-1);

  if (pt/p < 1.89293)
    return sqrt(5.0*(pow(pt/p, 1./3.5) -1)); // Mach < 1
  else {
    // Mach >= 1
    double mach = sqrt(0.77666*pt/p); // Initial guess is based on a quadratic approximation of the Rayleigh formula
    double delta = 1.;
    double target = pt/(166.92158*p);
    int iter = 0;

    // Find the root with Newton-Raphson. Since the differential is never zero,
    // the function is monotonic and has only one root with a multiplicity of one.
    // Convergence is certain.
    while (delta > 1E-5 && iter < 10) {
      double m2 = mach*mach; // Mach^2
      double m6 = m2*m2*m2;  // Mach^6
      delta = mach*m6/pow(7.0*m2-1.0,2.5) - target;
      double diff = 7.0*m6*(2.0*m2-1)/pow(7.0*m2-1.0,3.5); // Never zero when Mach >= 1
      mach -= delta/diff;
      iter++;
    }

    return mach;
  }
}
```

Taking a look at Luizmonteiro's JavaScript this is what he is using (I've converted it to Python).

```python
def CAStoMach(cas, altitude):
    pressure = ISAtmosphere.Pressure(altitude)
    asl = ISAtmosphere.sealevel_speed_of_sound

    if cas < ISAtmosphere.sealevel_speed_of_sound:
        qc = ISAtmosphere.sealevel_pressure * (math.pow(1 + 0.2 * math.pow(cas/asl, 2), 3.5) - 1)
    else:
        qc = ISAtmosphere.sealevel_pressure * (((166.9215801 * math.pow(cas/asl, 7)) / math.pow(7 * math.pow(cas/asl, 2) - 1, 2.5)) - 1)

    mach = math.sqrt(5 * (math.pow(qc/pressure + 1, 2/7) - 1))

    if mach > 1:
        for i in range(7):
            mach = 0.88128485 * math.sqrt((qc/pressure + 1) * math.pow(1 - 1/(7*mach*mach), 2.5))

    return mach
```

Now if I only use the first `qc` calculation, i.e. the one used when the CAS is less than the speed of sound at sea level then I get identical results to the JSBSim implementation.

I haven't found any references so far showing the derivation of the `qc` formulas used by Luizmonteiro to confirm that they are accurate, but they do seem to agree with the USAF graph publication and one or two EM (Energy Maneuverability) diagrams I've seen with CAS and Mach on the x-axis.

So I'm looking for any possible feedback to confirm whether Luizmonteiro's implementation is more accurate over a wider CAS/Mach range than the current JSBSim implementation and therefore whether we should consider updating JSBSim's implementation.

djlinse, 2018-08-27:
Sean,

You've discovered the issue.

If you track down a copy of the US Air Force Aircraft Performance Flight Testing Manual (AFFTC-TIH-99-01) [There are various places it is available on the interweb.], you'll find all of these equations in section 4.6.  In particular, you'll find the equations for qc (the impact pressure) which is the total pressure minus the static pressure.  For subsonic speeds, it is calculated the same in both JSBSim and Luizmonteiro.  But for supersonic cases (when the calibrated airspeed is greater than the sea-level speed of sound (equivalent to pt/p < 1.89293)), it needs to be calculated behind the normal shock created at the pitot-tube, so it uses the Rayleigh Supersonic Pitot Equation, which is the "else" part in Luizmonteiro's calculation.  JSBSim never recalculates pt (i.e. qc + p) for the supersonic case and it needs to.  The equations for qc give identical results at the crossing point, so that is why the subsonic version is calculated for the check, since it is the easier one.

Luizmonteiro's equations follow most directly those presented in AFFTC-TIH-99-01. It uses a simple iteration to solve things.  I *think* the JSBSim implementation (using Newton-Raphson) is correct, but it is a bit "more complicated" in that it calculates the derivative with respect to Mach to converge "faster".

So to fix JSBSim, just add this in the else bracket and try it:

  pt = p + psl*(((166.921580* pow(cas/asl, 7)) / pow(7 * pow(cas/asl, 2) - 1, 2.5)) - 1);

I think that should work.  You can add the above reference to the code, if it isn't there already.

Dennis

seanmcleod, 2018-08-28:
Dennis thanks for the useful insights and reference tip!

I did a fair bit of searching on "CAS to Mach", "pitot pressure equations" and didn't come across anything as useful and complete as the AFFTC. 

I was also initially a bit confused because of the reference to Rayleigh in the JSBSim source code for the Mach > 1 case and so I assumed that the Rayleigh formula was being used. Plus the differences only really started showing from Mach 1.68 onwards. So I assumed they were both using the Rayleigh formula but there was some other difference.

I'm in two minds in terms whether to just apply your suggested fix or whether to convert it to match Luizmonteiro's version which matches the AFFTC equations directly.

seanmcleod, 2018-08-28:
Yep. I did test out your 1-line fix proposal and it mostly worked except for showing up a discrepancy around Mach 1.

Same conditions, 20,000ft standard ISA day.

| CAS (kt) | JSBSim Fixed Mach | Luizmonteiro Mach |
| --- | --- | --- |
| 400 | 0.85 | 0.85 |
| 500 | 1.09 | 1.05 |
| 600 | 1.24 | 1.24 |
| 700 | 1.45 | 1.45 |
| 800 | 1.68 | 1.68 |
| 900 | 1.91 | 1.91 |
| 1000 | 2.13 | 2.13 |

So pretty much identical except for the 500kt case around Mach 1.

For Luizmonteiro's version which follows section 4.8 Calibrated Airspeed of the AFFTC:

>  The subsonic and supersonic Mach number equations are used with the simple substitutions of (CAS/Asl) for M and Psl for P. However, the condition for which the equations are used is no longer subsonic (M <1) or supersonic (M >1) but rather calibrated airspeed being less or greater than the speed of sound ( SL a ), standard day, sea level (661.48 knots).

It means that for 500kt CAS the `if` statement is true so the first `qc` is used for calculating dynamic pressure which is consistent with the comment above.

```python
    if cas < ISAtmosphere.sealevel_speed_of_sound:
        qc = ISAtmosphere.sealevel_pressure * (math.pow(1 + 0.2 * math.pow(cas/asl, 2), 3.5) - 1)
    else:
        qc = ISAtmosphere.sealevel_pressure * (((166.9215801 * math.pow(cas/asl, 7)) / math.pow(7 * math.pow(cas/asl, 2) - 1, 2.5)) - 1)
```

But with the quick 1-liner fix for JSBSim the check for which formula to use for `qc` results in the `if` statement being false and so in this case the supersonic version for `qc` is used.

```python
    pt = pressure + ISAtmosphere.sealevel_pressure*( ((1+cas*cas*ISAtmosphere.sealevel_density/(7*ISAtmosphere.sealevel_pressure)) ** 3.5) - 1)

    if pt/pressure < 1.89223:
        # Mach < 1
        return math.sqrt(5*((pt/pressure)**(1/3.5) - 1))
    else:
        # Mach >= 1
        asl = ISAtmosphere.sealevel_speed_of_sound
        pt = pressure + ISAtmosphere.sealevel_pressure*(((166.921580* math.pow(cas/asl, 7)) / math.pow(7 * pow(cas/asl, 2) - 1, 2.5)) - 1)
```

So I'm planning on switching to using the Luizmonteiro/AFFTC version.

djlinse, 2018-08-28:
I really, really **want** to figure out what is going wrong because I know that I could eventually figure it out. But at the same time, I don't really need to spend any more effort trying to figure out what is going wrong.  :-/ 

Fully documented versions sounds like a better use of time.

bcoconni, 2018-09-01:
While investigating the code and reading documentation, I also wondered what is the sea level values that a real speed indicator uses ?
In JSBSim the true/real pressure and density at sea level are passed to the routines that compute the CAS. However in real life the speed indicator has no access to these values. It either uses constant values (standard sea level values I guess) or a parameter set by the pilot (such as the QNH for the altimeter) ?

I am wondering if we are not computing a CAS that is "too close to perfection" ?

seanmcleod, 2018-09-01:
> Historically, airspeed indicators were constructed with a single pressure input being the differential pressure qc. The gauge is “calibrated” to read true airspeed at sea level standard pressure and temperature. 

> .... calibrated air speed is directly proportional to compressible dynamic pressure. The two measures of airspeed are identical at sea level, standard day. 

So the airspeed indicator "uses" the constant standard day sea level values set via calibration.

So my understanding is we shouldn't be using sea-level pressure or sea-level speed of sound that is different to the standard day values, e.g. when the user specifies a non-standard day via our atmosphere model.

I see that `FGAuxiliary` has some code intialising things with the ISA standard day values for sea-level pressure and temperature.

```c++
FGAuxiliary::FGAuxiliary(FGFDMExec* fdmex) : FGModel(fdmex)
{
  pt = 2116.23; // ISA SL pressure
  tatc = 15.0; // ISA SL temperature
```

So just taking the subsonic case as an example my understanding is we should be doing something along these lines for working out CAS. We have the TAS as output from the EOM etc. and we know the ambient pressure (P) for the aircraft's current position and we know the Mach number based on the TAS and local speed of sound.

```c++
Mach = Vt / in.SoundSpeed;
```

So we use formula (4.11) from the AFFTC manual to calculate `qc` given we know the Mach and P.

[image]

Then we use formula (4.20) to calculate CAS (Vc) where `asl` and `Psl` are the standard day values and where `qc` comes from (4.11).

[image]

seanmcleod, 2018-09-01:
Bertrand I tested out your proposed code and it seems to pretty much match Luizmonteiro's implementation.

As a summary here are some plots showing the differences.

The first plot shows what I originally came across in terms of the difference between JSBSim's current implementation and Luizmonteiro's implementation with a divergence starting around Mach 1.7 at 20,000ft.

[image]
 
The next plot shows the output using Dennis's quick 1 line fix versus Luizmonteiro's, a good match except for the blip around Mach 1.

[image]

And lastly Bertrand's proposed code changes which are pretty much a 100% match to Luizmonteiro's.

[image]

seanmcleod, 2018-09-01:
I was about to submit a push request with a C++ version of the following code until Bertrand's proposed code changes above, so now I'm in 2 minds :wink:

```python
'''
Based on the formulas in the US Air Force Aircraft Performance Flight Testing Manual (AFFTC-TIH-99-01).
In particular sections 4.6 to 4.8.
The subsonic and supersonic Mach number equations are used with the simple substitutions of (Vc/asl) for M 
and Psl for P. However, the condition for which the equations are used is no longer subsonic (M < 1) or 
supersonic (M > 1) but rather calibrated airspeed being less or greater than the speed of sound ( asl ), 
standard day, sea level (661.48 knots). 
'''
def CAStoMach(cas, pressure):

    if cas < ISAtmosphere.sealevel_speed_of_sound:
        # Bernoulli’s compressible equation (4.11)
        qc = ISAtmosphere.sealevel_pressure * (math.pow(1 + 0.2 * math.pow(cas/ISAtmosphere.sealevel_speed_of_sound, 2), 3.5) - 1)
    else:
        # Rayleigh's supersonic pitot equation (4.16) 
        qc = ISAtmosphere.sealevel_pressure * (((166.9215801 * math.pow(cas/ISAtmosphere.sealevel_speed_of_sound, 7)) / math.pow(7 * math.pow(cas/ISAtmosphere.sealevel_speed_of_sound, 2) - 1, 2.5)) - 1)

    # Solving for M in equation (4.11), also used as initial condition for supersonic case
    mach = math.sqrt(5 * (math.pow(qc/pressure + 1, 2/7) - 1))

    if mach > 1:
        # Iterate equation (4.22) since M appears on both sides of the equation
        for i in range(7):
            mach = 0.88128485 * math.sqrt((qc/pressure + 1) * math.pow(1 - 1/(7*mach*mach), 2.5))

    return mach
```

bcoconni, 2018-09-02:
> So my understanding is we shouldn't be using sea-level pressure or sea-level speed of sound that is different to the standard day values, e.g. when the user specifies a non-standard day via our atmosphere model.

That is my understanding as well. So after the code for CAS computation has been fixed, I will track down the calls to `MachFromVcalibrated` and `VcalibratedFromMach` to make sure they are called with the standard SL parameters.

BTW, the SL density `rhosl` is passed to the routines but it is actually the SL speed of sound `asl` that is actually needed, I think we should fix that as well.

> I see that `FGAuxiliary` has some code intialising things with the ISA standard day values for sea-level pressure and temperature.

True but this is for the class construction only and these initial values are overridden when `FGAuxiliary::InitModel` is called.

I would prefer to use the standard SL parameters from `FGAtmosphere` rather than duplicating hard coded values (remember how we struggled with the precision of the standard atmosphere constants while fixing the atmosphere model ? :smiley:)

---

Another thing that bothers me is that the formulas used for CAS computation are using the true ambient pressure for `p` but I am uncertain how this value is measured in real life in supersonic flight ? I mean Pitot tubes measure the total pressure **and** the static pressure so they are both measured behind the shock wave.
[image]

In that case the static pressure measured by the Pitot tube in supersonic flight is not the ambient pressure `Pamb` but a different pressure `Ps` that can be obtained from the normal shock wave theory
[image]
while we are assuming `qc = Pt - Pamb`, it is not the case if `qc` is measured by a Pitot tube where `qc = Pt - Ps` since `Ps != Pamb` in supersonic flight as shown by the formula above.

Below is a picture of an F/A 18 Pitot tube from [Wikipedia](https://commons.wikimedia.org/wiki/File:FA18pitot.png) on which it is clear that the static pressure is taken from the Pitot tube (through the small holes on the tube wall).

[image]

Of course the ambient pressure can be measured elsewhere on the plane but in that case, I am wondering:
* How a static pressure can be measured upstream of a shock wave ?
* What the static pressure from the Pitot tube is used for ?

bcoconni, 2018-09-02:
> I was about to submit a push request with a C++ version of the following code until Bertrand's proposed code changes above, so now I'm in 2 minds :wink:

I don't mind which code is pushed to git as long as both routines `MachFromVcalibrated` and `VcalibratedFromMach` are addressed.

However since this code
```python

    if cas < ISAtmosphere.sealevel_speed_of_sound:
        # Bernoulli’s compressible equation (4.11)
        qc = ISAtmosphere.sealevel_pressure * (math.pow(1 + 0.2 * math.pow(cas/ISAtmosphere.sealevel_speed_of_sound, 2), 3.5) - 1)
    else:
        # Rayleigh's supersonic pitot equation (4.16) 
        qc = ISAtmosphere.sealevel_pressure * (((166.9215801 * math.pow(cas/ISAtmosphere.sealevel_speed_of_sound, 7)) / math.pow(7 * math.pow(cas/ISAtmosphere.sealevel_speed_of_sound, 2) - 1, 2.5)) - 1)
```
is exactly the same as in `FGJSBBase::PitotTotalPressure` (with the only exception that your code return `qc = pt-p` while the C++ code returns `pt`)

And this code
```python
    # Solving for M in equation (4.11), also used as initial condition for supersonic case
    mach = math.sqrt(5 * (math.pow(qc/pressure + 1, 2/7) - 1))

    if mach > 1:
        # Iterate equation (4.22) since M appears on both sides of the equation
        for i in range(7):
            mach = 0.88128485 * math.sqrt((qc/pressure + 1) * math.pow(1 - 1/(7*mach*mach), 2.5))
```
is exactly the same as in `MachFromImpactPressure`, I am suspecting that you will end up with something very similar to the C++ code I submitted above :wink:

Also the unit test `tests/unit_tests/FGJSBBaseTest.h` will need to be updated, more specifically the method `testCASConversion` which tests the results from `MachFromVcalibrated` and `VcalibratedFromMach` against hard coded values.

File `tests/unit_tests/FGJSBBaseTest.h` [lines 101-107](https://github.com/JSBSim-Team/jsbsim/blob/450618a9dc7fe33a7003c794d8fc821db2c6aaf9/tests/unit_tests/FGJSBBaseTest.h#L101)
```c++
    TS_ASSERT_DELTA(VcalibratedFromMach(0.5, p, psl, rhosl), 170.1469, 1E-4);
    TS_ASSERT_DELTA(VcalibratedFromMach(1.0, p, psl, rhosl), 340.2939, 1E-4);
    TS_ASSERT_DELTA(VcalibratedFromMach(1.5, p, psl, rhosl), 493.2217, 1E-4);
    TS_ASSERT_EQUALS(MachFromVcalibrated(0.0, p, psl, rhosl), 0.0);
    TS_ASSERT_DELTA(MachFromVcalibrated(170.1469, p, psl, rhosl), 0.5, 1E-4);
    TS_ASSERT_DELTA(MachFromVcalibrated(340.2939, p, psl, rhosl), 1.0, 1E-4);
    TS_ASSERT_DELTA(MachFromVcalibrated(493.2217, p, psl, rhosl), 1.5, 1E-4);
```

seanmcleod, 2018-09-04:
I've submitted a pull request - https://github.com/JSBSim-Team/jsbsim/pull/114 which basically incorporates Bertrand's proposed code changes which on closer inspection as he mentioned are pretty much the same as my sample changes with some refactoring.

djlinse, 2018-09-05:
@seanmcleod If you can find a copy of the US Navy Test Pilot School, Flight Test Manual 108, Fixed Wing Performance (You can get if from Amazon! USNTPS FTM 108), there is a large section on measuring pressure and many of the issues before subsonic and supersonic.  I think I have/know of a similar NASA document that I'll try to track down as well.

djlinse, 2018-09-05:
@seanmcleod Found it.  NASA RP 1046.  Measurement of Aircraft Speed and Altitude.  1980.  Has most/all of the same equations at the USN TPS and AFFTC.

seanmcleod, 2018-09-06:
@djlinse thanks for the links. I took a look at the NASA report - [Measurement of Aircraft Speed and Altitude](https://ntrs.nasa.gov/archive/nasa/casi.ntrs.nasa.gov/19800015804.pdf) and yes chapter III in particular is pretty detailed and comprehensive,

seanmcleod, 2018-09-06:
One thing I noticed in testing is the difference in the error during the CAS to Mach to CAS roundtrip between subsonic and supersonic. So at 20,000ft standard day.

| CAS (kt) | Mach | CAS Difference |
| --- | --- | --- |
| 100 | 0.222 | 5.54e-13 |
| 200 | 0.44 | -1.14e-13 |
| 300 | 0.651 | 0.00e+00 |
| 400 | 0.854 | -2.27e-13 |
| 500 | 1.05 | 1.45e-04 |
| 600 | 1.24 | 1.37e-04 |
| 700 | 1.45 | 3.16e-04 |
| 800 | 1.68 | 3.22e-04 |

Changing the number of iterations in `MachFromImpactPressure` from 10 to 100 makes virtually no difference. Maybe the Newton-Raphson method would result in higher accuracy? But maybe this sort of difference on the order of 3e-04 is good enough?

djlinse, 2018-09-06:
If you're talking about 3e-4 knots difference, I'm thinking that is probably pretty good from a simulation standpoint.  As far as I know, Mach and CAS are both outputs of the EOM.  So there is no accumulation of error over time.

Theoretically, you would expect better results.  Since the change of iterations from 10 to 100 makes almost no difference, I would suspect that 1) Newton-Raphson isn't going to help and 2) there is a (subtle?) rounding error in one or more of the constants or in the calculation itself that makes the "round trip" not as accurate as you might like from the computer-science-y side of things.

seanmcleod, 2018-09-07:
Okay so I took a look at the 2 constants in use and calculated them from the AFFTC, i.e.

```python
166.92158 = 166.92158009316827 = (1.2**3.5)*(2.5**2.5)*(2.4**2.5)
0.881285 = 0.8812848543473311 = math.sqrt((7**2.5)/((1.2**3.5)*(2.5**2.5)*(2.4**2.5)))
```

In using the higher precision versions the difference in CAS after the round-trip was 2-3 orders of magnitude smaller. Previously there was no change in the order of magnitude when increasing the number of iterations from 10 to 100, but now there is a massive difference.

 CAS (kt) | Mach | CAS Difference (10 iter) | CAS Difference (100 iter) |
| --- | --- | --- | --- |
| 100 | 0.222 | 5.54e-13 | |
| 200 | 0.44 | -1.14e-13 | |
| 300 | 0.651 | 0.00e+00 | |
| 400 | 0.854 | -2.27e-13 | |
| 500 | 1.05 | -2.96e-06 | -2.27e-13 |
| 600 | 1.24 | -5.31e-06 | 0.00e+00 |
| 700 | 1.45 | -6.06e-06 | -2.27e-13 |
| 800 | 1.68 | -8.89e-06 | 2.27e-13 |
| 900 | 1.9 | -2.62e-06 | 0.00e+00 |
| 1000 | 2.13 | -6.26e-07 | 0.00e+00 |
| 1100 | 2.36 | -1.50e-07 | 0.00e+00 |

seanmcleod, 2018-09-07:
So how many iterations should we use for the supersonic case?

 CAS (kt) | Mach | 10 iter | 15 iter | 20 iter | 25 iter |
| --- | --- | --- | --- | --- | --- |
| 500 | 1.05 | -2.96e-06 | -2.18e-08  | -1.60e-10 | -1.14e-12 |
| 600 | 1.24 | -5.31e-06 | -5.73e-09 | -6.03e-12 | 0.00e+00 |
| 700 | 1.45 | -6.06e-06 | -3.48e-08 | -2.27e-10 | -1.71e-12 |
| 800 | 1.68 | -8.89e-06 | -1.28e-08 | -1.82e-11 | 2.27e-13 |
| 900 | 1.9 | -2.62e-06 | -1.04e-09 | -5.68e-13 |  0.00e+00 |
| 1000 | 2.13 | -6.26e-07 | -8.04e-11 | 0.00e+00 | 0.00e+00 |
| 1100 | 2.36 | -1.50e-07 | -6.82e-12 | 0.00e+00 | 0.00e+00 |

djlinse, 2018-09-07:
If it truly made any difference for any other calculation, I'd say that you put a convergence condition rather than just an arbitrary 10 iterations.  But since it doesn't really make any difference, as far as I can tell, I'd say that 10 iterations is plenty.  The errors are telling me that CAS is accurate to 9-10 digits, and that's given all of the assumptions already inherent in the atmospheric model.  That's _**WAAAAYYYYY**_ beyond the needed level for any realistic use.

bcoconni, 2018-09-09:
@seanmcleod 
> So how many iterations should we use for the supersonic case ?

FWIW if we really want to get a result driven by precision rather than by the number of iterations, the code can easily be modified to something like
```c++
  if (M > 1.0)
    for(unsigned int i=0; i<10; i++) {
      double newMach = 0.881285*sqrt(A*pow(1-1.0/(7.0*M*M),2.5));
      if (abs(newMach-M) < 1E-6 break;
      M = newMach;
    }
```
Regarding the speed of convergence, we have moved from a Newton-Raphson (NR) scheme to a fixed point algorithm. The latter is simpler but its convergence speed depends on the Lipschitz constant `k` of the function: the closer this constant is to 1.0, the slower the convergence (as it converges as `k^n/(1-k)` after `n` iterations). I have not checked the Lipschitz constant of the function above but it is possible that it is somewhat close to 1.0 in some ranges of Mach number.

AFAIK NR does not suffer from this limitation and as long as you correctly guess the initial point, NR convergence is amazingly fast i.e. it grossly doubles the number of correct digits at each iteration (meaning you need approximately 4 iterations to get 16 significant digits). However this is at the expense of more complicated code and of being able to find an initial value that is in the basin of attraction.

Since our function of interest is quite simple and we are only dealing with 10 iterations, its CPU cost is negligible. In addition it is called in `FGInitialCondition.cpp` and in `FGAuxiliary.cpp` and given that 
* Optimization is mostly irrelevant for functions executed during initialization and,
* the function `VcalibratedFromMach` is called only once per simulation time step 

then I guess we can live with the current code since it is easier to maintain.

closed 2018-09-10 by seanmcleod