JSBSim-Team/jsbsim #116: Expose air-fuel ratio of Piston engine to the property tree

agodemar, 2018-09-09:
*From: Alan Teeder*
A user on the FG forum [asks to have air-fuel ratio of Piston engine exposed](https://forum.flightgear.org/viewtopic.php?f=66&t=34762&sid=3ccba873d448bf34b9e84c643ab8a50b&p=336225#p336225) so that he can trigger exhaust smoke with an over-rich mixture.

seanmcleod, 2018-09-10:
Alan you mentioned in the FlightGear forum that there is some commented out code in FGPiston.cpp to calculate the AFR (Air Fuel Ratio). I presume you're referring to the following?

```c++
/**
 * Calculate the fuel flow into the engine.
 *
 * Inputs: Mixture, thi_sea_level, p_amb, m_dot_air
 *
 * Outputs: equivalence_ratio, m_dot_fuel
 */

void FGPiston::doFuelFlow(void)
{
  double thi_sea_level = 1.3 * in.MixturePos[EngineNumber]; // Allows an AFR of infinity:1 to 11.3075:1
  equivalence_ratio = thi_sea_level * 101325.0 / p_amb;
//  double AFR = 10+(12*(1-in.Mixture[EngineNumber]));// mixture 10:1 to 22:1
//  m_dot_fuel = m_dot_air / AFR;
  m_dot_fuel = (m_dot_air * equivalence_ratio) / 14.7;
  FuelFlowRate =  m_dot_fuel * 2.2046;  // kg to lb
  if(Starved) // There is no fuel, so zero out the flows we've calculated so far
  {
    equivalence_ratio = 0.0;
    FuelFlowRate = 0.0;
    m_dot_fuel = 0.0;
  }
  FuelFlow_pph = FuelFlowRate  * 3600;
  FuelFlow_gph = FuelFlow_pph / FuelDensity;
}
```

So in theory we could just uncomment the AFR calculation and expose it via the property tree?

alanjt, 2018-09-10:
The forum post has turned into one of those squabbles about something unrelated,  3 pages of it so far, and none of it about the original request.........
The OP (who is not involved in the rants)  wants it so that he can trigger smoke when the mixture is over rich. No doubt there will be other uses.
Alan

bcoconni, 2018-09-10:
@seanmcleod 
> So in theory we could just uncomment the AFR calculation and expose it via the property tree?

No by definition the AFR is the ratio of the air mass flow to the fuel mass flow which is `14.7/equivalence_ratio` given the line of code below
```c++
m_dot_fuel = (m_dot_air * equivalence_ratio) / 14.7;
```
I think the lines above have been commented out because they have been superseded by this code.

alanjt, 2018-09-10:
From: Bertrand Coconnier 
Sent: Monday, September 10, 2018 8:52 PM
To: JSBSim-Team/jsbsim 
Cc: alanjt ; Comment 
Subject: Re: [JSBSim-Team/jsbsim] Expose air-fuel ratio of Piston engine to the property tree (#116)

@seanmcleod

  So in theory we could just uncomment the AFR calculation and expose it via the property tree?

No by definition the AFR is the ratio of the air mass flow to the fuel mass flow which is 14.7/equivalence_ratio given the line of code below

m_dot_fuel = (m_dot_air * equivalence_ratio) / 14.7;I think the lines above have been commented out because they have been superseded by this code.

----------------------------------

Bertrand

Agreed, it needs a little more thought.

AFR, in conjunction with equivalence_ratio, occurs several times in commented out sections of the code, It just seems that AFR was not used in the final version of the code and performance was calculated from the equivalence ratio, so AFR was removed. 

AFAIK neither AFR nor equivalence_ratio are exposed to the property tree. Perhaps both could be, but AFR is the more commonly understood of the two by lay persons.

Alan

seanmcleod, 2018-09-13:
I've submitted a pull request https://github.com/JSBSim-Team/jsbsim/pull/118 to expose an Air Fuel Ratio (AFR) property for piston engines.

seanmcleod, 2018-09-13:
I decided to compare the engine power output versus AFR.

[image]

I used the C172p aircraft model at sea-level with the engine running and brakes on and `fcs\throttle-cmd-norm = 1` and then ramped the `fcs/mixture-cmd-norm from 1.3 down to 0.7`.

To compare against the Wikipedia graph:

[image]

seanmcleod, 2018-09-13:
Just wondering whether the power peak appearing so far left around an AFR of 10 is related to this issue - https://github.com/JSBSim-Team/jsbsim/issues/96 ?

alanjt, 2018-09-14:
From: Sean McLeod 
Sent: Thursday, September 13, 2018 5:19 PM
To: JSBSim-Team/jsbsim 
Cc: alanjt ; Comment 
Subject: Re: [JSBSim-Team/jsbsim] Expose air-fuel ratio of Piston engine to the property tree (#116)

Just wondering whether the power peak appearing so far left around an AFR of 10 is related to this issue - #96 ?

It looks like #96  is reporting exactly the same problem with the fgpiston model.

That should be fun trying to sort out 

Who wrote the original model, and what documents was it based upon?

Alan

bcoconni, 2018-09-15:
I have just pushed JSBSim bleeding edge to FlightGear so this new AFR feature is available there.

alanjt, 2018-09-15:
Many thanks for this. I have updated the forum thread.
Alan

closed 2018-09-26 by seanmcleod

legoboyvdlp, 2018-11-28:
To implement something such as spark plug fouling  (https://www.simcoders.com/2016/06/02/20160602keep-your-spark-plugs-clean/)

is AFR the ratio that I want? So if AFR is less than 15, the spark plugs should begin to foul (slowly - maybe quicker at AFR of ~10)?

alanjt, 2018-11-28:
From: Jonathan Redpath 
Sent: Wednesday, November 28, 2018 1:13 PM
To: JSBSim-Team/jsbsim 
Cc: Alan Teeder ; Comment 
Subject: Re: [JSBSim-Team/jsbsim] Expose air-fuel ratio of Piston engine to the property tree (#116)

To implement something such as spark plug fouling (https://www.simcoders.com/2016/06/02/20160602keep-your-spark-plugs-clean/)

is AFR the ratio that I want? So if AFR is less than 15, the spark plugs should begin to foul (slowly - maybe quicker at AFR of ~10)?

—

A lot depends upon the engine design and the heat range and the age of the spark plug. I would expect zero fouling if the mixture is approximately correct.

Alan