JSBSim-Team/jsbsim #119: Sensor element: fail_stuck behaviour

legoboyvdlp, 2018-09-13:
Hi,
I'm trying to replicate the Air Data Reference Unit of the A320 using JSBSIM <sensor> elements. 

[image]
I set fail_stuck to 1.0. 
I expected the speed to get stuck at 74.5 knots, instead it actually went to 0. Is this intended behaviour?

Thanks,
Jonathan

legoboyvdlp, 2018-09-14:
This might help:
https://en.wikipedia.org/wiki/Pitot-static_system#Pitot-static_errors

seanmcleod, 2018-09-14:
If you want to model the blocked pitot tube case and the blocked static port case then you'll need 2 sensor elements, 1 to model the pitot sensor and 1 to model the static port and you would then set `fail_stuck` on the relevant sensor and you would need to calculate the airspeed using the 2 sensors.

In other words you can't have a single sensor element representing an airspeed indicator and then set `fail_stuck` on that and assume you can model the blocked pitot and blocked static cases. In this case you would only be modeling the airspeed indicator/computer itself failing/getting stuck.

legoboyvdlp, 2018-09-14:
`So at first glance it looks like fail_stuck should output the last sensor input. However PreviousOutput is only set in the Lag() method. So if you haven't configured any lag parameters, hence the reason I was asking how you had set things up, then PreviousOutput will always be 0, it's initialized value on sensor element creation.

So this does seem to be a bug in the FGSensor class.`

Ah - that's exactly right, I'm not using Lag! Thanks :)

seanmcleod, 2018-09-14:
> Ah - that's exactly right, I'm not using the lag parameters at all! That would explain it.

But it's still a bug since you shouldn't be **required** to use lag, but adding some lag, even very minimal, is a work around for now.

bcoconni, 2018-10-28:
The problem should now be fixed by commit f543d63.
@legoboyvdlp 
Please test and let me know if that fixes the issue on your side as well.

bcoconni, 2018-10-28:
@legoboyvdlp 
If you are testing with FlightGear, the bug fixes have just been pushed to FlightGear source code.

legoboyvdlp, 2018-10-28:
Great!

May I ask if the fgturbine fixes I did were pushed?

bcoconni, 2018-10-28:
> May I ask if the fgturbine fixes I did were pushed?

Yes they will be pushed in a separate commit because bug fixes could make their way to the release 2018.3 of FlightGear while new features are intended to stay in the branch `next` until the next release.

The commit in question should be pushed to FlightGear today.

bcoconni, 2018-11-17:
@legoboyvdlp 
Have you had the opportunity to check if the bug you reported in this issue is fixed ? The fix has been pushed in FlightGear 2018.3 and in its `next` branch

legoboyvdlp, 2018-11-17:
Thanks! No, I haven't - I've been busy elsewhere. I can check in a few days.

bcoconni, 2018-12-27:
@legoboyvdlp 
> No, I haven't, sorry - I've been busy elsewhere. I can check in a few days.

Have you had the opportunity to check if the bug you reported in this issue is fixed ?

legoboyvdlp, 2018-12-27:
again I've been quite busy. but if its in 2018.3, then I suppose I can check right now

bcoconni, 2018-12-27:
OK thanks. Issue closed.

closed 2018-12-27 by bcoconni