JSBSim-Team/jsbsim #120: /gear/*/rollspeed-ms stuck when wow=false

mike402, 2018-09-24:
In FlightGear, ``/gear/*/rollspeed-ms`` seem to stay at the last value they had on the ground when ``wow`` becomes false, as described in https://sourceforge.net/p/flightgear/codetickets/2053/

I think this could be JSBSim-side, because YASim is not affected.

mike402, 2018-09-24:
Additionally, I remember seeing that applying brakes does not seem to stop wheels from rotating in the air.

seanmcleod, 2018-09-24:
Looking at the JSBSim source I see the following in FGLGear.h and .cpp:

```c++
    property_name = base_property_name + "/wheel-speed-fps";
    PropertyManager->Tie( property_name.c_str(), (FGLGear*)this,
                          &FGLGear::GetWheelRollVel);
```

I didn't find any references to `rollspeed` in the JSBSim source code so I'm assuming the FlightGear interface is mapping `wheel-speed-fps` to `rollspeed-ms`.

```c++
  double GetWheelRollVel(void) const   { return vWhlVelVec(eX)*cos(SteerAngle)
                                              + vWhlVelVec(eY)*sin(SteerAngle);  }
```

And in the `WOW == false` case in `FGLGear::GetBodyForces` there is explicit code to spin the wheel down:

```c++
      // Let wheel spin down slowly
      vWhlVelVec(eX) -= 13.0 * in.TotalDeltaT;
      if (vWhlVelVec(eX) < 0.0) vWhlVelVec(eX) = 0.0;
```

So it would appear that JSBSim does spin down the wheel once `WOW == false`. I haven't run a script yet with a take-off to confirm though.

legoboyvdlp, 2018-09-27:
Is it possible to add the brake affecting the wheelspeed in the air if that is not yet present?

Some aircraft also have a "brake pad" on the nose gear (which tends not to have a brake) to slow and stop the wheels after they are retracted.

bcoconni, 2018-10-06:
IMHO extracting the wheel spinning velocity from `FGLGear::GetWheelRollVel` was a bad idea from the start. This method is actually returning the aircraft velocity projected along the rolling direction of a gear for the purpose of ground reactions management.

Now some people made creative uses of that value included aliasing it with `/gear/*/rollspeed-ms` in FlightGear. If you want a proper model of a wheel, you need its diameter, inertia and friction parameters. These are not modeled in JSBSim but instead a hard coded value `-13.0` is used which is supposed to model all possible wheel design...

Honestly, either the physics of the wheel is modeled properly or I would leave it in its current state. I vote against adding new magic numbers to the existing ones.

bcoconni, 2018-12-27:
And the code has just been pushed to FlightGear.
Let me know if that fixes the issue you reported.

mike402, 2018-12-27:
Thank you, I will try it out tonight!

On Thu, Dec 27, 2018 at 05:54:19PM +0000, Bertrand Coconnier wrote:
> And the code has just been pushed to FlightGear.
> Let me know if that fixes the issue you reported.

legoboyvdlp, 2018-12-27:
Excellent!

legoboyvdlp, 2018-12-27:
Excellent! Now the Spitfire's tailwheel won't crazily rotate :)

bcoconni, 2018-12-28:
Thanks, @mike402 also reported that the problem is fixed on the FlightGear mailing list.

Issue closed.

closed 2018-12-28 by bcoconni