JSBSim-Team/jsbsim #122: Possibility to have a reset command on the integrator filter

abassign, 2018-10-10:
I am writing the JSBSim code to reproduce a gauge (PHI) for navigation via radar doppler. The instrument was created in the 50s and 60s and was mounted on NATO's G91s and on Italian F104s. My problem is the need to reproduce the electronic supplement used to advance the mile counter. To do this I use the integration filter with an input obtained by a processing of the sensor function that gives a data of the speed affected by an error. Everything works very well, but I would like to succeed, in order to be more faithful with the technology of the time, to have a supplement that resets at every mile traveled (this is to avoid going out of scale being an electronic integrator). I noticed that currently the integrator filter defines the "trigger" method that blocks the integration, but does not define a method to reset the integrator and therefore make it restart from zero or better from a certain predefined value. I think it's a useful function to simplify the JSBSim code in case we want to simulate the systems functions. The same problem is when using the integrator to determine the amount of fuel in the tanks. In this case, the reset would be convenient if it were a value (the amount of initial fuel) that is then consumed by the engine.

Thank you for your interest
Best regards Adriano

bcoconni, 2018-10-17:
I guess you are talking about the `<pid>` element ?

If so, it seems that the docs are misleading because the `<trigger>` property does not put the integration on hold but resets the integrator to zero

[Below is `FGPID.cpp line 211`](https://github.com/JSBSim-Team/jsbsim/blob/61f1e8c3af050ce75f2032dbd8e43a6d4897502a/src/models/flight_control/FGPID.cpp#L211)
```c++
  if (test < 0.0) I_out_total = 0.0;  // Reset integrator to 0.0
```
Could you please check if setting the property in `<trigger>` to a negative value resets the integrator as you need ?

In any case, the documentation definitely needs an update.

bcoconni, 2018-10-17:
BTW, I'm pretty sure that the lines 221-222 need to be swapped.
```c++
  Input_prev = Input;
  Input_prev2 = Input_prev;
```
and the documentation of the parameter `<pvdot>` is incomplete.

abassign, 2018-10-18:
I did a test with the PD used as a filter integrator.
It works with negative values of the reset, but only if the parameter Kp (?!) Is set to zero. The integrator gives a zero output value until the trigger returns to 0. If the triger is set to 1, the integration stops until the trigger is set to zero. In this case the presence of a non-zero value in Kp has no effect.

I did a test with an integrator filter, the reset parameter if zero does not block the integration, if it is positive or negative it stops the integration, but it does not reset.

At this point I think it would be enough to activate the negative reset parameter to zero the integration so as to be identical to the PID. It would also be interesting to activate the possibility to define the type of integration as for the PID.

bcoconni, 2018-10-27:
> the integration filter that is necessary to manage the distance measurement through a real doppler radar subjected to noise (real sensor).

Honestly I don't see why an `<integration>` filter is **necessary**. Whether the component is named `<integration>` or `<pid>` (with `<kp>` and `<kd>` set to zero) does not make much difference: the same computations are made behind the scenes.

> ... Using PID as a filter integrator is possible, but it seems rather bizarre ... ;)

I don't see anything bizarre in using a `<pid>` for your particular purpose. I guess it is a matter of taste.

> At this point I think it would be enough to activate the negative reset parameter to zero the integration so as to be identical to the PID. It would also be interesting to activate the possibility to define the type of integration as for the PID.

Will do. But I will actually modify the code in JSBSim to use a `<pid>` behind the scenes when an `<integrator>` will be requested because I am not willing to maintain duplicated code. This way you will get all the features of an integration with `<pid>` and still be able to use the tag `<integrator>`.

bcoconni, 2018-10-27:
> BTW, I'm pretty sure that the lines 221-222 need to be swapped.

Indeed this was needed, otherwise the integration with Adams-Bashforth 3 would fail. This has been fixed in commit 6adf268

bcoconni, 2018-10-27:
> Will do. But I will actually modify the code in JSBSim to use a `<pid>` behind the scenes when an `<integrator>` will be requested.

Done in commit 0c189fe. The component `<integrator>` now has the features you requested:
- the ability to reset it by setting the `<trigger>` property to a negative value
- the ability to choose the integration scheme among `rect` (Euler), `trap` (trapezoidal), `ab2` (2nd order Adams-Bashforth) or `ab3` (3rd order Adams-Bashforth)
```xml
<integrator name="test">
  <input> test-input </input>
  <c1 type="ab2"> 1.0 </c1> <!-- Uses 2nd order Adams-Bashforth -->
</integrator>
```
@abassign 
Could you please test and let me know if that works ?

bcoconni, 2018-10-27:
> In any case, the documentation definitely needs an update.

> the documentation of the parameter `<pvdot>` is incomplete.

Both are now fixed.

abassign, 2018-10-30:
> The code is now also pushed to FlightGear next branch.

I am very happy that you made the change on the integrator I requested, I
just tested it and I must say that it works flawlessly. I used the
second-order integration, it works correctly. I did a test with the PHI
(The tool that inetegra the ground speed data with the distance, simulating
a doppler radar affected by Gaussian error), even this worked very well and
the data are within the average error hypothesized for this tool. Therefore
I consider my request completely satisfied. I hope you will soon update the
manual so that others can use this useful feature.

thank you for your assistance 😊

agodemar, 2018-10-31:
@abassign Now that the problem is addressed and solved, I close this issue. Thanks.

closed 2018-10-31 by agodemar