JSBSim-Team/jsbsim #123: Incorrect effect of humidity on denisty at very high altitudes

bcoconni, 2018-10-17:
The code that manages the effect of humidity on the air density can give incorrect results at very high altitudes. The problem comes from the fact that at very high altitudes (above 150,000 ft) the saturated vapor pressure might become higher than the ambient pressure. In such cases, humidity rates of 100% gives inconsistent states and can even lead to negative densities.

In the real world, when the saturated vapor pressure is higher than the ambient pressure, water boils until the vapor partial pressure reaches the saturated pressure. In open atmosphere, one cannot bring enough water at very high altitudes to elevate the partial pressure to the saturated pressure. In other words, water cannot exist in a liquid state at very high altitudes and evaporates.

In addition, according to the US Standard Atmosphere (1976), measurements show that the quantity of water vapor is very low at high altitudes i.e. less than 20 ppm (parts per million) for altitudes higher than 12 km/39 kft (see tables 20 and 21 in chapter 3.3 "Water vapor" ). Basically this means that the HR is almost zero up there.

My suggestion is to force the water vapor quantity to the values reported by ISA 1976 above a certain altitude threshold (value to be defined). That would avoid unrealistically high humidity rates and nonsensical physical parameters such as a negative density.

Until this problem is fixed, I suggest to disable the effect of humidity on air density in FlightGear.

Phillip-M-Feldman, 2019-02-06:
If someone can point me to the relevant code, I'm willing to take a look at this issue.

When switching from one model to another at a threshold, one can introduce a non-physical discontinuity.  Perhaps there is a way to avoid this.

seanmcleod, 2019-02-08:
@Phillip-M-Feldman take a look at [FGStandardAtmosphere.cpp](https://github.com/JSBSim-Team/jsbsim/blob/61f1e8c3af050ce75f2032dbd8e43a6d4897502a/src/models/atmosphere/FGStandardAtmosphere.cpp) and the tests in [TestStdAtmosphere.py](https://github.com/JSBSim-Team/jsbsim/blob/6be80dee85d338b701f4987a5a9a66988f2ea7d2/tests/TestStdAtmosphere.py)

bcoconni, 2019-02-09:
@Phillip-M-Feldman Thanks for your help.

Since JSBSim atmosphere model is based on the International Standard Atmosphere (1976), I would encourage you to read the document and especially its section 3.3 "Water Vapor".

The PDF is available from the NASA Technical Reports Server https://ntrs.nasa.gov/archive/nasa/casi.ntrs.nasa.gov/19770009539_1977009539.pdf

Phillip-M-Feldman, 2019-02-10:
I got the document and am now reading the relevant material.  Thanks!

bcoconni, 2019-05-18:
The problem has been fixed (see the commit message of 94d5877 for the details) and has been pushed to FlightGear for 2 weeks without anyone reporting a problem

The issue is fixed and closed.

closed 2019-05-18 by bcoconni