JSBSim-Team/jsbsim #124: <function>: no <floor>, <ceiling> and <fmod> available.

mike402, 2018-10-27:
<floor>, <ceiling> and <fmod> would have been very convenient, because their equivalents are very bulky and typo-prone.

mike402, 2018-10-27:
For example this piece of code used to implement an indicator with Geneva drive would get twice, if not more, as long if I had to make it work with negative input values when using ``<integer>`` instead of ``<floor>``:

~~~
  <fcs_function name="fcs/NPK/instr/UVO[1]/counter/H-100m[0]">
   <function>
    <!-- Geneva drive -->
    <!-- JSBBUG Moves sharply for negative values, because there is no <floor>, and <integer> is used instead! -->
    <sum>
     <!-- Simple step -->
     <integer>
      <product>
       <!-- lower digit -->
       <p>fcs/NPK/instr/UVO[1]/gauge/H-m</p>
       <!-- = 1 / base -->
       <v>0.01</v>
      </product>
     </integer>
     <!-- Transition -->
     <min>
      <v>1.0</v>
      <max>
       <v>0.0</v>
       <product>
        <difference>
         <product>
          <!-- lower digit -->
          <p>fcs/NPK/instr/UVO[1]/gauge/H-m</p>
          <!-- = 1 / base -->
          <v>0.01</v>
         </product>
         <integer>
          <product>
           <!-- lower digit -->
           <p>fcs/NPK/instr/UVO[1]/gauge/H-m</p>
           <!-- = 1 / base -->
           <v>0.01</v>
          </product>
         </integer>
         <!-- transition length, 0.0 to 1.0 -->
         <v>0.9</v>
        </difference>
        <!-- = 1 / transition length -->
        <v>10.0</v>
       </product>
      </max>
     </min>
    </sum>
   </function>
  </fcs_function>
~~~

bcoconni, 2018-10-27:
Issue closed as it is a duplicate of issue #94.
The new suggestions `<floor>`, `<ceiling>` and `<fmod>` have been copied to #94.

closed 2018-10-27 by bcoconni