JSBSim-Team/jsbsim #125: Inertial tensor matrix initialization error

jinzhu1911, 2018-10-29:
The inertia matrix J for rigid body is  
FGMatrix33(  bixx,  -bixy,  -bixz,
                    -bixy,  biyy,  -biyz,
                    -bixz,  -biyz,  bizz )
  (ref. Stevens and Lewis, "Flight Control & Simulation"),
but in the function FGMassBalance::Load of FGMassBalance class , 
you pass FGMatrix33(  bixx,  -bixy,  bixz,
                                 -bixy,  biyy,  -biyz,
                                   bixz,  -biyz,  bizz ) to SetAircraftBaseInertias.
I think this is wrong.
Wait for your answer. Thanks very much!

bcoconni, 2018-11-01:
I first thought you were right but after some investigations, the code is actually correct :smiley: 

The trick is that the matrix in Steven & Lewis is given in the **body** frame while the inertia in JSBSim are specified in the **structural** frame (see [JSBSim reference manual](https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/user-manual-frames-of-reference) about frame definitions).

The opposite sign of `bixz` with respect to Steven & Lewis is therefore a result of the transformation from the structural frame to the body frame (see the message of commit f04489499d477).

I will add some comments to the code since, I agree, the motivation behind the sign modification is not obvious.

bcoconni, 2018-11-01:
After some further investigations, the problem was raised in 2006 (see the [SF ticket 46](https://sourceforge.net/p/jsbsim/bugs/46/)) and, as I mentioned above, solved by the commit f044894 in 2007.

At the time @jonsberndt wrote
> The aircraft inertias are read in correctly. However, they are given in structural coordinates, with the X axis positive aft, and the Y axis positive out the right wing. When they are read in, they are transferred to body frame, which involves a 180 degree roation about the Y axis. That changes the sign of the x and z terms, so any product of inertia that involves the x and y axis, or z and y axis should flip it sign. Unfortunately, the xz term was also having its sign flipped, which it should not have. So, that has been fixed. The relevant setting function is now:
```c++
SetAircraftBaseInertias(FGMatrix33( bixx, -bixy,  bixz,
                                   -bixy,  biyy, -biyz,
                                    bixz, -biyz,  bizz ));
```
> Before, there was a minus sign in front of the "bixz" variable. Now, there is not.

Another point that must be clarified is that inertia cross products (`Ixy`, `Ixz` and `Iyz`) must be specified **with a negative sign** in the XML definition file. Said in other words, **JSBSim does not invert the sign of inertia cross products for you**.

See for example the [DHC6 XML definition file](https://github.com/JSBSim-Team/jsbsim/blob/b914b2480217354c377fbec909a933875d8e8223/aircraft/DHC6/DHC6.xml#L97):
```xml
<mass_balance>
    <ixx unit="SLUG*FT2"> 19317 </ixx>
    <iyy unit="SLUG*FT2"> 24679 </iyy>
    <izz unit="SLUG*FT2"> 35344 </izz>
    <ixz unit="SLUG*FT2"> -1099 </ixz>
    <emptywt unit="LBS"> 8100 </emptywt>
```
This certainly needs to be clarified in the docs as well.

bcoconni, 2018-11-01:
> I will add some comments to the code since, I agree, the motivation behind the sign modification is not obvious.

> Said in other words, **JSBSim does not invert the sign of inertia cross products for you**. [...] This certainly needs to be clarified in the docs as well.

Done in commit 0376337

jinzhu1911, 2018-11-07:
Thanks very much!

closed 2018-11-07 by agodemar

rega0051, 2018-11-07:
Remember that: Frames and Coordinates are not the same.

If the Inertia tensor is defined in the Structural-frame and transformed to the Body-frame. There should be tensor rotation (which can be reduced to sign changes as discussed previously). But this is only the coordinate transformation from S to B. The frame also needs to be translated, Structural-frame origin is typically at the nose, Body-frame origin is typically at the CG.

So, as it stands the XML specifies Inertia elements at the Body-frame, but in the Structural-coordinates! This is transformed into Inertia at the Body-frame in the Body-coordinates.

This is rather confusing because the CG location is specified in the XML as a location with respect to the  Structural-Frame in the Structural-coordinates. So, the Frame is not consistent, but the Coordinates are...

mike402, 2019-12-09:
Personally, it is still unclear whether I should invert the sign of `Ixz`, because it is with a plus sign in the mentioned code?

bcoconni, 2019-12-09:
> Personally, it is still unclear whether I should invert the sign of Ixz, because it is with a plus sign in the mentioned code?

Very simple: you don't need to. The bug was in JSBSim not in your model.

mike402, 2019-12-09:
The code comment seems ambiguous: `in most￼ cases, negative values should be provided for <ixy>, <ixz> and <iyz>` -- "negative" as in "sign opppsite to your data" or "negative"? That is, if I have negative cross products from the literature, do I put them as provided or make them positive?

Are there any quick rules-of-thumb for cross products like "positive Ixz is when starting to roll right you also start to yaw right, negative is when you start to yaw left" and "positive Ixz is when weights with longer moment arm lie in I and III quadrants of xz plane, and negative is when II and IV quadrants" (and did I get this example right?)?

bcoconni, 2019-12-14:
>  "negative" as in "sign opppsite to your data" or "negative"?

"Negative values" means "values lower than zero".
> That is, if I have negative cross products from the literature, do I put them as provided or make them positive?

Well as I explained [above](https://github.com/JSBSim-Team/jsbsim/issues/125#issuecomment-435035000), inertia matrices must be supplied to JSBSim in the structural frame. That means that:
* If the inertia matrix you got from the literature is expressed in the **structural frame** then you need **not** to change the sign of `Ixy` and `Iyz` when you copy the inertia matrix to the XML definition file.
* If the inertia matrix you got from the literature is expressed in the **body frame** then you **must** change the sign of `Ixy` and `Iyz` when you copy the inertia matrix to the XML definition file.

In any case, you **do not** need to change the sign of `Ixz`.
> Are there any quick rules-of-thumb for cross products like [...]

Well, sort of:
1. Most airplanes are symmetrical (mirrored) with respect with the `XZ` plane so `Ixy` and `Iyz` are both null (i.e. equal to zero).
2. The sign of `Ixz` describes if the body frame should be rotated clockwise (i.e. if `Ixz` < 0) or anticlockwise (i.e. if `Ixz` > 0) around the `Y` axis to match the inertia principal axes.

Since on most aircraft the tail is installed on the upper side of the rear fuselage then `Ixz` is negative as in the example the DHC6 XML definition file https://github.com/JSBSim-Team/jsbsim/blob/b914b2480217354c377fbec909a933875d8e8223/aircraft/DHC6/DHC6.xml#L97-L102
[image]

Unconventional tail layout such as the MQ-1 Predator will give you a positive value (higher than zero) for the inertia `Ixz`
[image]

mike402, 2019-12-14:
Thank you for clarification, now I get it!

I think then https://github.com/JSBSim-Team/jsbsim/blob/master/src/models/FGMassBalance.h#L69 has a typo that it mentions ´<ixz>´?

NikolaiVChr, 2019-12-20:
>In any case, you **do not** need to change the sign of `Ixz`.

In a document about F-16, Ixz is given in Body frame and is positive, but the tail is on upper side of fuselage.
Could that be cause it has a big inlet under the fuselage I wonder..