JSBSim-Team/jsbsim #126: Extraction of single digits ...

abassign, 2018-10-31:
Using JSBSim to replace NASAL for Flightgear, I found it quite complicated to extract individual digits from a number. This function is very convenient for simulating electromechanical meters or electromechanical processing devices typical of 50-90 aircraft. Currently I solved the problem with a series of functions that use the division-integer method, but it would seem much simpler to have a method of the function class that extracts a single digit from a number using the comma as a reference. This method could be called "extrdigit" with two parameters, the first is the number from which to make the extraction and the second the position with respect to the comma. If this second parameter is negative, the digit to the right of the comma is extracted, if positive on the left. If zero returns a zero.
This method would greatly simplify the construction of interfaces and would encourage the use of JSBSim in complete replacement of NASAL.

Thank you for your interest.

mike402, 2018-10-31:
I made snippets for Geneva-driven digits, look for "Geneva drive" comment, but they require the new ``<floor>`` function to work correctly with negative values:

https://gitlab.com/mdanil/Tu-144/blob/master/Tu-144-jsbsim/NPK/DISS.xml

bcoconni, 2018-11-01:
As for similar requests from @mike402, I would rather address this issue with a mechanism which allows code re-use such as I proposed in issue #94.

This is because I don't think programming every conceivable function in C++ (sawtooth function, stair functions, digit extraction, etc.) is a viable option for the long-term.

abassign, 2018-11-03:
For a 3-digit Genoa-Drive in the G91R1B I use these three functions, they
are quite compact, but obviously I would like to have an explicit function.
It would be desirable to have a function similar to that actuator, this
device is very common in aircraft from 1940 up to 2000.

  <fcs_function>
            <function>
                <integer>
                    <quotient>
                        <p>systems/gauges/PHI/indicator/mil-direction</p>
                        <v>100</v>
                    </quotient>
                </integer>
            </function>
            <output>systems/gauges/PHI/indicator/digit1A</output>
        </fcs_function>

        <fcs_function>
            <function>
                <integer>
                    <quotient>
                        <difference>

<p>systems/gauges/PHI/indicator/mil-direction</p>
                            <product>
                                <p>systems/gauges/PHI/indicator/digit1A</p>
                                <v>100</v>
                            </product>
                        </difference>
                        <v>10</v>
                    </quotient>
                </integer>
            </function>
            <output>systems/gauges/PHI/indicator/digit2A</output>
        </fcs_function>

        <fcs_function>
            <function>
                <integer>
                    <difference>
                        <p>systems/gauges/PHI/indicator/mil-direction</p>
                        <sum>
                            <product>
                                <p>systems/gauges/PHI/indicator/digit1A</p>
                                <v>100</v>
                            </product>
                            <product>
                                <p>systems/gauges/PHI/indicator/digit2A</p>
                                <v>10</v>
                            </product>
                        </sum>
                    </difference>
                </integer>
            </function>
            <output>systems/gauges/PHI/indicator/digit3A</output>
        </fcs_function>

mike402, 2018-11-03:
I used to have the same functions until Richard Harrison told me how
to make proper transitions.

Down the road we're gonna find many functions that are just as useful,
but do not directly map to C/C++ and are big enough to get some hidden
mistakes or have some detail different from aircraft to aircraft -- it
would be beter to have them in some kind of a "library" that you can
override from aircraft, without having to wait until the fix has made
it into the release, because many users are on precompiled version.
Not that many, if any, people ever fly my aircraft, but hey -- it's a
very bad habit to require unstable version. :)

For example, I read/heard somewhere the transition is actally not
linear, but S-shaped, so the functions I have are only first
approximation. Also I'm sure the length/shape of transition varies
from indicator to indicator.

On Sat, Nov 03, 2018 at 02:00:27AM -0700, Adriano Bassignana wrote:
> For a 3-digit Genoa-Drive in the G91R1B I use these three functions, they
> are quite compact, but obviously I would like to have an explicit function.
> It would be desirable to have a function similar [...]

> On Thu, Nov 1, 2018 at 3:36 PM Bertrand Coconnier <notifications@github.com>
> wrote:
> 
> > This is because I don't think programming every conceivable function in
> > C++ (sawtooth function, stair functions, digit extraction, etc.) is a
> > viable option for the long-term.

abassign, 2018-11-03:
Unfortunately it is the problem of JSBSim to be compiled in C ++ and
therefore it is not possible to insert particular library functions related
to a certain application. That is possible, but only for personal use. An
interesting solution could be by binding to other languages, such as
Python, and using those to implement the missing functions to a certain
project. However, having a function for extracting a single digit would be
convenient for solving many problems with the construction of
instrumentation interfaces. Other simulation languages do it. The problem
was then linked to an actuator that allows to operate with repetitive
values, this because the actuator is an excellent concept, but operates
only in a linear way. We hope that someone can make this improvement. You
have to think that for the 6 digits I use for the PHI of the G91R1B, over
400 lines of code are only for the 6 numbers to show :)! It is not only a
problem of workload, but also of code to follow.

abassign, 2018-11-03:
I would add however that JSBSim is a language for the management of
continuous systems and the modifications must be compatible with these
systems. The problem of digit extraction can pose this problem. Many
simulation environments (eg Openmodelica) use separate networks for the
management of continuous systems from binary systems and this allows to
interface then the two worlds through interfaces that avoid the generation
/ propagation of errors due to the presence of singularities in the
derivation.

bcoconni, 2019-01-21:
@abassign 
I think PR #160 also solves this issue ?

abassign, 2019-01-22:
Hi, no, but I'm thinking of a solution I've just described in the
linearizer development thread. My idea is to make a sort of shift-register
with a variable numerical base, something like a gearbox, like the ones
used for clock drives. I think to make a prototype to better explain the
concept and try to understand together if it makes sense. Maybe there is
only one exertion to learn new things, but I'm curious to try.
Shift-registers are the basis of mechanical calculation systems and analog
clocks and can be used to build serial transmission systems. Even with the
linearizer I think I could do the same thing with the <hysteresis>
parameter I try, if it works I insert it in the documentation as well.