JSBSim-Team/jsbsim #127: Bogey: no way to adjust side force

mike402, 2018-10-31:
There is no way to separately adjust the side force from bogeys. ``<rolling_friction>`` seems to adjust both front and side force, and if I try to increase it I get too much front force.

The bogeys are too "slippery", and I can not reproduce the crosswind landing behaviour for the Tu-144. The reports say it should align itself with the runway after a crabbed touchdown without much rudder, but I always have to apply rudder to keep it from rolling off. Neither can I turn during taxi without too much braking.

https://gitlab.com/mdanil/Tu-144/blob/master/Tu-144-jsbsim/gndreac.xml , first 3 contact points.

mike402, 2018-10-31:
Actually, ``<rolling_friction>`` does not change side force at all.

abassign, 2018-10-31:
I also verified this problem with the G91R1B, for example if there is a side wind the G91 rotates like a flag ... The thing is very bad because it makes very unrealistic a takeoff and landing with side wind or slope track lateral. I thought I had something wrong in the code, but this post confirmed my suspicion. One solution is to place three lateral force vectors that follow the rules of the carts (static friction coefficient, surface of the truck in contact with the ground, load), but what are these rules? Once identified it is simple because they are normal forces to motion. It would be better, before using a "do it yourself" solution to have an official solution.

bcoconni, 2018-11-01:
At the moment, JSBSim is using [Pacejka "magic formula" for tires model](https://en.wikipedia.org/wiki/Hans_B._Pacejka#The_Pacejka_%22Magic_Formula%22_tire_models) with coefficient which are hard coded in the [C++ code of FGLGear.cpp](https://github.com/JSBSim-Team/jsbsim/blob/b914b2480217354c377fbec909a933875d8e8223/src/models/FGLGear.cpp#L221).

The problem I see with the formula, as it is implemented in JSBSim, is that as long as there is no side slip velocity, there is no side friction force. This is most likely the reason why you are observing the behaviors you describe above.

> There is no way to separately adjust the side force from bogeys.

That is incorrect you can provide a `<table>` that describes how side friction coefficient depends from the wheel slip angle.
```xml
<table type="CORNERING_COEFF">
  ...
  {wheel_slip angle} {friction coefficient}
  ...
</table>
```
I have to admit that the documentation is not very clear about that functionality :flushed: In addition, since it is seldom used, I give no guarantees about its results.

Let me know if you face problems and I will try to address them.

Finally, there is also the issue #89 on a related topic and we might need to address the 2 issues as the 2 sides of the same problem. And my experience is that friction is definitely not an easy topic to solve :disappointed:

mike402, 2018-11-01:
Wow, if that works I won't have to wait until the next release, I'll give it a try tonight.

What is the default value of ``CORNERING_COEFF`` , or what is better to use there as a starting point?

bcoconni, 2018-11-01:
> What is the default value of `CORNERING_COEFF` , or what is better to use there as a starting point?

As a starting point I would grossly mimic the shape of Pacejka's formula curve
[image]

The first column of the table should give the side slip angle in degrees and second column the friction coefficient

```xml
<table type="CORNERING_COEFF">
  -10. -1.0
  -2.0 -1.0
   2.0  1.0
   10.  1.0
</table>
```
From there I would modify the profile of the curve until it gives the desired results.

Note that as soon as the wheel/bogey slips, this table will be ignored and the dynamic friction coefficient will be used instead. And as I am writing these lines I am realizing that the whole thing might probably not work !?!

If you don't mind, I will let you (@mike402 and @abassign) drive the experiments and then based on your feedback we will discuss how we can proceed further.

bcoconni, 2018-11-01:
Another interesting experiment would to check what happens if you specify a positive friction coefficient even for negative slip angles ?
```xml
<table type="CORNERING_COEFF">
  -10. 1.0
  -2.0 1.0
   0.0 0.1 <!-- and what about a non zero coefficient at zero slip angle ? -->
   2.0 1.0
   10. 1.0
</table>
```

mike402, 2018-11-01:
Thanks, I will try it and do some crabbed landings either tonight or on the weekend.

mike402, 2018-11-01:
Looking at this quickly, maybe it's a good idea to add some simple scaling factor for Pacejka formula, which would roughly be the number of tyres on bogey?

mike402, 2018-11-01:
For the ``<table type="CORNERING_COEFF">`` FlightGear-next crashes with ``Fatal error (const char*):   An unknown table type attribute is listed: . Execution cannot continue.``. If I remove `` type="CORNERING_COEFF"`` or put any other value in the quotes, it loads ok but the table does not have any effect.

mike402, 2018-11-02:
Thanks, just tried it with nose wheel steering and it looks very promising -- there was no longer that icy-slippery feel to taxiing.

bcoconni, 2018-11-02:
> Thanks, just tried it with nose wheel steering and it looks very promising -- there was no longer that icy-slippery feel to taxiing.

Great ! :+1: 

Let us know what you learnt in the process once you will have gathered enough experience with that feature. Possibly the default function to compute the side force (Pacejka) can benefit from your feedback :smile: ?

mike402, 2018-11-02:
I think I should start with a Python script to print Pacejka for user-defined parameters in the <table> format -- I'll try to write it tomorrow.

mike402, 2018-11-03:
Ah, I think I get it! Pacejka is actually fine -- it is just that many people including me have been using ``<static_friction>`` to set maximum braking force of the wheel brakes, but it is used as scaling factor for Pacejka formula at https://github.com/JSBSim-Team/jsbsim/blob/b914b2480217354c377fbec909a933875d8e8223/src/models/FGLGear.cpp#L225

I put a larger ``<static_friction>`` on the nose wheel and the steering improves a lot.

mike402, 2018-11-04:
Looks like I'm done. I had expected to spend a week or more on it, but it took only a day, so it was fast ^_^.

1. For FlightGear release I managed to make taxi steering satisfactory by setting larger ``<static_friction>``. But it's a bit too stiff for landings, causing rollout tendency and rolling moment if you are not careful enough and derotate with side slip:

https://gitlab.com/mdanil/Tu-144/commit/662910fdbabdcca8c227a0a9050121e2c3358003

2. For FlightGear-next, now that ``CORNERING_COEFF`` is working (I really really hope that commit gets cherrypicked), I made nose wheels less stiff and gave the main bogeys enough force to help align with the runway, but made them soft enough not to cause dangerous rolling moment in max crosswind.

https://gitlab.com/mdanil/Tu-144/blob/2018.3.0/Tu-144-jsbsim/gndreac.xml

Later I may want to check and reduce strut damping, because it seems to cause a bit of bouncing, but at least the side-force part is okay now.

Thank you for the timely bugfix!

P.S.: Might be useful for someone scratching their head with "why in the world can't I decrab": I also found a problem in my FDM: the "Yaw moment due to beta" was too large, making the aircraft turn into the wind on the rollout beyond rudder authority. Naturally, that also kept the aligning effect of main gear from being felt -- so I wasted most of my time with too-high side forces, only to watch the plane turn upside down because of rolling moment.

P.P.S.: Is there a repository for JSBSim development tools to which I can give my Pacejka script, so that it does not get lost?

mike402, 2018-11-04:
About improving built-in Pacejka. Personally, I I think I would no longer need to manually give ``CORNERING_COEFF`` tables, if the built-in Pacejka could be like:

1. Its parameters exposed to some tags under ``<contact>``, something like ``<cornering_peak>``, ``<cornering_stiffness>``, etc.
2. ``Peak`` at https://github.com/JSBSim-Team/jsbsim/blob/b914b2480217354c377fbec909a933875d8e8223/src/models/FGLGear.cpp#L225 is optionally decoupled from from ``<static_friction>`` by supplying ``<cornering_peak>`` under ``<contact>``.

The second scaling times ``<static_friction>`` https://github.com/JSBSim-Team/jsbsim/blob/b914b2480217354c377fbec909a933875d8e8223/src/models/FGLGear.cpp#L588 should be kept for compatibility, but perhaps there should be made a way to override it, because what if some case needs zero forward force, but non-zero side force.

I think with these changes the built-in formula should be enough for most cases.

legoboyvdlp, 2018-11-07:
copying my comment:

> Hi,
Many users are tuning brakes using <static_friction> - this results in correct braking behaviour, but really seems to mess up steering - is it possible to have a property that tunes the brake force without affecting the static friction of the wheels ( resistance to lateral force if I remember right? )

legoboyvdlp, 2018-11-07:
For reference, I'm speaking not about side force when the wheels hit the ground on a crabbed landing, but trying to turn at ~20 knots, especially when on grass - sometimes, when I command full right, I actually go left!

legoboyvdlp, 2018-11-07:
@it0uchpods would know more about this than me, he is the one that adjusted the brakes on the A320

mike402, 2018-11-07:
By the way, far as I understand typically there is no brakes on the
front wheel, apart from the small pads to stop it when it's retracting
-- so the ``<static_friction>`` can be arbitrarily high there, but
like I said, it can get too stiff and kill you on accidental
derotation with side speed.

On Wed, Nov 07, 2018 at 03:09:02AM -0800, Jonathan Redpath wrote:
> @it0uchpods would know more about this than me, he is the one that adjusted the brakes on the A320

bcoconni, 2018-11-10:
@legoboyvdlp 
The side friction that is generated by a steering wheel is not different than the side friction that occurs during a crabbed landing. Both are managed by the Pacejka function that has been discussed above. 

A steering wheel can modify the orientation of the side friction force. And since the side friction force is generally significantly higher than the rolling friction the aircraft, which tends to follow the path of least resistance, will follow the rolling/steering direction.

Now that I fixed the bug with the `CORNERING_COEFF` tables, you can specify the side friction via this feature.
> but also noticeable on asphalt - sometimes, when I command full right, I actually go left!

I am still bothered about the fact that the Pacejka function returns a negative friction coefficient for negative steering angles. I would be curious to know if that direction inversion still occurs if the friction coefficient from the Pacejka would be forced to an always positive value.

There might be another explanation for that behavior: is there some strong cross wind when you observe the inverted steering command behavior ? I mean while you are on the grass, FlightGear might reduce the friction coefficient making the steering wheel unable to resist cross wind ?

legoboyvdlp, 2018-11-10:
Yes, I think that is what is happening.

legoboyvdlp, 2018-11-10:
Regardless, I'll see if the nose wheel can be adjusted by changing static friction and using cornering coeff once 2018.4 is released.

bcoconni, 2018-11-17:
@legoboyvdlp 
> I'll see if the nose wheel can be adjusted by changing static friction and using cornering coeff once 2018.4 is released.

FYI the fix for the `CORNERING_COEFF` feature has been pushed to FlightGear 2018.3.

legoboyvdlp, 2018-11-17:
Excellent, thanks.

bcoconni, 2018-12-27:
@mike402 and @legoboyvdlp 
Can this issue be considered closed ? Or do you need some further support ?

mike402, 2018-12-27:
I think now that this works it can be closed, unless someone wants to
expand the default Pacejka the way I had described.

Thank you for fixes!

legoboyvdlp, 2018-12-27:
I am experimenting with it, and it works really well for nose wheel steering!
@mike402  python file helps a lot :)

bcoconni, 2018-12-27:
OK thanks. Issue closed.

closed 2018-12-27 by bcoconni