JSBSim-Team/jsbsim #128: Problem for use of the actuator in rotary displacement when it exceeds 360 ° ...

abassign, 2018-10-31:
The use of the actuator allows obtaining a very realistic result for linear or rotary displacement systems but limited within a certain range of values. While it does not work properly for multi-turn rotary systems. For example this wind speed display is made by simulating an electromechanical component. The effect between 0 and 9 is perfect, but between 9 and 0 or between 0 and 9 (backward rotation) the actuator finds its equilibrium state after a complete rotation. The problem is that Actuator does not provide the ability to manage these particular types of devices. The same problem is present for the lap-compass needle which, when it exceeds 360 °, makes a further rotation to find the equilibrium point.

This is an example:
https://youtu.be/SnTTpJdyFlw

`<actuator name="digit3AW-inc">`
`            <input>systems/gauges/PHI/wind/digit3A</input>`
`            <lag>0.0</lag>`
`            <rate_limit>10</rate_limit>`
`            <bias>0.002</bias>`
`            <deadband_width>0.01</deadband_width>`
`            <hysteresis_width>0.01</hysteresis_width>`
`            <output>systems/gauges/PHI/wind/digit3AW</output>`
` </actuator>`

I thought a lot about this problem, and I sensed that in fact the "actuator" function works very well for linear displacements or with values within a certain range, but it can not operate correctly on cyclic values, like those presented here. I solved the problem with a couple of complex fcs-functions, but I'm certainly not satisfied. I think it would be much better to insert a control that, in the case of actuators that have to control systems that perform an indeterminate number of turns, correct the discontinuity both forward and backward, avoiding the problem shown here in the movie.

My solution is in the list displayed: https://i.postimg.cc/RFyGmqYz/Screenshot-20181031-222326.png

That is, I transform the problem by constructing a sort of counter, then first calculating the variation and then adding the value currently used by the actuator function, if it exceeds the maximum value (in the example 9.9 for pisitive variation) I set the displayed value to zero. On the other hand, if the new value is less than zero, I set the displayed value to 9.9.

Obviously there are many different ways, but it would not be difficult to allow the actuator to manage this type of device.

I'd like to know if this change can be made.

Thank you

mike402, 2018-10-31:
My solution would be to filter sine and cosine separately, but it is way too bulky and perhaps has hidden gotchas.

For now I have to leave 360-degree needles in instruments and such unfiltered. Same problem with FlightGear propertyrule-based filtering.

abassign, 2018-11-01:
> My solution would be to filter sine and cosine separately, but it is way too bulky and perhaps has hidden gotchas.

The Actuator module is very nice and gives excellent effects, but it has been realized for linear and non-rotary actuators, like the numbered drum of my example or a gyrocompass needle. Looking at the code it is possible, with a few interventions, to solve the problem by introducing the concept of min-max. In my opinion it is enough to introduce these two parameters and by analyzing the difference between the input value and the real value in which the actuator is located, correctly manage the min-max.

bcoconni, 2018-11-01:
Makes sense to me to extend `<actuator>` to cyclic devices. However I'm not sure what you mean by "min-max" I'd rather address the problem with a floating point [modulo operation](https://en.wikipedia.org/wiki/Modulo_operation) ?

mike402, 2018-11-01:
This is probably not the right place to mention this, but I noticed if there's ``<clipto>`` in integrators and such, they still integrate beyond ``<min>`` and ``<max>`` inside, which is not obvious and imho is error prone. What one has to do in this case is make a ``<switch>`` that modifies its ``<input>`` or ``<cX>`` to limit the integrations. It would probably be best to make the filters respect ``<clipto>`` internally or have additional tags for that, and a ``<cyclic>`` tag that would make them jump between the two values when either of them is exceeded.

abassign, 2018-11-01:
The min-max method (in the case that I presented with the code is 0-9 because the drum is divided into 10 faces) was made by me in order to obtain a simple and fast solution to the problem. But today I verified that it does not work if its output is inserted into the actuator! The problem lies in the fact that the actuator "follows" the proposed value at the input and if I apply the method proposed by me when it reaches the passage point 0-> 9 or 9-> 0 the actuator starts to oscillate around these two values . I have also tried another method, insert a block of subtraction between the proposed value and the output one from the actuator and use this difference as input to the actuator. My idea was to make the actuator work like a counter, but unlike an integrator, the actuator has no memory and therefore the value of roration is half of the real one and then become correct at the point of passage between 9 - > 0 or 0 -> 9. Actually it's not a simple problem :)

bcoconni, 2018-11-10:
@abassign 
I gave some thoughts to your problem, my proposition is to add an attribute to `<clipto>` such as
```xml
<clipto mode="cyclic">
  <min> 0.0 </min>
  <max> 10.0 </max>
</clipto>
```
The actuator would then apply a `modulo` to the output such as if the input value is `11.0` then it will return `1.0` or if the input value is `-2.0` then it will return `8.0`. That way the output will follow a cycle such as is needed for a needle.

For backward compatibility, if the attribute `mode` will not be set or will be set to `"clamp"` then `<clipto>` will behave as usual.

However, for the actuator to interpret an increase (i.e. for the needle to turn clockwise), the current value must be **higher** than the previous value (i.e. `9.0` -> `11.0`) which means you should supply an input value (`11.0`) higher than the `<max>` value specified in `<clipto>` (which `10.0` in the example above).

Similarly, for the needle to turn counterclockwise, you should supply an input value that is always  lower than the current value and could be therefore lower than the `<min>` value specified in `<clipto>`.

The problem is that if the current value is `1.0` while the previous value was `9.0` there is no way to determine if the value actually decreased from `9.0` to `1.0` or increased from `9.0` to `1.0` (which should be interpreted as `11.0` clipped to `1.0`).

Would such a feature address your problem ?

mike402, 2018-11-10:
On Sat, Nov 10, 2018 at 04:14:31AM -0800, Bertrand Coconnier wrote:
> However, for the actuator to interpret an increase (i.e. for the needle to turn clockwise), the current value must be **higher** than the previous value (i.e. `9.0` -> `11.0`) which means you should [...]

I think what you can do to address this is to choose the shortest of
the 2 turns to the target position, for example for an actuator that
does 0..360 it would be determined by the +-180 degree sawtooth.

bcoconni, 2018-11-10:
> I think what you can do to address this is to choose the shortest of the 2 turns to the target position, for example for an actuator that does 0..360 it would be determined by the +-180 degree sawtooth.

This is a plausible assumption but how would you know for sure that it is the desired behavior ?

abassign, 2018-11-10:
The problem is that a cyclic rotating actuator does not work with the concept of absolute value, but with the concept of increase-decrease. This means that its value can take more than 360 ° (if you measure the rotation in degrees) and if this happens it must be subtracted or added up (depending on the direction).
For example, let's say that the minimum is zero and the maximum nine, when we exceed 9, for example 10, it must be subtracted 10. For example: 8,9,10 -> 0 vice versa 2,1,0, -1 becomes 2,1 , 0,9. 
I find excellent the idea of the "cyclic" attribute, the extremes will be set for example: 0-9 the difference is calculated 10, this is the value to subtract or add up according to the direction of rotation, for a number of times up to arrive at a value included within the min-max limits.

mike402, 2018-11-10:
On Sat, Nov 10, 2018 at 09:39:38AM -0800, Adriano Bassignana wrote:
> The problem is that a cyclic rotating actuator does not work with the concept of absolute value, but with the concept of increase-decrease. This means that its value can take more than 360 ° (if you [...]

The actuators for which the number of turns matters should not use
cyclic, imho. Cyclic implies something like ``<fmod>``.

abassign, 2018-11-10:
Mike402, i think is correct is a particular "mod" type ... in this moment I not remember the exact math notation of this particular type module. But i think my concept is clear.

bcoconni, 2018-12-30:
@abassign 
The feature for cyclic clipping has been committed to JSBSim and pushed to FlightGear.
The feature is the same as described above and its syntax is
```xml
<clipto type="cyclic">
  <min> 0.0 </min>
  <max> 10.0 </max>
</clipto>
```
Could please test and let me know if that works ?

abassign, 2019-01-02:
@bcoconni 
Hi, I did a test but ... [FIAT G91R1B test JSBSim actuator in rotary displacement when it exceeds 360 test 01](https://youtu.be/Gv5_nfvFcTI) 
This is the code:
```
        <actuator name="systems/gauges/PHI/attitude/gyrocompass-magnetic-deg-var-operative-lag">
            <input>systems/gauges/PHI/attitude/gyrocompass-magnetic-deg-var-operative</input>
            <lag>2.0</lag>
            <rate_limit>100</rate_limit>
            <bias>0.1</bias>
            <deadband_width>1</deadband_width>
            <hysteresis_width>0.5</hysteresis_width>
            <clipto mode="cyclic">
                <min> 0.0 </min>
                <max> 360.0 </max>
            </clipto>
        </actuator>
```
I've done some tests, it seems ok, but it does not work in case I have placed in my request. The reason is due to the fact that the cyclic function when it passes from 9 to 0 (in your example) or from 0 to 360 in the case that I show you generates a discontinuity that is interpreted by the XML function "<animation> <type> rotate </ type> ... "as a return to zero.

If, however, I think for a moment it is right that the system does so, a rotation is continuous so when it goes through zero clockwise it should insert an angle of 360 ° (maximum value of the clip), if instead it goes through zero counterclockwise It was supposed to take a corner off. In fact, every time that passes from zero, ie the discontinuity takes place, this is solved with this method of adding or subtracting values equal to the maximum permissible value. Obviously there is the problem of negative values that must be solved in the same way. In topography (field in which I worked) we have the same problem with the corners and it is solved in the same way.

The solution you have put is not a cyclic solution, but a cycloid transformation. In fact, I would change the name to cycloid, which is always an important function, for example in alternative motions such as the plunger of a locomotive. Therefore, I would keep it as I find it interesting even if I would change its name.

The cyclic solution instead must give a continuous result, without discontinuity, so as not to give problems. Not only that, but this solution allows the immediate calculation of the number of laps and the possibility of integrating and deriving the results obtained.

mike402, 2019-01-02:
Animations have nothing to do with JSBSim. But if a 0..360 rotate
animation is done right, the user will not see any discontinuity
because animations work immediately without any lagging.

abassign, 2019-01-02:
@mike402 
I understand your perplexity, but we are not talking about an animation problem, but about an actuator that has to solve the problem of a rotation. If clipping produces a non-derivable function, a discontinuity is generated, which is not very compatible with a continuous simulation system. What I'm trying to make clear is the importance of introducing an actuator that can handle a rotary device from an angular input. In the example I inserted, it is clear that it is not possible to manage a magnetic compass (or ADF) by means of an actuator, since the information produced by JSBSim is a limited angle between two values (for example, 0..360 °). So if you apply a lag filter, it can not operate properly if there is a point of discontinuity. To resolve this problem can operate by first transforming the appropriate angular value into an absolute angular value and then apply the actuator ... it is a considerable complication and I'm sorry that is not already present in jsbsim.
However I understand that it is not an easy concept and certainly I'm explaining it badly! In short, the angular actuator must solve a dynamic problem linked to the direction of rotation. In the implementation of this angular actuator there should be a flag indicating the direction of rotation, so as to apply the sum of the maximum value (or its subtraction) at the point of discontinuity. I hope I can have the time these days to do a C ++ test in order to better explain the problem.

mike402, 2019-01-03:
I've just tried this on fg-next

~~~
  <actuator name="fubar">
   <output>fubar-out</output>
   <rate_limit>1000.0</rate_limit>
   <input>fubar-in</input>
   <clipto mode="cyclic">
    <min>0.0</min>
    <max>360.0</max>
   </clipto>
  </actuator>
~~~

and it doesn't work at all, I see it just stops at 360 if the input exceeds that. I have also tried replacing ``mode="cyclic"`` with ``type="cyclic"``, same result.

mike402, 2019-01-03:
Sorry, I had made a comment about it not working, turns out I did not have that commit.

It works very well, exactly like the sawtooths I had suggested in #84!

Thank you, now I can rid my plane of lots of ugly <fcs_function> copypasta! \o/

mike402, 2019-01-05:
On Sat, Jan 05, 2019 at 05:27:41AM -0800, Bertrand Coconnier wrote:
> This is the modulo function
> [image]

Wait, isn't modulo symmetric around the 0? If it weren't, I wouldn't have suggested the addition of sawtooths?

bcoconni, 2019-01-05:
> Wait, isn't modulo symmetric around the 0? If it weren't, I wouldn't have suggested the addition of sawtooths?

That is just an illustration. It really depends on the `<min>` and `<max>` values you provide to the clipping function. If you set `<min>` to zero then you should get what you are expecting.

abassign, 2019-01-07:
Hi, I made a prototype in the my local SRC library ... _JSBSim/models/flight_control to test the possibility of creating a device that solves the problem object of this discussion. [JSBSim - Linear Actuator test ](https://youtu.be/9WE-ElasXrY).
From Wikipedia [the linear actuator](https://en.wikipedia.org/wiki/Linear_actuator) it is described as a converter of motion of the rotary type (linear gear coupled to a rotating gear, motor axis, worm gear type Genova etc ...). This clear description allowed me to realize my prototype of a _linear actuator_ that should not be confused with the actuator already present in JSBSim. The two objects can be coupled together because the linear actuator converts the rotation into a linear displacement, while the standard actuator inserts some features in order to make it closer to reality.
The final result seems to me to be excellent and solves the problem perfectly without changing the current JSBSim structure. Introduces some features that I have not seen in the JSBSim modules as a SET parameter used to store the output value (SET = 1/0) similarly to what happens in the [JK Flip-flops](https://it.wikipedia.org/wiki/Flip-flop) And a parameter RESET = 0/1 when is necssary set the initial device position. 

I hope the code can be inserted into the JSBSim experimental repository so that we can solve the problem and take advantage of the new opportunities that the module offers.

File: FGLinearActuator.h

`/*%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
 * 
 * Header:  FGLinearActuator.h
 * Author:  Adriano Bassignana
 * Date started: 2019-01-02
 * 
 * ------------- Copyright (C)  -------------
 * 
 * This program is free software; you can redistribute it and/or modify it under
 * the terms of the GNU Lesser General Public License as published by the Free Software
 * Foundation; either version 2 of the License, or (at your option) any later
 * version.
 * 
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS
 * FOR A PARTICULAR PURPOSE.  See the GNU Lesser General Public License for more
 * details.
 * 
 * You should have received a copy of the GNU Lesser General Public License along with
 * this program; if not, write to the Free Software Foundation, Inc., 59 Temple
 * Place - Suite 330, Boston, MA  02111-1307, USA.
 * 
 * Further information about the GNU Lesser General Public License can also be found on
 * the world wide web at http://www.gnu.org.
 * 
 * HISTORY
 * --------------------------------------------------------------------------------
 * 
 * %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
 * SENTRY
 * %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%*/

#ifndef FGLINEARACTUATOR_H
#define FGLINEARACTUATOR_H

/*%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
 * INCLUDES
 * %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%*/

#include "FGFCSComponent.h"
#include "simgear/props/propertyObject.hxx"

/*%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
 * FORWARD DECLARATIONS
 * %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%*/

namespace JSBSim {
    
    class Element;
    
    /*%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
     * CLASS DOCUMENTATION
     * %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%*/
    
    /** Models a flight control system summing component.
     *    linear actuators typically operate by conversion of rotary motion into linear motion.
     *    The linear_actuator component is decrobed in the Wikipedia link:
     *    https://en.wikipedia.org/wiki/Linear_actuator
     *    Converts a rotation into a linear movement or a multiturn rotation.
     *    For the conversion it is necessary to declare a module
     *    that is defined by the difference of the maximum and minimum value Input.
     *    List of parameters:
[... 401 lines cut ...]
                for (auto node: OutputNodes)
                    cout << "     OUTPUT: " << node->GetName() << endl;
            }
        }
        if (debug_lvl & 2 ) { // Instantiation/Destruction notification
            if (from == 0) cout << "Instantiated: FGLinear_Actuator" << endl;
            if (from == 1) cout << "Destroyed:    FGLinear_Actuator" << endl;
        }
        if (debug_lvl & 4 ) { // Run() method entry print for FGModel-derived objects
        }
        if (debug_lvl & 8 ) { // Runtime state variables
        }
        if (debug_lvl & 16) { // Sanity checking
        }
        if (debug_lvl & 64) {
            if (from == 0) { // Constructor
            }
        }
    }
}
`
The following system files were then modified to activate the module:

JSBSim/models/FGFCS.cpp
JSBSim/CMakeLists.txt

abassign, 2019-01-08:
[This is the GIT](https://github.com/abassign/JSBSim-Test-new-functions.git)

abassign, 2019-01-08:
Sorry if I inadvertently closed the discussion, I reopened in the hope of having comments and help in being able to correctly enter the code I wrote in the test repository of JSBSim.
Thanks to all those who are collaborating to solve this problem.

mike402, 2019-01-12:
By the way, as ``<clipto>`` only clips/sawtooths the output of integrators, actuators, the internal integration still goes on as usual -- can an overflow happen at some point and what would be the consequences?

bcoconni, 2019-01-12:
@mike402 
> By the way, as `<clipto>` only clips/sawtooths the output of integrators, actuators, the internal integration still goes on as usual -- can an overflow happen at some point and what would be the consequences?

The parameter `<trigger>`  of the PID was created for that very purpose. Quoting [the docs](https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGPID.html):
> trigger - Property which is used to sense wind-up, optional. Most often, the trigger will be driven by the "saturated" property of a particular actuator. When the relevant actuator has reached it's limits (if there are any, specified by the <clipto> element) the automatically generated saturated property will be greater than zero (true). If this property is used as the trigger for the integrator, the integrator will not continue to integrate while the property is still true (> 1), preventing wind-up. The integrator can also be reset to 0.0 if the property is set to a negative value.

bcoconni, 2019-01-21:
@abassign 
I guess that now PR #160 has been committed, this issue can be closed ?

bcoconni, 2019-01-23:
> I would close the #160 because the work is finished. Please, tell me how I should do

Well PR #160 is now closed since it has been merged. If you submit a new PR it will have another ID.

I was just asking if the issue #128 (this issue) could be closed as well since it is basically requesting what has been implemented by the PR #160. So unless you think there are still some topics to discuss about issue #128, you can click on the button `Close issue` below.

abassign, 2019-01-30:
The issue was solved by creating the linear_actuator component.

closed 2019-01-30 by abassign