JSBSim-Team/jsbsim #131: As Massbalance execute before Propulsion,  the TankInertia is not in the same CG with basJ.

jinzhu1911, 2018-11-07:
In the function Run of FGMassBlance ,  use statement "GetPointmassInertia( lbtoslug * EmptyWeight, vbaseXYZcg )" to transfer the baseJ to the new CG, but in the statement "mJ += in.TankInertia" which TankInerita is calculated in the old CG.
Thanks for your help!

bcoconni, 2018-12-15:
> in the statement `mJ += in.TankInertia` which TankInerita is calculated in the old CG.

That's correct. Not sure this makes much difference since the CG position is changing very slowly but strictly speaking you are correct.

If you need higher precision for your application, feel free to submit a patch.

bcoconni, 2019-03-17:
No feedback for 3 months. Issue closed.

closed 2019-03-17 by bcoconni