JSBSim-Team/jsbsim #132: How to use an external wind map?

fabioandraderj, 2018-11-14:
I am working with Unmanned Aerial Vehicles and I would like to use forecast wind maps on JSBSim simulations. I know I should use the function "SetWindNED(x,y,z)", but I don't know exactly how to implement this. I only use JSBSim by starting it with the Ardupilot simulator.

I have a file with the wind data for latitudes and longitudes (2.5 km resolution) and would like that JSBSim used this wind data when calculating the flight dynamics. I would appreciate if you guys could give me some hints on how to start and which documentation to check.

seanmcleod, 2018-11-28:
You have 2 options. 

One option is to write some C++ code and add it to the JSBSim source code in `copy_to_JSBsim()`, where your code would lookup the wind vector during each simulation time step and then call `SetWindNED()`. In this case you would build a new custom JSBSim executable.

The second option is to leave the JSBSim executable as-is and write your own external program to query JSBSim for the current aircraft position and then after looking up the relevant wind vector set it via a network (UDP/TCP) connection by updating the following properties:

`
atmosphere/wind-north-fps   
atmosphere/wind-east-fps   
atmosphere/wind-down-fps
`

In terms of the second option take a look at this issue - https://github.com/JSBSim-Team/jsbsim/issues/74

fabioandraderj, 2018-12-03:
Hey. Thank you very much for the reply. I will go for the first option then. 

I will write the code bellow in the copy_to_JSBSim() function, right? But where is this function? Is it the "bool FGJSBsim::copy_to_JSBsim()" at JSBSim.cxx? If so, the function SetWindNED is called there, so would I replace it by the new one?

```
wNorth = function_wind_north(Propagate->GetLatitudeDeg(),Propagate->GetLongitudeDeg(),Propagate->etGeodAltitude());
wEast = function_wind_east(Propagate->GetLatitudeDeg(),Propagate->GetLongitudeDeg(),Propagate->etGeodAltitude());
Atmosphere->SetWindNED(wNorth, wEast, 0);
```

bcoconni, 2018-12-08:
> If so, the function SetWindNED is called there, so would I replace it by the new one?
```c++
wNorth = function_wind_north(Propagate->GetLatitudeDeg(), Propagate->GetLongitudeDeg(), Propagate->GetGeodAltitude());
wEast = function_wind_east(Propagate->GetLatitudeDeg(), Propagate->GetLongitudeDeg(), Propagate->GetGeodAltitude());
Atmosphere->SetWindNED(wNorth, wEast, 0);
```
That seems reasonable.

You may also use `Propagate->GetGeodLatitudeDeg()` to get the geodetic latitude and be consistent with the use of the geodetic altitude ?

bcoconni, 2019-01-13:
@fabioandraderj 
Has your question been answered ? And if so, can we close this issue ?
Thanks.

fabioandraderj, 2019-01-13:
Hi! Thank you very much for the support. I am coming back from holidays on Monday and will try to implement it. Please let me see if I am able to do it before closing the issue. Thanks!

fabioandraderj, 2019-01-19:
I implemented the code but now I am struggling building the modified JSBSim. I am using https://github.com/tridge/jsbsim repository. The executable file in the folder jsbsim/src. I installed jsbsim using the command ./autogen.sh. Now, if I change the file jsbim/src/JSBSim.cxx including my code and use the command ./autogen.sh, it doesn't use my code. I also tried to cmake and make but it didn't work. How can I compile the executable using my modifications on JSBSim.cxx? Thanks!

bcoconni, 2019-01-20:
@fabioandraderj 
> I am using https://github.com/tridge/jsbsim repository.

First of all, I have checked your repo and you are using an old version of JSBSim (forked from [arktools/jsbsim](https://github.com/arktools/jsbsim) which has not been updated for 7 years and from [SourceForge](https://sourceforge.net/projects/jsbsim) which is no longer updated for almost one year).

I would rather encourage you to clone our GitHub repo [JSBSim-Team/jsbsim](https://github.com/JSBSim-Team/jsbsim) to get the latest stuff including bug fixes.
> Now, if I change the file jsbim/src/JSBSim.cxx including my code and use the command ./autogen.sh, it doesn't use my code. I also tried to cmake and make but it didn't work. 

It does not seem you have committed your code to your repo https://github.com/tridge/jsbsim. At least I could not find any commit by your name except on the branch `gethostbyaddr-fix` which dates back from 2015. So it is unlikely that it contains the code you are currently working on :smile: 

It is difficult to answer because you do not give much details. However my guess is that you added new source files to the project and you did not tell the build tool to include them in the build.

Say you have a new source file `ExternalWinds.cpp` in the directory `src/models/atmosphere` then you have to update `src/models/atmosphere/CMakeLists.txt`
https://github.com/JSBSim-Team/jsbsim/blob/10768d2b12a58853cf78a75a43a8c8153f54e55c/src/models/atmosphere/CMakeLists.txt#L1-L5
The modification to the file would look like the diff below.
```diff
-   FGWinds.cpp)
+   FGWinds.cpp
+   ExternalWinds.cpp)
```
If that still does not work, could you please post the error messages you get from `cmake`, `make`, etc. ?

bcoconni, 2019-01-20:
> Therefore, I would like to know where I should call the function to load the txt file so it is permanently loaded and then I can just access the array on the FGWinds::Run function.

Your best option is to load it from `FGWinds::InitModel` which is just called once during the initialization of the `FGWinds` model (hence the name):
https://github.com/JSBSim-Team/jsbsim/blob/10768d2b12a58853cf78a75a43a8c8153f54e55c/src/models/atmosphere/FGWinds.cpp#L121-L135

fabioandraderj, 2019-01-22:
Thanks again @bcoconni ! Now it is working but there is something wrong with the latitude.

I added the following under FGWinds() function:
`Propagate = fdmex->GetPropagate();`

Then, when I use Propagate->GetLatitude() and Propagate->GetLongitude(), I get the correct value for the longitude but a wrong value for latitude. I also tried with GetLatitudeDeg() and also got a wrong value for the latitude. I am investigating if there is something wrong with jsbsim source code also. Do you have any idea of what the problem may be?

The source code is at my repo.

bcoconni, 2019-02-09:
@fabioandraderj 
Has your problem been fixed ? Or do you need some further help ?

fabioandraderj, 2019-02-10:
Yes, it is fixed and working. Thank you very much for the help @bcoconni! I also reported the required corrections on Ardupilot's SIM_JSBSim.cpp.

closed 2019-02-10 by fabioandraderj