JSBSim-Team/jsbsim #133: Dynamic parameters for the sensor class

abassign, 2018-11-15:
Often the sensors have problems, for example the lack of power supply, at this point it would be useful to be able to insert directly into the sensor its operating status (for example in the form of a gain parameter that is not constant, but associated with a parameter), it would be useful also to be able to have a sensor value when it is not working, also in the form of a possibly recursive parameter (for example the value of the sensor itself when it is placed in an inoperative state). Also the error parameter should be a parameter to be able to simulate the degradation of the sensor for various causes, for example an excessive acceleration etc ...

bcoconni, 2018-12-27:
Yes, this will be part of some refactoring of the gain/bias/delay feature. At the moment, there is a lot of code duplication to handle these features in the sensors, I will take this opportunity to allow "dynamic" parameters for them.

abassign, 2019-01-09:
@bcoconni Thanks for your interest, I believe that the possibility of extending variable parameters over time is very useful to simplify the code and make simulation more realistic.

bcoconni, 2019-01-10:
> Thanks for your interest, I believe that the possibility of extending variable parameters over time is very useful to simplify the code and make simulation more realistic.

No problem but don't expect it to happen any time soon as I have already plenty on my plate.

legoboyvdlp, 2019-10-31:
Just on the topic of sensors, I wonder if someone could explain what the value in `<bits> number </bits>` means; what does this number control?

seanmcleod, 2019-12-15:
@legoboyvdlp the `bits` field if specified is used to quantize the output, so for example if you specify `10 bits` then you will only see 2^10 = 1024 distinct values output, so in effect limiting the resolution you will see from the sensor.