JSBSim-Team/jsbsim #134: build failure

legoboyvdlp, 2018-11-28:
Severity	Code	Description	Project	File	Line	Suppression State
Error		command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.15.26726\\bin\\HostX64\\x64\\link.exe' failed with exit status 1120	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\CMakeLists.txt	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\EXEC	1	

Hi,
While trying to build following https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/quickstart-building-using-visualstudio/, I got this error. Does anyone know of any solution? 

Thanks,
Jonathan

seanmcleod, 2018-11-28:
A while back I remember having an issue in terms of building a debug version if I tried to build all targets via CMake. The issue was because the linker was looking for a debug version of the Python dll when building a debug version of our JSBSim Python wrapper and at least on my machine my installation of Python didn't have a debug version of the Python dll.

So I typically build a release version of the JSBSim-Python wrapper and just build a debug version of the JSBSim executable if I want to debug anything within JSBSim.

legoboyvdlp, 2020-04-15:
May as well close now...

bcoconni, 2020-04-17:
Closed as per @legoboyvdlp feedback.

closed 2020-04-17 by bcoconni